evcc-io/evcc #9043: VW/Audi/BMW vehicles not getting recongnized

mxhf, 2023-07-21:
### Describe the bug

Scince some time (couple of months?) none of our 3 vehicles get configures correctly anymore. I suspect change in the manufacturers API?

### Steps to reproduce

Try configuring any of
Audi Q4 Etron
BMW i3
VWBus T7 hybrid

Tried today with the most current stable release

### Configuration details

```yaml
- name: i3
    type: template
    template: bmw 
    title: BWM i3  
    user: xxx
    password: xxx
    vin: xxx  
    capacity: 33  
    phases: 3  
    cache: 15m  
    mode: now  
    minSoC: 00  
    targetSoC: 80  
    minCurrent: 6  
    maxCurrent: 16  
  - name: Q4_2022 
    type: template
    template: etron
    title: Q4ETRON
    user: xxx 
    password: xxx
    vin:xxx
    capacity: 50 
    phases: 3 
    cache: 15m 
    mode: now 
    minSoC: 00 
    targetSoC: 80 
    minCurrent: 6 
    maxCurrent: 16 
  - name: Q4_2023
    type: template
    template: etron 
    title: Q4ETRON  
    user: xxx  
    password: xxx  
    vin: xxx  
    capacity: 83
    phases: 3
    icon: car
    cache: 15m
    mode: now  
    minSoc: 00  
    targetSoc: 80  
    minCurrent: 6  
    maxCurrent: 32  
    priority: 1
```

### Log details

```text
Jul 21 06:20:22 raspberrypi systemd[1]: Started evcc.
Jul 21 06:20:25 raspberrypi evcc[3659]: [main  ] INFO 2023/07/21 06:20:25 evcc 0.114.1 (62a68fba)
Jul 21 06:20:25 raspberrypi evcc[3659]: [main  ] INFO 2023/07/21 06:20:25 using config file: /etc/evcc.yaml
Jul 21 06:20:25 raspberrypi evcc[3659]: [main  ] INFO 2023/07/21 06:20:25 starting ui and api at :7070
Jul 21 06:20:27 raspberrypi evcc[3659]: [db    ] INFO 2023/07/21 06:20:27 using sqlite database: /var/lib/evcc/evcc.db
Jul 21 06:20:29 raspberrypi evcc[3659]: [main  ] ERROR 2023/07/21 06:20:29 creating vehicle Q4_2022 failed: cannot create vehicle 'template': cannot create vehicle 'etron': invalid character 'd' looking for beginning of object key string
Jul 21 06:20:29 raspberrypi evcc[3659]: [main  ] ERROR 2023/07/21 06:20:29 creating vehicle Q4_2023 failed: cannot create vehicle 'template': cannot create vehicle 'etron': invalid character 'd' looking for beginning of object key string
Jul 21 06:20:30 raspberrypi evcc[3659]: [main  ] ERROR 2023/07/21 06:20:29 creating vehicle vbbus failed: cannot create vehicle 'template': cannot create vehicle 'vw': invalid character 'd' looking for beginning of object key string
Jul 21 06:20:31 raspberrypi evcc[3659]: [main  ] ERROR 2023/07/21 06:20:31 creating vehicle i3 failed: cannot create vehicle 'template': cannot create vehicle 'bmw': cannot get vehicles: unexpected status: 404 (Not Found)
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] WARN 2023/07/21 06:20:31 vehicle 'unavailable' does not support automatic detection
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] WARN 2023/07/21 06:20:31 vehicle 'unavailable' does not support automatic detection
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] WARN 2023/07/21 06:20:31 vehicle 'unavailable' does not support automatic detection
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] WARN 2023/07/21 06:20:31 vehicle 'unavailable' does not support automatic detection
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] INFO 2023/07/21 06:20:31 site config:
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] INFO 2023/07/21 06:20:31   meters:      grid ✓ pv ✓ battery ✗
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] INFO 2023/07/21 06:20:31     grid:      power ✓ energy ✗ currents ✗
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] INFO 2023/07/21 06:20:31     pv 1:      power ✓ energy ✗ currents ✗
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] INFO 2023/07/21 06:20:31   vehicles:
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] INFO 2023/07/21 06:20:31     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] INFO 2023/07/21 06:20:31     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] INFO 2023/07/21 06:20:31     vehicle 3: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Jul 21 06:20:31 raspberrypi evcc[3659]: [site  ] INFO 2023/07/21 06:20:31     vehicle 4: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Jul 21 06:20:31 raspberrypi evcc[3659]: [lp-1  ] INFO 2023/07/21 06:20:31 loadpoint 1:
Jul 21 06:20:31 raspberrypi evcc[3659]: [lp-1  ] INFO 2023/07/21 06:20:31   mode:        now
Jul 21 06:20:31 raspberrypi evcc[3659]: [lp-1  ] INFO 2023/07/21 06:20:31   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✓
Jul 21 06:20:31 raspberrypi evcc[3659]: [lp-1  ] INFO 2023/07/21 06:20:31   meters:      charge ✓
Jul 21 06:20:31 raspberrypi evcc[3659]: [lp-1  ] INFO 2023/07/21 06:20:31     charge:    power ✓ energy ✗ currents ✓
Jul 21 06:20:32 raspberrypi evcc[3659]: [main  ] INFO 2023/07/21 06:20:32 new version available: 0.118.9
Jul 21 06:20:32 raspberrypi evcc[3659]: [lp-1  ] INFO 2023/07/21 06:20:32 car connected
Jul 21 06:20:32 raspberrypi evcc[3659]: [lp-1  ] INFO 2023/07/21 06:20:32 start charging ->
Jul 21 06:20:33 raspberrypi evcc[3659]: [main  ] WARN 2023/07/21 06:20:33 couldn't download release notes: POST https://api.github.com/markdown: 403 API rate limit exceeded for 84.136.138.151. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.) [rate reset in 16m29s]
```

### What type of operating system are you running?

Linux

StefanSchoof, 2023-07-21:
0.114.1 is not the current version. Update to 0.118.9 and try again.

mxhf, 2023-07-21:
Indeed, usererror. Apologies. Can be closed.

closed 2023-07-21 by mxhf