evcc-io/evcc #9047: Easee: outdated ProductUpdate overwrites current charger state

GrimmiMeloni, 2023-07-21:
### Describe the bug

I have seen proof in logs, that Easee sends Product Updates via SignalR out of order, i.e. older (outdated) ProductUpdates are received by evcc last. This leads to the internal representation of the charger state in evcc become out of sync with reality.

In my specific case, the charger Op Mode was overwritten. The last received state was "waiting for vehicle", although the charge had already started. This lead to evcc thinking that the charger is still waiting for the vehicle, while in fact charging had already started.
Charger Op mode is specifically bad to be out of sync, because it rarely changes (once charging started), so evcc does not really have a chance to "self-heal" and become in sync again.

### Steps to reproduce

Not easily reproducible, but I have seen this a few times. The fact that we now are logging the timestamps of the ProductUpdates helps detect this. In the log excerpt below the message containing CHARGER_OP_MODE 6 is arriving later than CHARGER_OP_MODE 3, however looking at the timestamps in those ProductUpdates, it is clear that CHARGER_OP_MODE 6 was processed one second earlier.

### Configuration details

```yaml
N/A
```

### Log details

```text
[easee ] TRACE 2023/07/21 13:16:08 ProductUpdate EHWHL6VE: (2023-07-21 11:16:05 +0000 UTC) CHARGER_OP_MODE 2
[easee ] TRACE 2023/07/21 13:16:08 ProductUpdate EHWHL6VE: (2023-07-21 11:16:05 +0000 UTC) PILOT_MODE B
...
[easee ] TRACE 2023/07/21 13:16:20 ProductUpdate EHWHL6VE: (2023-07-21 11:16:11 +0000 UTC) CHARGER_OP_MODE 3
...
[easee ] TRACE 2023/07/21 13:16:20 ProductUpdate EHWHL6VE: (2023-07-21 11:16:10 +0000 UTC) CHARGER_OP_MODE 6
...
[easee ] TRACE 2023/07/21 13:16:20 ProductUpdate EHWHL6VE: (2023-07-21 11:16:11 +0000 UTC) PILOT_MODE C

then on next intervals, loadpoint is out of sync with reality.

[lp-1  ] DEBUG 2023/07/21 13:16:21 charger status: B
[lp-1  ] DEBUG 2023/07/21 13:16:22 vehicle status: C

and then finally we end up with this

[lp-1  ] DEBUG 2023/07/21 13:16:36 charge power: 4317W
[lp-1  ] DEBUG 2023/07/21 13:16:36 charge currents: [5.89 6.02 5.86]A
[lp-1  ] DEBUG 2023/07/21 13:16:36 charger status: B
```

### What type of operating system are you running?

Docker container

### Version

0.118.9

GrimmiMeloni, 2023-07-21:
@andig please assign to me

andig, 2023-07-21:
Where does the final B come from?

GrimmiMeloni, 2023-07-21:
> Where does the final B come from?

CHARGER_OP_MODE 6.

jheinitz, 2023-07-21:
Moin! 

Ist das genau das, was ich in #9054 beschrieben habe?
Wenn ja, dann könnt ihr das als Duplicate schließen. Sorry.

Gruß Jens

closed 2023-07-22 by andig

jheinitz, 2023-07-24:
Moin Männer!

Ich habe gerade das Laden mit dem letzten Nightly begonnen und erwartet, dass ich jetzt kein `charger out of sync: expected disabled, got enabled` zu Beginn des Ladevorgangs erhalte. Ist nun aber doch wieder passiert. 

War das hier jetzt der richtige Issue oder bin ich im falschen Film??? :-)

Wenn ihr das Log braucht, dann sagt Bescheid.

Viele Grüße

Jens

GrimmiMeloni, 2023-07-24:
Hi Jens!
Falscher Film. 😁 - das Out of Sync Thema schauen wir aktuell getrennt in #8298 an

PR #9053 (2023-07-22): easee: ignore outdated observations
fix #9047