evcc-io/evcc #9048: easee schaltet von 3P->1P aber dann nicht zurück zu 3P

ministryofsillywalks, 2023-07-21:
### Describe the bug

Bei ordentlich Sonne lädt easee wie erwartet auf 3P mit ca. 8kw. (was das Dach halt hergibt)
Wolken haben die Produktion dann gedrosselt, sodass die dynamisch auf 1P geschaltet wurde.
Jetzt scheint wieder die Sonne und es ist genug Überschuss vorhanden, aber easee bleibt auf 1P
[image]

Wenn der Ladevorgang über die UI beendet wird und dann neu gestartet wird, springt es wieder zurück auf 3P.

### Steps to reproduce

1. Wolken zwingen Wechsel von 3P auf 1P
2. Genug Strom für 3P Überschuss
3. easee bleibt auf 1P

### Configuration details

```yaml
uri: 10.10.10.3:7070 # uri for ui
interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: XXX

# log settings
log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug
  easee: trace

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/evcc-io/config#meters
meters:
  - name: pv
    type: custom
    power:
      source: calc
      add:
      - source: modbus
        model: sunspec
        uri: 10.10.10.231:1502 # Port 502 (SetApp) or 1502 (LCD)
        id: 1
        value: 103:DCW
      - source: modbus
        uri: 10.10.10.231:1502 # Port 502 (SetApp) or 1502 (LCD)
        id: 1
        register:
          address: 62836 # Battery 1 Instantaneous Power
          type: holding
          decode: float32s
          
  - name: grid
    type: custom
    power:
        source: modbus
        model: sunspec
        uri: 10.10.10.231:1502 # Port 502 (SetApp) or 1502 (LCD)
        id: 1
        subdevice: 1 # Metering device
        value: 203:W
        scale: -1

  - name: battery
    type: custom
    power:
      source: modbus
      uri: 10.10.10.231:1502 # Port 502 (SetApp) or 1502 (LCD)
[... 751 lines cut ...]
[lp-1  ] DEBUG 2023/07/21 14:17:28 charge power: 0W
[lp-2  ] DEBUG 2023/07/21 14:17:28 charge power: 3243W
[lp-1  ] DEBUG 2023/07/21 14:17:29 charge currents: [0.01 0.009 0.011]A
[lp-1  ] DEBUG 2023/07/21 14:17:29 charge total import: 62.700kWh
[lp-1  ] DEBUG 2023/07/21 14:17:29 charger status: A
[lp-1  ] DEBUG 2023/07/21 14:17:38 charge power: 0W
[lp-2  ] DEBUG 2023/07/21 14:17:38 charge power: 3243W
[lp-2  ] DEBUG 2023/07/21 14:17:39 charge currents: [14.6 0.009 0.01]A
[lp-2  ] DEBUG 2023/07/21 14:17:39 detected active phases: 1p
[lp-2  ] DEBUG 2023/07/21 14:17:39 charge total import: 12.053kWh
[lp-2  ] DEBUG 2023/07/21 14:17:39 charger status: C
[lp-2  ] DEBUG 2023/07/21 14:17:39 soc estimated: 40.00% (vehicle: 40.00%)
[lp-2  ] DEBUG 2023/07/21 14:17:39 vehicle soc: 40%
[lp-2  ] DEBUG 2023/07/21 14:17:39 vehicle soc limit: 80%
[lp-2  ] DEBUG 2023/07/21 14:17:39 vehicle range: 174km
[lp-2  ] DEBUG 2023/07/21 14:17:39 pv charge current: 37.1A = 16A + 21.1A (-4862W @ 1p)
```

### What type of operating system are you running?

Docker container

### Version

0.118.9

andig, 2023-07-21:
Wo wäre der Fehler im Log?

andig, 2023-07-21:
Dein MQTT PW ist verbrannt

ministryofsillywalks, 2023-07-21:
> Dein MQTT PW ist verbrannt

mist und ich dachte ich hätte alle Dinge ersetzt :(

ministryofsillywalks, 2023-07-21:
> Wo wäre der Fehler im Log?

14:12:39 will er zurück zu 3P (glaube ich)

14:13:59 merkt er das er auf 1P ist und bleibt dann auch da

ein error gibt es nicht, aber das system springt einfach nicht zurück auf 3P

premultiply, 2023-07-21:
Das nächste mal einfach: `evcc dump --cfg`

andig, 2023-07-21:
Es wird auf 3p geschaltet:

```
[easee ] TRACE 2023/07/21 14:12:39 POST https://api.easee.com/api/chargers/EH848820/settings
[easee ] TRACE 2023/07/21 14:12:39 {"phaseMode":2}
--
[{"device":"EH848820","commandId":38,"ticks":638255383591749140}]
[easee ] TRACE 2023/07/21 14:12:39 CommandResponse EH848820: {SerialNumber:EH848820 ID:38 Timestamp:2023-07-21 12:12:39.174914 +0000 UTC DeliveredAt:2023-07-21 12:12:39.252 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638255383591749140}
[easee ] TRACE 2023/07/21 14:12:39 POST https://api.easee.com/api/chargers/EH848820/commands/pause_charging
[easee ] TRACE 2023/07/21 14:12:39 {"device":"EH848820","commandId":48,"ticks":638255383598437380}
[easee ] TRACE 2023/07/21 14:12:40 ProductUpdate EH848820: (2023-07-21 12:12:39 +0000 UTC) PHASE_MODE 2
```

Warum das Auto keine 3p konsumiert ist unklar.