evcc-io/evcc #9049: prevent Keba charging wakeup after car reached charging SOC

motdule, 2023-07-21:
### Describe the bug

Ich beobachte mit der Version 0.118.8 das wenn die Ladesitzung bei erreichen des Ziel SOC (80%) vom Fahrzeug beendet wird, die Regelung versucht nach x Minuten den Ladevorgang wieder zu starten. Die Ladung beginnt dann auch für 1-2 Sekunden und wird dann vom Fahrzeug wieder beendet, weil es in den 80% Ladestand reinläuft. Dieses an / aus wird fortgesetzt bis keine PV Leistung mehr vorhanden ist. 

Wallbox: Keba P30 X (Firmware 1.16.1) / Modbus TCP mit Phasenumschaltung

### Steps to reproduce

1. Fahrzeug läuft in Ladebegrenzung und beendet Ladevorgang
2. Fahrzeug bleibt angesteckt
3. EVCC versucht Ladevorgang automatisch nach x Minunten zu starten und wird kurz darauf beendet.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

#log: info error warn debug trace
log: debug
levels:
  cache: error

# unique installation id
plant:

interval: 10s # control cycle interval

meters:
  - type: template
    template: fronius-solarapi-v1
    usage: grid
    host: 192.168.2.x
    name: grid1
  - type: template
    template: fronius-solarapi-v1
    usage: pv
    host: 192.168.2.x
    name: pv1
  - name: VictronBattery
    type: template
    template: victron-energy
    usage: battery
    host: 192.168.2.x # IP-Adresse oder Hostname
    port: 502 # Port (Optional)

chargers:
  - name: kebax
    type: template
    template: keba-modbus
    # Modbus TCP
    modbus: tcpip
    id: 255
    host: 192.168.2.x # Hostname
    port: 502 # Port
sponsortoken: 

loadpoints:
- title: Keba
  charger: kebax
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  enable:
    threshold: 0
    delay: 3m
[... 182 lines cut ...]
Jul 21 13:35:07 raspberrypi evcc[81968]: [lp-1  ] DEBUG 2023/07/21 13:35:07 charge power: 0W
Jul 21 13:35:07 raspberrypi evcc[81968]: [site  ] DEBUG 2023/07/21 13:35:07 pv power: 9083W
Jul 21 13:35:07 raspberrypi evcc[81968]: [site  ] DEBUG 2023/07/21 13:35:07 battery soc: 96%
Jul 21 13:35:07 raspberrypi evcc[81968]: [site  ] DEBUG 2023/07/21 13:35:07 battery power: -80W
Jul 21 13:35:07 raspberrypi evcc[81968]: [site  ] DEBUG 2023/07/21 13:35:07 grid power: -8816W
Jul 21 13:35:07 raspberrypi evcc[81968]: [site  ] DEBUG 2023/07/21 13:35:07 site power: -8846W
Jul 21 13:35:07 raspberrypi evcc[81968]: [lp-1  ] DEBUG 2023/07/21 13:35:07 charge currents: [0 0 0]A
Jul 21 13:35:07 raspberrypi evcc[81968]: [lp-1  ] DEBUG 2023/07/21 13:35:07 charge total import: 473.948kWh
Jul 21 13:35:07 raspberrypi evcc[81968]: [lp-1  ] DEBUG 2023/07/21 13:35:07 charger status: B
Jul 21 13:35:07 raspberrypi evcc[81968]: [lp-1  ] DEBUG 2023/07/21 13:35:07 final charge energy: 5.21kWh
Jul 21 13:35:07 raspberrypi evcc[81968]: [lp-1  ] INFO 2023/07/21 13:35:07 stop charging <-
Jul 21 13:35:07 raspberrypi evcc[81968]: [lp-1  ] DEBUG 2023/07/21 13:35:07 wake-up timer: start
Jul 21 13:35:07 raspberrypi evcc[81968]: [lp-1  ] DEBUG 2023/07/21 13:35:07 charge total import: 473.948kWh
Jul 21 13:35:07 raspberrypi evcc[81968]: [lp-1  ] DEBUG 2023/07/21 13:35:07 pv charge current: 12.8A = 0A + 12.8A (-8846W @ 3p)
Jul 21 13:35:07 raspberrypi evcc[81968]: [lp-1  ] DEBUG 2023/07/21 13:35:07 max charge current: 12.8A
...
```

### What type of operating system are you running?

Linux

### Version

0.118.8

andig, 2023-07-21:
> die Regelung versucht nach x Minuten den Ladevorgang wieder zu starten

Woran machst Du das fest? Dann müsste im Logfile etwas von enable/disable stehen- tut es aber nicht. Sieht für mich eher aus, als wäre der target soc _nicht_ erreicht (anderenfall würde der PV Mode gar nciht mehr zum Einsatz kommen) und du bist hier im normalen PV Mode. Anscheinend ist es das Auto selbst dass hier etwas steuert.

/cc @premultiply

andig, 2023-07-23:
Keine weiteren Erkenntnisse

closed 2023-07-23 by andig