evcc-io/evcc #9050: kein Browser Dashboard ab V 0.118.6

Murphy1985, 2023-07-21:
### Describe the bug

ab der Version 0.118.6 oder höher wird das Browserdashboard nicht mehr angezeigt.

ab Version 0.118.8 wird für die Keba Wallbox ein Token benötigt, war vorher auch nicht.

Tapo P Steckdose funktioniert auch nicht mehr, ständig Log In Fail (auch nach restart)
es wurde an der Konfiguration nichts geändert.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: db4ab0f04de37b33a2ce379ec60d82895e2e8b09eadecba3d09df4024ec31b70

interval: 10s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.46  
  port: 502  
  usage: grid  
  storageunit: 1  
  modbus: tcpip  
  timeout: 15s  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.46  
  port: 502  
  usage: pv  
  storageunit: 1  
  modbus: tcpip  
  timeout: 15s  
  name: pv2
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.46  
  port: 502  
  usage: battery  
  storageunit: 1  
  modbus: tcpip  
  timeout: 15s  
  name: battery3

chargers:
- type: template
  template: keba 
  host: 192.168.178.47  
  serial: 25582749  
  name: wallbox4
- type: template
  template: tapo 
  host: 192.168.178.39  
  user: zitzl.thomas@web.de  
  password:XXXXX  
  standbypower: -1  
  integrateddevice: true  
  icon: waterheater  
  name: wallbox5

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv
  phases: 3
  mincurrent: 7
  maxcurrent: 16
  resetOnDisconnect: true
- title: Heizung
  charger: wallbox5
  mode: pv
  phases: 1
  mincurrent: 13
  maxcurrent: 13
  resetOnDisconnect: true

site:
  title: Zitzl Home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
evcc --log debug
[main  ] INFO 2023/07/21 15:33:59 evcc 0.118.5
[main  ] INFO 2023/07/21 15:33:59 using config file: /etc/evcc.yaml
[main  ] INFO 2023/07/21 15:33:59 starting ui and api at :7070
[db    ] INFO 2023/07/21 15:33:59 using sqlite database: /home/zitzl/.evcc/evcc.db
[site  ] DEBUG 2023/07/21 15:34:00 static grid tariff
[site  ] WARN 2023/07/21 15:34:00 battery configured but residualPower is missing (add residualPower: 100 to site)
[site  ] INFO 2023/07/21 15:34:00 site config:
[site  ] INFO 2023/07/21 15:34:00   meters:      grid â pv â battery â
[site  ] INFO 2023/07/21 15:34:00     grid:      power â energy â currents â
[site  ] INFO 2023/07/21 15:34:00     pv 1:      power â energy â currents â
[site  ] INFO 2023/07/21 15:34:00     battery 1: power â energy â currents â soc â capacity â
[lp-1  ] INFO 2023/07/21 15:34:00 loadpoint 1:
[lp-1  ] INFO 2023/07/21 15:34:00   mode:        pv
[lp-1  ] INFO 2023/07/21 15:34:00   charger:     power â energy â currents â phases â wakeup â
[lp-1  ] INFO 2023/07/21 15:34:00   meters:      charge â
[lp-1  ] INFO 2023/07/21 15:34:00     charge:    power â energy â currents â
[lp-2  ] INFO 2023/07/21 15:34:00 loadpoint 2:
[lp-2  ] INFO 2023/07/21 15:34:00   mode:        pv
[lp-2  ] INFO 2023/07/21 15:34:00   charger:     power â energy â currents â phases â wakeup â
[lp-2  ] INFO 2023/07/21 15:34:00   meters:      charge â
[lp-2  ] INFO 2023/07/21 15:34:00     charge:    power â energy â currents â
[site  ] DEBUG 2023/07/21 15:34:00 static grid tariff
[lp-1  ] DEBUG 2023/07/21 15:34:00 phase timer inactive
[lp-1  ] DEBUG 2023/07/21 15:34:00 pv timer inactive
[lp-1  ] DEBUG 2023/07/21 15:34:00 guard timer inactive
[lp-2  ] DEBUG 2023/07/21 15:34:00 phase timer inactive
[lp-2  ] DEBUG 2023/07/21 15:34:00 pv timer inactive
[lp-2  ] DEBUG 2023/07/21 15:34:00 guard timer inactive
[lp-2  ] ERROR 2023/07/21 15:34:01 charger: tapo error 9999: Login failed, invalid user or password
[site  ] DEBUG 2023/07/21 15:34:01 ----
[lp-1  ] DEBUG 2023/07/21 15:34:01 charge power: 0W
[main  ] INFO 2023/07/21 15:34:02 new version available: 0.118.9
```

### What type of operating system are you running?

Linux

### Version

0.118.5

andig, 2023-07-21:
Ohne trace log nicht bewertbar

closed 2023-07-21 by andig

Murphy1985, 2023-07-21:
Was ist ein trace log?
Welchen Befehl muss ich da ausführen?
Danke

thierolm, 2023-07-21:
Siehe https://docs.evcc.io/docs/installation/linux#testen

`-l debug` einfach durch `-l trace` ersetzen ...

thierolm, 2023-07-21:
Meine Tapo Dose hat keine Probleme:

```
thierolm@LAPTOP-KQMMVN3H:evcc -l debug charger
[main  ] INFO 2023/07/21 19:14:04 evcc 0.118.7 (7be43c8c)
[main  ] INFO 2023/07/21 19:14:04 using config file: ./evcc_tapotest.yaml
[db    ] INFO 2023/07/21 19:14:04 using sqlite database: /home/thierolm/.evcc/evcc.db
Power:         0W
Charge status: B
Enabled:       false
Charged:       0.0kWh
Features:      []
```

Murphy1985, 2023-07-21:
Sobald ich das einsetze gehts im Debug charger nicht mehr weiter.
Das hat vorher alles einwandfrei funktioniert.
chargers:
- type: template
  template: tapo 
  host: 192.168.178.39  
  user: zitzl.thomas@web.de  
  password: xxx  
  standbypower: -1  
  integrateddevice: true  
  icon: waterheater  
  name: wallbox5

loadpoints:
- title: Heizung
  charger: wallbox5
  mode: pv
  phases: 1
  mincurrent: 13
  maxcurrent: 13
  resetOnDisconnect: true

zitzl@zitzl:~ $ evcc -l debug charger
[main  ] INFO 2023/07/21 21:25:44 evcc 0.118.7
[main  ] INFO 2023/07/21 21:25:44 using config file: /etc/evcc.yaml
[db    ] INFO 2023/07/21 21:25:44 using sqlite database: /home/zitzl/.evcc/evcc.db
Ab hier hängt es

Trace Log:
zitzl@zitzl:~ $ evcc -l trace charger
[main  ] INFO 2023/07/21 21:28:25 evcc 0.118.7
[main  ] INFO 2023/07/21 21:28:25 using config file: /etc/evcc.yaml
[db    ] INFO 2023/07/21 21:28:25 using sqlite database: /home/zitzl/.evcc/evcc.db
[db    ] TRACE 2023/07/21 21:28:25 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/07/21 21:28:25 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/07/21 21:28:25 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/07/21 21:28:25 SELECT * FROM `settings` 6 <nil>
[tapo  ] TRACE 2023/07/21 21:28:27 request: {"method":"login_device","params":{"password":"VG9tMjQyODUh","username":"M2RjM2MxNzgzZmVlMGUwYzg2OWIwMzQ5M2JjMmFiZTE1OGY2YmNlZQ=="}}
[tapo  ] TRACE 2023/07/21 21:28:27 POST http://192.168.178.39/app
[tapo  ] TRACE 2023/07/21 21:28:27 {"method":"securePassthrough","params":{"request":"kIxXxrnJtBOnbislmE/ixgg1Ft08P+fXcVcOhB1kliN2olHRIWPN2bsrxxlCp9EBqy//ST/cf1awjem8nbMeT8FZ4cc1jqr68MvzEN0jubbJ8f+xn6ckI2F2yVjoveIjmNCFtJcvyIFijQQSg5uSaGogMLcmVNz7vD6NqZhZ73Lal+5zAOSpqK22Ytca9IuP"}}
--
{"error_code":0,"result":{"response":"YM4bCbK6LjuPc1L5KbWRAm8Qr00gfGp6JJtqEaUhzGg="}}
[tapo  ] TRACE 2023/07/21 21:28:27 decrypted result: {"error_code":-1501}
[tapo  ] TRACE 2023/07/21 21:28:27 request: {"method":"get_device_info","requestTimeMils":2045559672}
[tapo  ] TRACE 2023/07/21 21:28:27 POST http://192.168.178.39/app?token=
[tapo  ] TRACE 2023/07/21 21:28:27 {"method":"securePassthrough","params":{"request":"pX9xzJY10CvpZ64tt81lgjBipkqbvh295YCDvpErcqx0TWMayfzIXr7NstSUjrmqZofGQH2mEotQ4zyXOl9Pyw=="}}
--
{"error_code":9999}
[tapo  ] TRACE 2023/07/21 21:28:27 request: {"method":"login_device","params":{"password":"VG9tMjQyODUh","username":"M2RjM2MxNzgzZmVlMGUwYzg2OWIwMzQ5M2JjMmFiZTE1OGY2YmNlZQ=="}}
[tapo  ] TRACE 2023/07/21 21:28:27 POST http://192.168.178.39/app
[tapo  ] TRACE 2023/07/21 21:28:27 {"method":"securePassthrough","params":{"request":"MSsrBle8Z3I3iLOJnQMBLl2kzdbt2jKI31HkIzUfU1mMSNv0DRrCWTIoeObeDLg+IdlzkL8CyX8A8Ba8Ax71fyUPezgUm2GJoD4q41DUy5e0UcPh4+nAfQo0/yEijxdqImMFWT7OhhgzKiIDgAk9y36qXbVqm/OcTzFVtpMCruDyIiHBxOye0LNZrph0hhBr"}}

Kann es auch an der Firmwareaktualisierung von der Steckdose liegen???

Danke
Thomas

thierolm, 2023-07-21:
Meine Tapo-Dose hat die neuste Firmware: 1.2.3 Build 230425 Rel.142542 und Hardware Version 1.0
Aber kann es sein dass du die neue 2-Faktor Authentifizierung aktiviert hast. Ich hab die Android App gerade aktualisiert und die bietet das an. Wenn du 2FA aktiviert hast, deaktiviere das mal und versuche es erneut.

thierolm, 2023-07-21:
Siehe auch https://www.tp-link.com/de/support/faq/3395/
...
 How to turn off Two-Step Authentication?

From ‘Me’ page, tap the Account Icon in the top left > Login Security> Then tap the Two-Step Verification ON/OFF toggle
...

thierolm, 2023-07-21:
Ich habe noch etwas gefunden:
https://github.com/fishbigger/TapoP100/issues/27

Hast du evtl einen Umlaut oder Sonderzeichen in deinem Passwort? Kannst du es mal ändern, so dass es mehr als 8 Zeichen lang ist und keine Sonderzeichen und oder Umlaute enthält?

thierolm, 2023-07-21:
Hab' mal einen Improvement PR aufgemacht (#9050), der die Fehlermeldung beim Login-Fehlercode -1501, den du ja auch in deinem Trace-Log hast, dann auch entsprechend erkennt und auswirft.

thierolm, 2023-07-21:
Letzte Frage nur zu Sicherheit: 
Du hast die Steckdose in der Tapo App auch registriert und kannst sie mit der App schalten, richtig?
Nur dann kannst du sie auch mit evcc und dem App User und Passwort nutzen ....

Murphy1985, 2023-07-22:
danke für die vielen Tipps
FW ist exakt die gleiche wie bei dir
im Passwort war ein Sonderzeichen -> funktioniert aber mit dem einfachen auch nicht.
2 Faktor ist aus
ich kann die Steckdose über die APP starten
Ping funktioniert auch

beim ausführen von config advanced springt er beim Test der Steckdose nicht in den nächsten Schritt
und wenn ich die Konfig manuell rein kopier, dann startet auch das Dashboard nicht mehr.
Die Website ist nicht erreichbar
localhost hat die Verbindung abgelehnt.
Versuche Folgendes:

Verbindung prüfen
[Proxy und Firewall prüfen](chrome-error://chromewebdata/#buttons)

ERR_CONNECTION_REFUSED

Murphy1985, 2023-07-22:
Jetzt gehts wieder
hab die Steckdose komplett aus meinen Account geschmissen und wieder neu hinzugefügt. Eine andere auch vorhandene ging problemlos
Gleiches Passwort wie vorher
in der advanced config springt der Schritt nicht weiter weil er im Log in Fail hängen bleibt und immer wieder versucht.
Ist anscheinend bei der neuen Verison 0.118.9 so, bei älteren kommt sofort Log In Fail

Danke für deine Unterstützung

Viele Grüße
Thomas

thierolm, 2023-07-22:
Ok, kurze Frage, funktioniert sie jetzt mit evcc oder nicht?

Murphy1985, 2023-07-22:
Ja einwandfrei mit der neuen Version 0.118.9

genauso wie immer.
Dann muss an Tapo was nicht gepasst haben