evcc-io/evcc #9054: evcc meldet nach Phasenwechsel 3p1p charger status: B obwohl geladen wird

jheinitz, 2023-07-21:
### Describe the bug

Ich nutze evcc mit einer Easee Wallbox zum Laden eines Skoda Enyaq. Mittlerweile eigentlich ganz zuverlässig. Nun ist mir aufgefallen, dass manchmal (leider nicht immer) evcc den Status nicht richtig mitbekommt. Nach dem Phasenwechsel von 3p auf 1p wird in angezeigt, dass 4,3 kW anliegen. Die Easee App zeigt, dass mit 4,3 kW geladen wird. Eigentlich sollte nach dem Phasenwechsel nur noch 1p aktiv sein und somit maximal 3,8 kW möglich sein. Im Log sehe ich jetzt immer nur `charger status: B` was ja bedeutet, dass nicht geladen wird. Ich verstehe das nicht so ganz. Das Verhalten habe ich schon 2-3x beobachtet, aber jetzt habe ich ein Log bzw. Trace.

### Steps to reproduce

1. Fahrzeug anschließen
2. PV laden startet mit 3p
3. PV Überschuss sinkt, so dass Phasenwechsel initiert wird
4. Phasenwechsel von 3p auf 1p
5. evcc stoppt Laden (stop charging)
6. dadurch charger status: B
7. Laden wird fortgesetzt
8. evcc meldet immer noch charger status: B
9. beim nächsten Phasenwechsel fängt sich das ganze dann meist wieder

### Configuration details

```yaml
---
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  site: trace
  # lp-1: debug
  # messaging: debug
  easee: trace
  # enyaq: debug
  # cache: error
  cache: debug

# unique installation id
plant: xxxxxx

interval: 20s # control cycle interval

sponsortoken: <censored>

# database configuration for persisting charge sessions and settings
database:
  type: sqlite
  dsn: /db/evcc.db

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

mqtt:
  broker: server2.example.net:1883
  topic: evcc
  clientid: evcc
  user: evcc
  password: <censored>

meters:
  - name: grid1
    type: custom
    power:
      source: mqtt
      topic: modbus/power/sun2000/meter
      jq: .fields.meter_active_power
      scale: -1
    energy:
      source: mqtt
      topic: modbus/power/sun2000/meter
      jq: .fields.meter_imported_energy
    currents:
      - source: mqtt
        topic: modbus/power/sun2000/meter
        jq: .fields.meter_a_line_current
        scale: -1
      - source: mqtt
        topic: modbus/power/sun2000/meter
        jq: .fields.meter_b_line_current
        scale: -1
      - source: mqtt
        topic: modbus/power/sun2000/meter
        jq: .fields.meter_c_line_current
        scale: -1

  - name: inverter1
    type: custom
    power:
      source: mqtt
      topic: modbus/power/sun2000/inverter/1
      jq: .fields.inverter_input_power
    energy:
      source: mqtt
      topic: modbus/power/sun2000/inverter/1
      jq: .fields.inverter_accumulated_energy_yield

  - name: storage1
    type: custom
    power:
      source: mqtt
      topic: modbus/power/sun2000/storage/1
      jq: .fields.storage_charge_discharge_power
      scale: -1
    soc:
      source: mqtt
      topic: modbus/power/sun2000/storage/1
      jq: .fields.storage_state_of_capacity
    capacity: 10

chargers:
  - type: template
    template: easee
    user: user@example.net
    password: <censored>
    timeout: 50s
    charger: EHxxxxxx
    name: wallbox4

vehicles:
  - type: template
    template: enyaq
    title: Skoda Enyaq
    user: user@exmple.net
    password: <censored>
    vin: TMB************
    # cache: 15m
    capacity: 82
    timeout: 50s
    name: ev1
    targetSoc: 85

loadpoints:
  - title: Carport
    charger: wallbox4
    mode: pv
    phases: 0
    minCurrent: 6
    maxCurrent: 16
    resetOnDisconnect: true
    enable:
      delay: 1m
    disable:
      delay: 5m

site:
  title: Hausweide-3
  meters:
    grid: grid1
    pv:
      - inverter1
    battery:
      - storage1
  # residualPower: 250
  residualPower: 10
  bufferSoc: 70
  bufferStartSoc: 80
  prioritySoc: 50

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.4422 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh

  # planner:
  co2:
    type: grünstromindex
    zip: "25xxx"

influx:
  url: http://server2.example.net:8086
  database: evcc
  token: evcc-pv-logging
  org: home
...
```
```

### Log details

```text
Log will be attached after bug creation.
```

### What type of operating system are you running?

Linux

### Version

0.118.9

jheinitz, 2023-07-21:
Please find logs and traces attached.

[evcc_ladebereit.txt](https://github.com/evcc-io/evcc/files/12129920/evcc_ladebereit.txt)

/cc @andig @GrimmiMeloni habt ihr das schon gesehen?

GrimmiMeloni, 2023-07-21:
Sieht in der Tat nach einem Duplicate aus.

andig, 2023-07-21:
Denke ich auch- probieren wir das erstmal

closed 2023-07-21 by andig