evcc-io/evcc #9059: Einheiten werden nicht einheitlich dargestellt

LiamHD, 2023-07-22:
### Describe the bug

Bei fast allen Leistungsangaben wird je nach Größe der Zahl W oder kW angezeigt. Bei Netzbezug und bei Ladung steht aber immer 0W. Wäre schön wenn das einheitlich dargestellt würde.
Siehe diese Beispiel:
[image]

### Steps to reproduce

Netzbezug oder Ladung müssen 0 sein.

### Configuration details

```yaml
Kann ich gerne nachreichen.
```

### Log details

```text
Kann ich gerne nachreichen.
```

### What type of operating system are you running?

Docker container

### Version

0.118.8

andig, 2023-07-22:
Jeder Wert wird für sich formatiert. Die Darstellung hängt am Betrag.

naltatis, 2023-08-01:
Wir haben uns damals dazu entscheiden alle Werte in dem Bereich in der gleichen Einheit darzustellen. Also alles kW oder alles W. Das hier ist die aktuelle Logik:

https://github.com/evcc-io/evcc/blob/72befffd9dbcf86de5f15ce1e62295d7bf7fe805/assets/js/components/Energyflow/Energyflow.vue#L244-L245

Ich find das auch weiterhin sinnvoll hier nicht die Einheiten zu mischen. Die Tatsache, dass bei 0-Werten nur W angezeigt wird würde ich auch als einen Bug sehen.

LiamHD, 2023-08-01:
680155

> ie Tatsache, dass bei 0-Werten nur W angezeigt wird würde ich auch als einen Bug sehen.

Und genau darum geht's mir in diesem Issue. Warum wurde es geschlossen?

naltatis, 2023-08-01:
Jo, da ging was durcheinander. Ich hab ihn wieder auf gemacht.

PR #9206 (2023-08-01): Energyflow: kW/W switch for non-changing values
- switch from W to kW if import, self or export power is above 1kW (before it was based on the sum of them)
  addresses #9112 
- force-update animated power values if unit changes
  fixes #9059
- add unit tests for `fmtKw`
closed 2023-08-01 by naltatis