evcc-io/evcc #9062: Citroen eC4 wird als Offline angezeigt

mailkluge, 2023-07-22:
### Describe the bug

Unser Citroen eC4 wird in evcc als Offline dargestellt

### Steps to reproduce

1. evvc öffnen
2. unter Fahrezeuge den eC4 auswählen
3. dann steht der eC4 mit dem konfigurierten Namen und hier diesem in Klammern (Offline)

### Configuration details

```yaml
die werde ich aus Datenschutzgründen nicht posten. Die Konfiguration des eC4 hat sich seit der Version 0.117.4 - mit dieser funktioniert es - nicht verändert

Anmerkung: Unten im Log Output habe ich unsere ClientID aus der URL gelöscht. Wenn man die URL mit Client_ID aufruft wird ein 401 geworfen.

>>>>
401 - Unauthorized!

Cannot pass the security checks that are required by the target API or operation, Enable debug headers for more details.
<<<<
```

### Log details

```text
[main  ] INFO 2023/07/22 11:27:58 evcc 0.118.10
[main  ] INFO 2023/07/22 11:27:58 using config file: /etc/evcc.yaml
[main  ] INFO 2023/07/22 11:27:58 starting ui and api at :7070
[db    ] INFO 2023/07/22 11:27:58 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/07/22 11:27:58 connecting evcc-1827471131 at tcp://localhost:1883
[mqtt  ] DEBUG 2023/07/22 11:27:58 tcp://localhost:1883 connected
[eebus ] INFO 2023/07/22 11:27:58 Local SKI:  xxxxxxxxxxxxxxxxxx
[eebus ] DEBUG 2023/07/22 11:27:58 mdns: announce
[eebus ] DEBUG 2023/07/22 11:27:58 mdns: using avahi
[eebus ] DEBUG 2023/07/22 11:27:58 starting websocket server on :4712
[eebus ] DEBUG 2023/07/22 11:27:58 error during websocket server starting: listen tcp :4712: bind: address already in use
[eebus ] DEBUG 2023/07/22 11:27:58 mdns: start search
[eebus ] DEBUG 2023/07/22 11:27:58 ski: xxxxxxx name: Elli-Wallbox-2127A1H3UU brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2127A1H3UU register: false host: wallbox-2127A1H3UU.local port: 4712 addresses: [192.168.178.26]
[eebus ] DEBUG 2023/07/22 11:27:58 delaying connection to xxxxx by 2.485s to minimize double connection probability
[eebus ] DEBUG 2023/07/22 11:28:01 trying to connect to xxxxx at 192.168.178.26
[eebus ] DEBUG 2023/07/22 11:28:01 initiating connection to xxxxx at 192.168.178.26:4712
[eebus ] DEBUG 2023/07/22 11:28:02 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2023/07/22 11:28:02 Recv: d:_i:47859_Elli-Wallbox-2127A1H3UU: to NodeManagement read 507287 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2023/07/22 11:28:02 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU reply 2 507287 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2023/07/22 11:28:02 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU result 3 507287 ResultData 0
[eebus ] DEBUG 2023/07/22 11:28:03 Recv: d:_i:47859_Elli-Wallbox-2127A1H3UU: to NodeManagement reply 507289 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2023/07/22 11:28:03 ev connected
[eebus ] DEBUG 2023/07/22 11:28:03 Send:  call 4 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU call 5 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU read 6 DeviceClassificationManufacturerData
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU read 7 NodeManagementUseCaseData
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU call 8 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU read 9 DeviceDiagnosisStateData
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU call 10 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU call 11 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU call 12 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU call 13 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU call 14 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU call 15 NodeManagementBindingRequestCall
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU read 16 DeviceConfigurationKeyValueDescriptionListData
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU read 17 DeviceClassificationManufacturerData
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU read 18 DeviceDiagnosisStateData
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU read 19 ElectricalConnectionDescriptionListData
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU read 20 ElectricalConnectionParameterDescriptionListData
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU read 21 MeasurementDescriptionListData
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU read 22 LoadControlLimitDescriptionListData
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU read 23 IdentificationListData
[eebus ] DEBUG 2023/07/22 11:28:03 Recv: d:_i:47859_Elli-Wallbox-2127A1H3UU: to NodeManagement call 507291 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2023/07/22 11:28:03 Send: d:_i:47859_Elli-Wallbox-2127A1H3UU result 24 507291 ResultData 0
[eebus ] DEBUG 2023/07/22 11:28:03 Recv: d:_i:47859_Elli-Wallbox-2127A1H3UU: to NodeManagement result 507297 4 ResultData 0
[eebus ] DEBUG 2023/07/22 11:28:03 Recv: d:_i:47859_Elli-Wallbox-2127A1H3UU: to NodeManagement result 507301 5 ResultData 0
[eebus ] DEBUG 2023/07/22 11:28:03 Recv: d:_i:47859_Elli-Wallbox-2127A1H3UU: to DeviceClassification reply 507304 6 DeviceClassificationManufacturerData
[eebus ] DEBUG 2023/07/22 11:28:03 Recv: d:_i:47859_Elli-Wallbox-2127A1H3UU: to DeviceDiagnosis reply 507310 9 DeviceDiagnosisStateData
[eebus ] DEBUG 2023/07/22 11:28:03 xxxxxx websocket read error:  websocket: close 4201: Double connection
[eebus ] DEBUG 2023/07/22 11:28:03 restarting mdns search
[eebus ] DEBUG 2023/07/22 11:28:03 ev disconnected
[eebus ] DEBUG 2023/07/22 11:28:03 delaying connection to xxxx by 2.694s to minimize double connection probability
[eebus ] DEBUG 2023/07/22 11:28:06 trying to connect to xxxxxx at 192.168.178.26
[eebus ] DEBUG 2023/07/22 11:28:06 initiating connection to xxxxx at 192.168.178.26:4712
[eebus ] DEBUG 2023/07/22 11:28:06 connection to xxxxxx failed:  dial tcp 192.168.178.26:4712: connect: connection refused
[eebus ] DEBUG 2023/07/22 11:28:06 trying to connect to xxxxx at wallbox-2127A1H3UU.local
[eebus ] DEBUG 2023/07/22 11:28:06 initiating connection to xxxxxxx at wallbox-2127A1H3UU.local:4712
[eebus ] DEBUG 2023/07/22 11:28:06 connection to xxxxxx failed: dial tcp: lookup wallbox-2127A1H3UU.local on [fd00::3ea6:2fff:fe2f:c596]:53: no such host
[eebus ] DEBUG 2023/07/22 11:28:06 restarting mdns search
[eebus ] DEBUG 2023/07/22 11:28:06 delaying connection to xxxxxx by 8.227s to minimize double connection probability
[eebus ] DEBUG 2023/07/22 11:28:10 avahi - error resolving service: Timeout reached
[main  ] ERROR 2023/07/22 11:28:13 creating vehicle ev2 failed: cannot create vehicle 'template': cannot create vehicle 'citroen': cannot get vehicles: Get "https://api.groupe-psa.com/connectedcar/v4/user/vehicles?client_id=xxxx(client ID habe ich manuell gelöscht)": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[site  ] WARN 2023/07/22 11:28:13 vehicle 'Denise eC4 (offline)' does not support automatic detection
[site  ] INFO 2023/07/22 11:28:13 site config:
[site  ] INFO 2023/07/22 11:28:13   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/07/22 11:28:13     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/07/22 11:28:13     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/07/22 11:28:13     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/07/22 11:28:13   vehicles:
[site  ] INFO 2023/07/22 11:28:13     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[site  ] INFO 2023/07/22 11:28:13     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/07/22 11:28:13 loadpoint 1:
[lp-1  ] INFO 2023/07/22 11:28:13   mode:        pv
[lp-1  ] INFO 2023/07/22 11:28:13   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/07/22 11:28:13   meters:      charge ✓
[lp-1  ] INFO 2023/07/22 11:28:13     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/07/22 11:28:13 phase timer inactive
[lp-1  ] DEBUG 2023/07/22 11:28:13 pv timer inactive
[lp-1  ] DEBUG 2023/07/22 11:28:13 guard timer inactive
```

### What type of operating system are you running?

Linux

### Version

0.118.10

mailkluge, 2023-07-22:
mhm...nach erneutem downgrade auf Version 0.117.4 funktioniert es weiter nicht. Dann scheint es nicht an evcc zu liegen. Stelle das Ticket zurück und beobachten erstmal weiter

mailkluge, 2023-07-22:
@andig Ich konnte den "Gut Fall" nicht reproduzieren. Daher habe ich das Ticket geschlossen. Sehe gerade, dass du es auch geschlossen hast - warum? Fehlte erneut eine wichtige Information? Lass es mich wissen

andig, 2023-07-22:
Wie schonmal geschrieben: es braucht ein trace log (nicht Debug). Und das muss natürlich den Fehler zeigen, nicht irgendwas.

mailkluge, 2023-07-22:
@andig Feedback: Das ist echt schwierig aus der Perspektive von jemanden, der ein Problem mit der Software hat. Das du ein TRACE brauchst hast du nicht geschrieben. Schau gerne in die Email. Da steht "du brauchst ein Log" und in dem Issue wird auch nur der DEBUG gefordert. Wie wird der Trace den du brauchst erstellt? Ich denke zusätzlich dass es kein guter Stil ist das Ticket stumpf zu schließen als "completed". Das ist total misleading, weil der Requester davon ausgeht, dass es gelöst ist. Dass Infos fehlen, kommt nicht raus. 

Und unsere Konversation zeigt auch, dass das nicht hilft und den Prozess besonders sowohl für dich als auch mich verlängert. Kürzer wäre, wenn du einfach schreibst: "Infos fehlen. Brauche ein Trace. bitte zusätzlich evcc -- ????" eingeben und Ergebnis posten."

andig, 2023-07-22:
Für die Diskussion fehlt mir die Zeit: https://github.com/evcc-io/evcc/issues/8995#issuecomment-1646531087

>  Hier brauchts ein trace log.

DerAndereAndi, 2023-07-22:
```
[eebus ] DEBUG 2023/07/22 11:27:58 error during websocket server starting: listen tcp :4712: bind: address already in use
```

evcc läuft scheinbar bereits. Das geht nicht. Und die Elli Wallbox beendet die Verbindung weil es bereits eine Verbindung zu einer anderen evcc Instanz hat.

Lösung: Bitte sicherstellen dass nur 1 Instanz von evcc läuft.

closed 2023-07-22 by mailkluge