evcc-io/evcc #9071: OCPP: nach neustart von EVCC keine neue Verbidnung

langschmidt, 2023-07-22:
### Describe the bug

Hallo zusammen,
ich hoffe es ist wirklich ein bug der bei EVCC liegt und nicht im OCPP Protokoll.

Ich habe 4 ABB Wallboxen, die alle via W-Lan oder Lan und OCPP verbunden sind.

Sobald ich EVCC neustarte, melden sich die Wallboxen nicht mehr an, im Wireshark kann ich aber sehen, dass die Wallboxen sich melden. Für mich sieht es so aus, dass die Wallboxen nachdem Neustart denken EVCC wäre immer noch da und keine neue Anmeldung generieren. 
Ich kann das umgehen, in dem ich einfach 3-5 min warte und EVCC erst dann neustarte, dann geht alles wieder.

Danke und Grüße
Johannes

### Steps to reproduce

1. EVCC stopppen
2. EVCC sofort wieder starten.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 678512825956f4a2cff276505fb865fc7f5faa6b1a69737fcb254538f921f9a2

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3ODI5MDAzNDYsImlhdCI6MTY4ODI5MjM0NiwiaXNzIjoiZXZjYy5pbyIsInN1YiI6ImxhbmdzY2htaWR0In0.5mgAiqth7-JFx7YjAHR_R1Q_otAII5BYgqjy5fRhtMY

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.81.8
  port: 502
  usage: pv
  storageunit: 0
  modbus: tcpip
  timeout: 30s
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.81.8
  port: 502
  usage: grid
  storageunit: 0
  modbus: tcpip
  timeout: 30s
  name: grid4

chargers:
- type: template 
  template: ocpp
  idtag: 1  
  stationid: TACW1144521G4608
  name: wallbox1
  getconfiguration: false
- type: template 
  template: ocpp
  idtag: 1  
  stationid: TACW1144521G5150
  name: wallbox3
  getconfiguration: false

loadpoints:
- title: Grüggelgraben-Garage1
  charger: wallbox1
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  circuit: Grüggelgraben
- title: Grüggelgraben-Garage3
  charger: wallbox3
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  circuit: Grüggelgraben

circuits:
- name: Grüggelgraben
  maxCurrent: 75
  maxPower: 0
  meter: 
  parent: 

site:
- title: Grüggelgraben
  meters:
    grid: grid4
    pv:
    - pv2
    battery:
```

### Log details

```text
C:\evcc>evcc --log debug
[main  ] INFO 2023/07/22 20:39:29 evcc 0.118.8-loadmanagement-next
[main  ] INFO 2023/07/22 20:39:29 using config file: C:\evcc\evcc.yaml
[main  ] INFO 2023/07/22 20:39:29 starting ui and api at :7070
[db    ] INFO 2023/07/22 20:39:30 using sqlite database: C:\Users\Administrator.LS\.evcc\evcc.db
[TACW1144521G5150] DEBUG 2023/07/22 20:39:31 waiting for chargepoint: 5m0s
[TACW1144521G4608] DEBUG 2023/07/22 20:39:31 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2023/07/22 20:39:31 chargepoint connected: TACW1144521G5150
[ocpp  ] DEBUG 2023/07/22 20:39:31 chargepoint connected: TACW1144521G4608
[circuit-1] DEBUG 2023/07/22 20:39:32 power checking disabled
[site  ] INFO 2023/07/22 20:39:32 site config:
[site  ] INFO 2023/07/22 20:39:32   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/07/22 20:39:32     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/07/22 20:39:32     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/07/22 20:39:32   circuits:
[site  ] INFO 2023/07/22 20:39:32      Power: ✗, Current: max 75.0A
[lp-1  ] INFO 2023/07/22 20:39:32 loadpoint 1:
[lp-1  ] INFO 2023/07/22 20:39:32   mode:        now
[lp-1  ] INFO 2023/07/22 20:39:32   circuit:     Grüggelgraben
[lp-1  ] INFO 2023/07/22 20:39:32   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/07/22 20:39:32   meters:      charge ✗
[lp-2  ] INFO 2023/07/22 20:39:32 loadpoint 2:
[lp-2  ] INFO 2023/07/22 20:39:32   mode:        now
[lp-2  ] INFO 2023/07/22 20:39:32   circuit:     Grüggelgraben
[lp-2  ] INFO 2023/07/22 20:39:32   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2023/07/22 20:39:32   meters:      charge ✗
[lp-1  ] DEBUG 2023/07/22 20:39:32 phase timer inactive
[lp-1  ] DEBUG 2023/07/22 20:39:32 pv timer inactive
[lp-1  ] DEBUG 2023/07/22 20:39:32 guard timer inactive
[lp-2  ] DEBUG 2023/07/22 20:39:32 phase timer inactive
[lp-2  ] DEBUG 2023/07/22 20:39:32 pv timer inactive
[lp-2  ] DEBUG 2023/07/22 20:39:32 guard timer inactive
[site  ] DEBUG 2023/07/22 20:39:32 ----
[lp-1  ] DEBUG 2023/07/22 20:39:32 charge power: 0W
[lp-2  ] DEBUG 2023/07/22 20:39:32 charge power: 0W
[main  ] INFO 2023/07/22 20:39:32 new version available: 0.118.10
[site  ] DEBUG 2023/07/22 20:39:33 pv power: 32W
[site  ] DEBUG 2023/07/22 20:39:34 grid power: 6W

C:\evcc>evcc --log debug
[main  ] INFO 2023/07/22 20:39:37 evcc 0.118.8-loadmanagement-next
[main  ] INFO 2023/07/22 20:39:37 using config file: C:\evcc\evcc.yaml
[main  ] INFO 2023/07/22 20:39:37 starting ui and api at :7070
[db    ] INFO 2023/07/22 20:39:37 using sqlite database: C:\Users\Administrator.LS\.evcc\evcc.db
[TACW1144521G4608] DEBUG 2023/07/22 20:39:38 waiting for chargepoint: 5m0s
[TACW1144521G5150] DEBUG 2023/07/22 20:39:38 waiting for chargepoint: 5m0s
```

### What type of operating system are you running?

Windows

### Version

0.118-8-loadmanagement

andig, 2023-07-22:
> Für mich sieht es so aus, dass die Wallboxen nachdem Neustart denken EVCC wäre immer noch da und keine neue Anmeldung generieren.

Bin nicht sicher, aber du könntest im service File ein Sleep einbauen damit die Pause länger wird. Sonst wüsste ich nicht was man da tun könnte.

closed 2023-07-22 by andig

langschmidt, 2023-07-22:
Wenn ich evcc starte, dann kann ich warte solange ich will es geht erst wieder wenn EVCC ca. 3-5min aus war.
Dann logge ich bei Gelgenheit nochmal per wireshark mit - wer den Fehler macht.

langschmidt, 2023-07-22:
[image]
er macht einfach mit Keep Alive weiter...

langschmidt, 2023-07-22:
jetzt ist er aber doch irgendwann angegangen. Komisch komisch...
Dann machen wir es erstmal so weiter...