evcc-io/evcc #9072: Value of savings.gridCharged and savings.selfConsumptionCharged too low after forceful shutdown of evcc

droerich, 2023-07-22:
### Describe the bug

I cut the power of the device (Raspi) evcc is running. After restarting the device and thus evcc the values of  database keys `savings.gridCharged` and `savings.selfConsumptionCharged` are smaller than before the shutdown. As far as I can tell, no loading sessions in the recent past are missing in the `sessions` database though.

### Steps to reproduce

1. Kill evcc (e.g. cut power)

I don't know if this always reproduces the described behavior.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ***

interval: 10s # control cycle interval

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.178.63
  name: grid1
- type: template
  template: sma-inverter
  usage: pv
  host: 192.168.178.61
  name: pv2

vehicles:
- type: template
  template: renault
  title: Megane
  user: ***
  password: ***
  vin: ***
  capacity: 60
  phases: 3
  icon: car
  cache: 15m
  minSoC: 20
  targetSoC: 80
  minCurrent: 6
  maxCurrent: 32
  name: ev3

sponsortoken: ***

chargers:
- type: template
  template: easee
  user: ***
  password: ***
  charger: EHZP2PDQ
  name: wallbox1

loadpoints:
- title: Stellplatz
  charger: wallbox1
[... 107 lines cut ...]
Jul 22 17:17:15 evctl evcc[796]: [site  ] DEBUG 2023/07/22 17:17:15 grid currents: [2.33 -0.871 -1.14]A
Jul 22 17:17:15 evctl evcc[796]: [site  ] DEBUG 2023/07/22 17:17:15 site power: 45W
Jul 22 17:17:15 evctl evcc[796]: [lp-1  ] DEBUG 2023/07/22 17:17:15 charge currents: [0 0 0]A
Jul 22 17:17:15 evctl evcc[796]: [lp-1  ] DEBUG 2023/07/22 17:17:15 charge total import: 1375.073kWh
Jul 22 17:17:15 evctl evcc[796]: [lp-1  ] DEBUG 2023/07/22 17:17:15 charger status: A
Jul 22 17:17:25 evctl evcc[796]: [site  ] DEBUG 2023/07/22 17:17:25 ----
Jul 22 17:17:25 evctl evcc[796]: [lp-1  ] DEBUG 2023/07/22 17:17:25 charge power: 0W
Jul 22 17:17:25 evctl evcc[796]: [site  ] DEBUG 2023/07/22 17:17:25 static grid tariff
Jul 22 17:17:25 evctl evcc[796]: [site  ] DEBUG 2023/07/22 17:17:25 pv power: 938W
Jul 22 17:17:25 evctl evcc[796]: [site  ] DEBUG 2023/07/22 17:17:25 grid power: 20W
Jul 22 17:17:25 evctl evcc[796]: [site  ] DEBUG 2023/07/22 17:17:25 grid powers: [471 -185 -265]W
Jul 22 17:17:25 evctl evcc[796]: [site  ] DEBUG 2023/07/22 17:17:25 grid currents: [2.23 -0.867 -1.13]A
Jul 22 17:17:25 evctl evcc[796]: [site  ] DEBUG 2023/07/22 17:17:25 site power: 20W
Jul 22 17:17:25 evctl evcc[796]: [lp-1  ] DEBUG 2023/07/22 17:17:25 charge currents: [0 0 0]A
Jul 22 17:17:25 evctl evcc[796]: [lp-1  ] DEBUG 2023/07/22 17:17:25 charge total import: 1375.073kWh
Jul 22 17:17:25 evctl evcc[796]: [lp-1  ] DEBUG 2023/07/22 17:17:25 charger status: A
```

### What type of operating system are you running?

Linux

### Version

0.118.1

andig, 2023-07-23:
Und worin besteht das Problem?

droerich, 2023-07-23:
Das Problem besteht im Verlust der Daten über die insgesamt bezogene Energie aus dem Netzt und aus der PV-Anlage in der Größenordnung von mehreren Dutzend kW. Die in der Datenbank gespeicherten Werte von `savings.gridCharged` und `savings.selfConsumptionCharged` entsprechen einem Stand, der mutmaßlich mehrere Wochen alt ist (genau konnte ich es nicht herausfinden) und nicht den Werten, die zuletzt, also vor dem harten Abschalten, auf der UI angezeigt wurden. Ich hätte erwartet, dass diese Werte regelmäßig in die Datenbank persistiert werden. So verstehe ich im Übrigen auch [den Code](https://github.com/evcc-io/evcc/blob/0.118.1/core/savings.go#L114): `Update(..)` wird regelmäßig aufgerufen und aktualisiert sowohl die UI als auch die entsprechenden Daten in der Datenbank. Da das bei mir aber offensichtlich nicht so funktioniert hat, gehe ich von einem Bug aus.

andig, 2023-07-23:
Dann brauchts ein trace log für db um das nachvollziehen zu können.

droerich, 2023-07-23:
Kann man das Log-Level für einzelne Komponenten individuell einstellen?

andig, 2023-07-23:
Ja, siehe levels

droerich, 2023-07-23:
Ich konnte das Verhalten gerade mit 0.118.10 reproduzieren. 

Vor hartem Abschalten auf Web-UI
* 692 kWh Sonne
* 420 kWh Netz

Nach Neustart auf Web-UI
* 675 kWh Sonne
* 419 kWh Netz

Leider noch keine Tracing-Logs für die DB, denn dazu hätte ich meine Konfig ändern und evcc neu starten müssen, was das Experiment verfälscht hätte durch das reguläre Beenden von evcc.

andig, 2023-07-23:
Was ist denn "reguläres" Beenden im Vergleich? In jedem Fall brauchts mal logs von regulär und dem Fehlerfall zum Vergleich.

droerich, 2023-07-23:
Mit regulärem Beenden meine ich ein `systemctl stop evcc` sprich ein `SIGTERM` an evcc.

Ich habe jetzt trace-Logs für DB aktiviert und werde das Experiment wiederholen, sobald ein neuer Ladevorgang hinzugekommen ist.

andig, 2023-07-23:
Wir behandeln die eigentlich alle gleich:

	// catch signals
	go func() {
		signalC := make(chan os.Signal, 1)
		signal.Notify(signalC, os.Interrupt, syscall.SIGTERM)

		<-signalC                        // wait for signal
		once.Do(func() { close(stopC) }) // signal loop to end
	}()

Beendest Du es vllt. per SIGKILL?

droerich, 2023-07-23:
Der Fehler tritt auf, wenn ich das Gerät, auf dem evcc läuft, hart abschalte, indem ich die Stromversorgung trenne.

andig, 2023-07-23:
In dem Fall... mache ich hier zu- denn darauf kann keine SW der Welt reagieren.

droerich, 2023-07-23:
Das kann ich nicht nachvollziehen. Es geht hier ja nicht um Daten aus den letzten Sekunden vor dem Abschalten, sondern um Daten aus einem Zeitraum von mehreren Wochen. Daten aus solchen Zeiträumen kann man ja wohl persistieren ohne dass sie bei einem Stromausfall verlorgengehen. Klappt ja übrigens mit den Ladevorgängen in der `sessions`-Datenbank auch, die in meinem Fall erhalten blieben -- trotz Trennung der Stromversorgung.

andig, 2023-07-23:
Klick. Jetzt hab ichs verstanden.

droerich, 2023-07-25:
Ich habe den Fehler mit trace-Level-Log für DB reproduziert

- evcc lief seit 23.7.23
- Es kamen Ladevorgänge hinzu
- Kurz vor dem Abschalten zeigte die UI 922 kWh Sonne & 508 kWh Netz
- Ich habe am 25.7.23 um ca. 22:25 den Stecker von meinem Raspi gezogen, auf dem evcc läuft
- Nach Neustart zeigt die UI 893 kWh Sonne & 463 kWh Netz

Hier die Logs: 
[evcc.log.gz](https://github.com/evcc-io/evcc/files/12165417/evcc.log.gz)

droerich, 2023-07-26:
Heute habe ich evcc zum Vergleich nochmal regulär heruntergefahren (`systemctl stop evcc`)
* evcc lief seit 25.7.23 ca. 22:30 durchgängig
* Es kamen Ladevorgänge hinzu
* Beenden von evcc mit `systemctl stop evcc` am 26.7.23 ca. 22:26

Logs (Trace-Level für DB): 
[evcc-230726.log.gz](https://github.com/evcc-io/evcc/files/12176963/evcc-230726.log.gz)

andig, 2023-07-26:
Und?

droerich, 2023-07-30:
Und die letzten Logs, die ich gepostet habe, sind nutzlos, weil durch einen Fehler von mir die Datenbank für evcc nicht schreibbar war. Muss das Experiment wiederholen.

andig, 2023-07-30:
Ich muss es auch nochmal wiederholen: hartes Abschalten ist kein Case, auf den wir optimieren wollen...

droerich, 2023-07-30:
Zwischen Optimierung für einen Edge-Case und Berücksichtigung eines Edge-Cases gibt es ja auch eine gewissen Bandbreite. Plötzliches Abschalten/Stromausfälle sind eine Realität von Embedded Devices, die eine robuste Anwendung meiner Meinung nach berücksichtigen sollte.

Ich mache diese Auswertungen, weil ich euch dabei helfen will, die App zu verbessern. Wenn daran kein Interesse besteht, akzeptiere ich das natürlich. Dann behalte ich halt im Hinterkopf, dass man evcc immer sauber runterfahren muss, um solche Datenverluste zu vermeiden. Aber sagt es mir halt bitte, damit ich hier keine Arbeit für die Tonne mache.

naltatis, 2023-08-01:
@droerich danke für die Analyse. Ich bin da total bei dir. Wir sollten alle zu persistierenden Daten in regelmäßigen Abständen (bspw. alle 30min) auch wirklich auf die Platte schreiben. Stromausfall und Stecker-Ziehen beim Raspberry sind in der Praxis nicht vermeidbar.

RTTTC, 2023-08-02:
I have also lost some data, noticed that my solar energy % went down after a power outage. (It was slowly going up the last few months, as there is plenty of sun.) 
Good catch and good effort @droerich ! 😎

ignazhabibi, 2023-08-05:
Ich habe das gleiche Problem auch schon mehrfach erlebt. Schade wenn die Statistiken dann mehrere Wochen alt sind, weil man den Raspi mal vom Strom trennen musste.

Passiert übrigens auch wenn man einen Home Assistant Restore macht und da evcc Add-on verwendet. Dies aber nur am Rande.

PR #9257 (2023-08-08): Settings/Savings: persist unsaved values regularly
fixes #9072

Write unsaved values at least every 30min.