evcc-io/evcc #9097: Stromkosten für Ladepunkt als Grenzkosten anzeigen

MarkusGH, 2023-07-24:
Die Anzeige der Stromkosten im UI für Verbrauch und Ladepunkt ist aus meiner Sicht aktuell nicht optimal.

Momentan wird so gerechnet:

Kosten (Verbrauch) = Kosten (Ladepunkt) = (Netzbezug * Bezugspreis - Eigenverbrauch * Einspeisepreis) / (Verbrauch + Ladepunkt)

Besser wäre:

(Alle Werte sind Leistungen, wie sie im UI benannt sind)

EigenverbrauchV = Min(Eigenverbrauch, Verbrauch)
BezugV = Verbrauch - EigenverbrauchV
EigenverbrauchL = Eigenverbrauch - EigenverbrauchV
BezugL = Ladepunkt - EigenverbrauchL

Kosten (Verbrauch) = (BezugV * Bezugspreis - EigenverbrauchV * Einspeisepreis) / Verbrauch
Kosten (Ladepunkt) = (BezugL * Bezugspreis - EigenverbrauchL * Einspeisepreis) / Ladepunkt

Auf die Art werden die Grenzkosten angezeigt, die entstehen wenn geladen wird. 

Die Erzeugung wird zunächst dem Hausverbrauch zugeschlagen, der ja ohnehin ansteht. Damit kann man qualifizierter entscheiden, ob man gerade laden möchte.

andig, 2023-07-24:
Wie würde das mit der Ladeplanung harmonieren?

MarkusGH, 2023-07-24:
Mit der Ladeplanung in meinem Hirn - super!
Spielt der Preis im PV Modus sonst noch eine Rolle?

andig, 2023-07-24:
Für die Ladeplanung.

MarkusGH, 2023-07-25:
Für die Ladeplanung sehe ich keine Relevanz, denn der Grenzpreis wird immer zwischen PV Einspeisepreis und Strom Einkaufspreis liegen. Also ist PV Laden immer günstiger als zu einem anderen Zeitpunkt Strom aus dem Netz zu ziehen. In der CO2 Betrachtung gilt dasselbe.

MarkusGH, 2023-07-25:
Beispiel:
PHEV, ab 40 ct kWh ist Strom teurer als Benzin.
Strom 50 ct, PV 8 ct
Hausverbrauch 4 kW, Erzeugung 2kW, Ladepunkt 2 kW, Import 4 kW.
Aktuell zeigt es als Strompreis für das  Laden des Fahrzeugs dann 36 ct / kWh an (8 * 2 + 50 * 4) / 6
Also lade ich, weil es ja billiger ist.
Tatsächlich kostet das laden in der Konstellation 60 ct / kWh, da die 2 kW Erzeugung schon komplett vom Haus verbraucht werden.

MarkusGH, 2023-07-25:
Für die Ladeplanung sollte das unschädlich sein, da PV Strom in der Gestehung immer günstiger ist als Netzstrom.

kalaws, 2023-07-25:
> Für die Ladeplanung sollte das unschädlich sein, da PV Strom in der Gestehung immer günstiger ist als Netzstrom.

Generally, this is not true where subsidies exist, like in Sweden. (According to Google translate, you're saying that "PV electricity is always cheaper to generate than mains electricity" – I am assuming you do not mean "generate", but "use" or "charge")

My suggestion of a `netPlanner` seems highly relevant to your feature request, to the extent that it relates to or impacts charge planning. I've written up what seems to me (not knowing much of anything about coding!) a relatively simple addition to the existing `planner`. UI implementation is through checkbox (or a new `Export` mode). It should also be beneficial from a CO2 perspective, assuming that energy prices tend positively to correlate with CO2 intensity.
[https://github.com/evcc-io/evcc/discussions/9042#discussioncomment-6528820](https://github.com/evcc-io/evcc/discussions/9042#discussioncomment-6528820)

MarkusGH, 2023-07-25:
OK - let's remove a simplification. 
The order of prices will always be:
PV price < mixed price (current implementation in the UI) < marginal Price (my suggestion) < mains price
or
PV price > mixed price (current implementation in the UI) > marginal Price (my suggestion) > mains price.
So any decision the planner makes will be exactly the same, no matter if marginal price or mixed price will be used.

kalaws, 2023-07-25:
> So any decision the planner makes will be exactly the same, no matter if marginal price or mixed price will be used.

Well, the point of the `netPlanner` that I proposed is precisely that the decision the planner makes will NOT be exactly the same, because there are savings to be made from making a different decision, as the table comparisons in that linked comment indicate. :)

Specifically, savings (compared to current `planner`) are due whenever, instead of charging, one could export PV at a feedin price greater than half of the grid cost of an hour scheduled for charging by `planner`.