evcc-io/evcc #9100: easee: CHARGER_OP_MODE 2 is incorrectly considered an "enabled" state

GrimmiMeloni, 2023-07-24:
### Describe the bug

Easee chargers implement a state machine moving through different states called `CHARGER_OP_MODE`. (Official docs [here](https://developer.easee.cloud/docs/enumerations#op-mode-109))

The Easee specific charger API implementation of `Enabled()` considers several op_modes as enabled, one of them being `2` (AWAITING_START). 
This behavior is incorrect, as the charger moves into this state when disabled by evcc. It should therefore not be considered as one of the enabled states.
 

### Steps to reproduce

1. Start charging by switching from Mode "off" to any other mode. 
2. Observe log, seeing Product Updates for CHARGER_OP_MODE, first 6 (waiting for EV), then 3 (charging)
3. Stop charging by switching back to Mode "off"
4. Observe log, seeing Product Update for CHARGER_OP_MODE moving to 2 (suspended EVSE).

### Configuration details

```yaml
any valid config with an Easee WB will do
```

### Log details

```text
[lp-1  ] DEBUG 2023/07/24 14:49:47 set charge mode: pv
[...]
[easee ] TRACE 2023/07/24 14:49:49 POST https://api.easee.com/api/chargers/EHWHL6VE/commands/resume_charging
[easee ] TRACE 2023/07/24 14:49:50 ProductUpdate EHWHL6VE: (2023-07-24 12:49:48 +0000 UTC) DYNAMIC_CHARGER_CURRENT 32
[easee ] TRACE 2023/07/24 14:49:52 ProductUpdate EHWHL6VE: (2023-07-24 12:49:50 +0000 UTC) CHARGER_OP_MODE 6
[easee ] TRACE 2023/07/24 14:49:54 ProductUpdate EHWHL6VE: (2023-07-24 12:49:52 +0000 UTC) PILOT_MODE C
[easee ] TRACE 2023/07/24 14:49:54 ProductUpdate EHWHL6VE: (2023-07-24 12:49:52 +0000 UTC) CHARGER_OP_MODE 3
[...]
[lp-1  ] DEBUG 2023/07/24 14:50:43 set charge mode: off
[...]
[easee ] TRACE 2023/07/24 14:50:46 POST https://api.easee.com/api/chargers/EHWHL6VE/commands/pause_charging
[easee ] TRACE 2023/07/24 14:50:47 ProductUpdate EHWHL6VE: (2023-07-24 12:50:45 +0000 UTC) DYNAMIC_CHARGER_CURRENT 0
[easee ] TRACE 2023/07/24 14:50:49 ProductUpdate EHWHL6VE: (2023-07-24 12:50:47 +0000 UTC) CHARGER_OP_MODE 2
[easee ] TRACE 2023/07/24 14:50:49 ProductUpdate EHWHL6VE: (2023-07-24 12:50:47 +0000 UTC) PILOT_MODE B
[...]
```

### What type of operating system are you running?

Docker container

### Version

0.118.9

GrimmiMeloni, 2023-07-24:
This was detected in the scenario described [here](https://github.com/evcc-io/evcc/issues/8298#issuecomment-1646931257)

andig, 2023-07-24:
@GrimmiMeloni magst du dich mal bei Slack anmelden?

andig, 2023-07-24:
Also hier ändern? https://github.com/evcc-io/evcc/blob/master/charger/easee.go#L383

GrimmiMeloni, 2023-07-24:
> Also hier ändern? https://github.com/evcc-io/evcc/blob/master/charger/easee.go#L383

Jepp. AwaitingStart ist der State der raus muss.

MaxV85, 2023-07-25:
Ist das ein Problem, welches mich vom Laden abhält? Oder generell: Ist das Laden mit der Easee nun halbwegs stabil? Tut mir leid wenn ich hier frage, aber ich bin immer noch unentschlossen, ob ich meine Hassliebe verkaufen soll oder nicht? Am Montag lade ich zum ersten Mal 😃

GrimmiMeloni, 2023-07-25:
TL;DR - Getestet gegen Master von heute. Nicht gut. Neuer Versuch morgen in Kombination mit #9075. Alternativ synchrones Warten auf CHARGER_OP_MODE updates.

Es macht was es soll - der Change an sich ist OK.
Leider erkennt evcc jetzt aber auch deutlich häufiger (korrekt!) Charger out-of-sync, da die Easee gerne mal länger zum Schalten braucht (oder auch der DCC Update direkt nach dem Resume ab und an schief geht). Bisher hat Enabled() effektiv eigentlich nur auf DCC gearbeitet - jetzt spielt OP_CODE eine Rolle und daher ist der Wert wichtiger geworden.

In Kombination mit der `syncCharger()` Logik im aktuellen Master halte ich diesen Change für nicht ok. 
Diese Logik sorgt immer noch für viele kurzzeitige pause/resume Aufrufe - mit diesem Change potentiell für noch mehr - je nach Timing.
Das kommt dann wiederum bei Wiederholungen bzw. Retries des Resume nicht gut. 
Bei mir hat das Auto z.B. aufgegeben, weil einfach zu häufig in kurzer Zeit gestartet und gestoppt wurde.
Außerdem kam die Easee API irgendwann mit den ganzen OP_CODE changes nicht mehr hinterher.

Ich habe zwei Vorschläge um hier weiter zu kommen.
Das einfachste ist, hier kurz Pause zu machen und auf #9075 zu warten, weil dann die unnötigen stop/start Zyklen verschwinden. Aus pause/resume/dcc wird dann ein einfaches resume/dcc. Das hat unter anderem den Effekt, daß der Charger nicht mehr ständig zwischen den verschiedenen OP_MODES (2 vs 3/6) hin und her springt. Im schlimmsten Fall wird ein aktiver Charger angewiesen nochmal zu aktivieren. Somit wird dann auch die Antwort von `Enable()` stabiler.
Das wird die Situation deutlich besser machen - ich muß diese Kombination aber erst noch (morgen?) testen.

Zusätzlich sehe ich noch folgenden zweiten Ansatz der das Verhalten unabhängig von Änderungen im Loadpoint generell robuster macht. Ich habe beobachtet daß bei einigen Aktivierungen das ProductUpdate für den CHARGER_OP_MODE erst sehr spät ankommt. Aktuell warten wir da aber nicht drauf. Das führt in ungünstigen Konstellationen dazu, daß der pause/resume Zyklus sich komplett überschneidet. Der Charger ist folglich noch dabei sich einzuschalten, wenn die API bereits wieder abschalten sagt. Das war auch der oben geschilderte Fall, bei dem mein Auto dann in letzter Instanz keine Lust mehr hatte.

Um das zu entzerren könnten wir - analog zum `waitForDynamicChargerCurrent()` - bei Resume und Pause ebenfalls auf ein ProductUpdate (für CHARGER_OP_MODE) warten - also ein `waitForChargerOpMode()` einbauen. Damit stellen wir sicher, daß der Charger auch physikalisch den Switch gemacht hat bevor wir das DCC Update senden. Dann kann es die gerade beschriebenen Überschneidungen nicht mehr geben und die Box pendelt nicht zwischen Disable() und Enable(). Außerdem macht es den Easee Adapter unabhängiger von der Logik im Loadpoint, was auch nicht verkehrt ist.

jheinitz, 2023-07-25:
Moin!

Bei uns sollte der Enyaq heute mit der Easee mit Nightly Version `0.118.10 eea661` laden. Eine knappe Stunde ging es gut, dann hat der Enyaq auch aufgegeben. 
Was ich damit andeuten will, ist, dass es vielleicht nicht am aktuellen Master liegt, sondern schon vorher das Problem da war. Ich habe dann auf den Modus **NOW** gestellt, da es schnell gehen musste.

Grüße Jens

andig, 2023-07-25:
@GrimmiMeloni warten auf OpMode klingt sinnig. Oder die Easee ist halt nur mit interval 1m zu betreiben. Ich warte auf deinen Test- allerdings sagtest du dass #9075 aktuell mit Easee nicht tut?

foto-andreas, 2023-07-25:
@jheinitz Ich habe beim Kona/Easee seit vermutlich Version 0.118.8 auch manchmal beobachtet, dass das Auto nach ein paar Stunden aufgibt und nicht mehr lädt, bis ichs dann merke und auf NOW stelle. Fand, dass es mit 0.118.9 besser war, hab in der Zeit aber nicht so viel zuhause = Solar geladen. Ich würde also eine Anpassung in diesem Bereich auch begrüßen.

GrimmiMeloni, 2023-07-25:
> allerdings sagtest du dass #9075 aktuell mit Easee nicht tut?

Ja, aber nur wenn wir #9075 alleine betrachten.

Soweit ich es sehen kann müssen wir #9075 **und** diesen Change hier mit reinnehmen, oder keinen von beiden.
Anders ausgedrückt: Die Anpassung des Verhaltens des Loadpoints bedingt dass für die Easee auch die Implementierung von `Enabled()` korrigiert wird.

Sorry, ich merke auch gerade, daß ich mich hiermit  
> hier kurz Pause zu machen und auf #9075 zu warten

etwas irreführend ausgedrückt habe. Was ich eigentlich sagen will ist: 

> hier kurz Pause machen und auf **_den Test in Kombination mit_** #9075 warten." 

Ich bin relativ optimistisch das diese Changes **zusammen** ok sind.

andig, 2023-07-30:
Fix moved to https://github.com/evcc-io/evcc/pull/9156

closed 2023-07-30 by andig