evcc-io/evcc #9112: Leistungsanzeige bei

Razer2412, 2023-07-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

naltatis, 2023-08-01:
Siehe auch https://github.com/evcc-io/evcc/issues/9059#issuecomment-1659782863

> ist es möglich, dass Leistungswerte unter 1 kW immer in W statt kW angezeigt werden?

Wir haben hier bewusst eine einheitliche Formatierung eingebaut damit Werte die nebeneinander stehen immer die gleiche Einheit haben und einfacher vergleichbar sind.

Für den ganzen Bereich gibt es eine Logik die von kW auf W wechselt wenn die Summe im Balkendiagramm größer 1kW ist. Diesen Mechanismus würde ich noch mal anpassen. Hier können wir mMn. bereits von kW auf W umstellen wenn keiner der Werte größer 1kW ist.

@Razer2412 Das würde an deinem konkreten Screenshot nichts ändern, wäre aber ein Schritt in die Richtung.

LiamHD, 2023-08-01:
Siehe auch meinen Kommentar in https://github.com/evcc-io/evcc/issues/9059#issuecomment-1659873540

closed 2023-08-01 by naltatis