evcc-io/evcc #9128: `evcc vehicle` Fehler Exit Code

StefanSchoof, 2023-07-26:
**Is your feature request related to a problem? Please describe.**
Der Befehl `evcc vehicle` gibt einen Exit Code 0 zurück auch, wenn das Fahrzeug nicht erstellt werden kann. Ich habe einen Cron Job der überprüft, ob evcc nightly noch mein Auto und mein PV Meter auslesen kann. Für Probleme mit dem meter kann ich einfach auf den Exit Code nutzen. Für vehicle muss dafür die Ausgabe geparst werden.

Beispiel config
```yaml
log: debug
meters:
  - name: pv
    type: template
    template: sunspec-inverter
    usage: pv
    modbus: tcpip
    id: 1
    host: wronghost
    port: 502

chargers:
  - name: idcharger
    type: custom
    enable:
      source: js
      vm: shared
      script: |
        set();
        var lp = state.loadpoints[0];
        lp.enabled = val;
        if (lp.enabled) lp.chargepower = lp.maxcurrent * 230 * lp.phases; else lp.chargepower = 0;
    enabled:
      source: js
      vm: shared
      script: |
        state.loadpoints[0].enabled;
    status:
      source: js
      vm: shared
      script: |
        if (state.loadpoints[0].enabled) "C"; else "B";
    maxcurrent:
      source: js
      vm: shared
      script: |
        set();
        var lp = state.loadpoints[0];
        lp.maxcurrent = val;
        if (lp.enabled) lp.chargepower = lp.maxcurrent * 230 * lp.phases; else lp.chargepower = 0;

vehicles:
  - type: template
    template: id
    title: ID.3
    user: user@example.com
    password: wrongpassword
    vin: wrongvin
    capacity: 58
    mode: pv
    minSoC: 40
    targetSoC: 80
    cloud: false
    name: ID3

site:
  title: Home
  meters:
    grid: grid
    pv: pv

loadpoints:
  - title: Carport
    charger: idcharger
    vehicle: ID3
    mode: pv
    phases: 1
    maxcurrent: 16
    resetOnDisconnect: true
```

Ausgabe

```shell
# evcc vehicle
[main  ] INFO 2023/07/26 18:51:13 evcc 0.118.10 (06d69a83)
[main  ] INFO 2023/07/26 18:51:13 using config file: /etc/evcc.yaml
[db    ] INFO 2023/07/26 18:51:13 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2023/07/26 18:51:14 creating vehicle ID3 failed: cannot create vehicle 'template': cannot create vehicle 'id': login.errors.password_invalid
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'id': login.errors.password_invalid
Capacity: 58.0kWh
Features: [Offline]
# echo $?
0
# evcc meter
[main  ] INFO 2023/07/26 18:51:27 evcc 0.118.10 (06d69a83)
[main  ] INFO 2023/07/26 18:51:27 using config file: /etc/evcc.yaml
[db    ] INFO 2023/07/26 18:51:27 using sqlite database: /root/.evcc/evcc.db
[main  ] FATAL 2023/07/26 18:51:27 cannot create meter 'pv': cannot create meter 'template': cannot create meter 'modbus': dial tcp: lookup wronghost on 192.168.178.1:53: no such host
# echo $?
1
```

**Describe the solution you'd like**
Sollte `evcc vehicle` ein Problem beim laden haben, sollte der Fehler Exit Code 1 zurück gegeben werden.

**Describe alternatives you've considered**
Ausgabe parsen und regex nutzen :(

**Additional context**
https://github.com/evcc-io/evcc/discussions/9063#discussioncomment-6519836

andig, 2023-07-26:
An den ursprünglichen Fehler kommt man nicht so elegant ran, aber zumindest ließe sich abbrechen:

```go
if _, ok := vehicle.(*wrapper.Wrapper); ok {
	fatal(fmt.Errorf("vehicle not available: %w", api.ErrNotAvailable))
}
```

StefanSchoof, 2023-07-26:
Danke