evcc-io/evcc #9143: easee: enhance charge start robustness

GrimmiMeloni, 2023-07-28:
Zusätzlich sehe ich noch folgenden zweiten Ansatz der das Verhalten unabhängig von Änderungen im Loadpoint generell robuster macht. Ich habe beobachtet daß bei einigen Aktivierungen das ProductUpdate für den CHARGER_OP_MODE erst sehr spät ankommt. Aktuell warten wir da aber nicht drauf. Das führt in ungünstigen Konstellationen dazu, daß der pause/resume Zyklus sich komplett überschneidet. Der Charger ist folglich noch dabei sich einzuschalten, wenn die API bereits wieder abschalten sagt. Das war auch der oben geschilderte Fall, bei dem mein Auto dann in letzter Instanz keine Lust mehr hatte.

Um das zu entzerren könnten wir - analog zum `waitForDynamicChargerCurrent()` - bei Resume und Pause ebenfalls auf ein ProductUpdate (für CHARGER_OP_MODE) warten - also ein `waitForChargerOpMode()` einbauen. Damit stellen wir sicher, daß der Charger auch physikalisch den Switch gemacht hat bevor wir das DCC Update senden. Dann kann es die gerade beschriebenen Überschneidungen nicht mehr geben und die Box pendelt nicht zwischen Disable() und Enable(). Außerdem macht es den Easee Adapter unabhängiger von der Logik im Loadpoint, was auch nicht verkehrt ist.

_Originally posted by @GrimmiMeloni in https://github.com/evcc-io/evcc/issues/9100#issuecomment-1650367297_

GrimmiMeloni, 2023-07-28:
@andig please assign to me

GrimmiMeloni, 2023-07-28:
FYI - ich tracke diese Idee in einem eigenen Issue, da dies auch mit dem aktuellen Master funktionieren sollte und wir dann die Changes nicht alle en-bloc reinnehmen müssen.

andig, 2023-07-29:
Wie siehts hier aus- weitere Tests notwendig?

GrimmiMeloni, 2023-07-29:
Gerade erfolgreich auf Basis von Master (zusammen mit dem CurrentGetter change, aber ohne alles andere von #8289) getestet. Diverse Starts / Stop und auch Phasenwechsel haben funktioniert. Kann meiner Ansicht nach auch gemerged werden.

PR #9156 (2023-07-30): Easee: wait for op_mode change when pausing/resuming
fix #9143
closed 2023-07-30 by andig