evcc-io/evcc #9145: Easee: support sync charger improvements in #8298

andig, 2023-07-28:
/cc @GrimmiMeloni

andig, 2023-07-28:
Aus https://github.com/evcc-io/evcc/issues/8298#issuecomment-1654578006:

> um einen sanity check erweitern der solche Situationen erkennt und der dann dazu führt das charger.MaxCurrent() aufgerufen wird, auch wenn es laut State vom Loadpoint nicht notwendig ist?

Das ist alles retrospektives Gewurschtel. Dann fände ich es besser zu schauen, ob der Vorgabestrom passt. PR folgt.

GrimmiMeloni, 2023-07-28:
Please assign to me, @andig - I will take care of this likely tomorrow.

GrimmiMeloni, 2023-07-29:
Die Idee aus dem Kommentar in #8298 für die Easee war bei einem "halben enable" (also resume funktioniert, aber DCC Update scheitert) wieder zu pausieren. Das ist aber nicht mehr notwendig, denn der Loadpoint kann seit gestern im nächsten Intervall den Strom prüfen und ggf. den Charger nachregeln.

Das würde auch wieder zu mehr stop/start Zyklen führen, die wir ja gerade versuchen loszuwerden.
Ich denke hier können wir einfach zu machen.

closed 2023-07-29 by andig