evcc-io/evcc #9146: RCT Power Batterie Fehlermeldung Batterie invalid response

hrncvk, 2023-07-28:
### Describe the bug

Hallo Zusammen,

ich bin bisher sehr zufrieden mit evcc! Vielen Dank erstmal dafür!

Leider bekomme ich sporadisch folgende Fehlermeldungen:

1. battery 1 soc: invalid response: Cmd Response (05) Id #INVALID (4BC0F974) Data [70 60 232 0]

2. battery 1 soc: invalid response: Cmd Response (05) Id #INVALID (C0A7074F) Data [58 189 129 108 80 83 32 49 48 46 48 32 86 87 52 81 0 0 0 0 0 0 0 0 0 0 0 0 50 51 42 69 79 14 25 195 35 158 21 63 0 0 7 13 0 0 0 0 50 46 51 46 53 54 54 49 0 0 0 0 0 0 0 0 48 48 54 53 66 52 48 50 51 57 49 55 0 0 0 0 0 0 0 0 223 21 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0]

Mir ist bekannt, das ich nicht parallel in EVCC und auf dem Smartphone die RCT native App nutzen darf, da es zu diesen konflikten kommen kann. Ich tue dies aber nicht.

### Steps to reproduce

1. EVCC öffnen
2. Browser mit EVCC offen lassen
3. Erster Fehler nach 1 Minute sichtbar
4. Zweiter Fehler nach 5 Minuten sichtbar

### Configuration details

```yaml
liefere ich gerne nach
```

### Log details

```text
liefere ich gerne nach
```

### What type of operating system are you running?

Linux

andig, 2023-07-28:
Könntest Du das Issue bitte neu in https://github.com/mlnoga/rct eröffnen?

/cc @mlnoga

TobiasHuber1980, 2023-07-28:
Habe ich auch immer wieder mit

`grid energy: invalid response: Cmd Response (05) Id #INVALID (C0A7074F) Data [71 240 52 38 80 83 32 56 46 48 32 68 83 76 69 0 0 0 0 0 0 0 0 0 0 0 0 0 119 61 110 69 200 249 61 193 0 0 0 63 0 0 5 13 0 0 0 0 50 46 51 46 53 54 54 49 0 0 0 0 0 0 0 0 48 48 54 53 66 52 56 50 51 50 52 54 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0]`

hab das issue neu aufgemacht @mlnoga

mlnoga, 2023-07-30:
Danke. Aufgenommen als [https://github.com/mlnoga/rct/issues/10](https://github.com/mlnoga/rct/issues/10), ich schaue es mir an.

TobiasHuber1980, 2023-08-12:
also bei mir kommt immer noch:
`battery 1 soc: invalid response to read of 42617474657279207374617465206F6620636861726765: Cmd Response (05) Id Battery power [W] (400F015B) Data [193 78 155 47]`

`pv 1 power: invalid response to read of 536F6C61722067656E657261746F72204120706F776572205B575D: Cmd Response (05) Id #INVALID (4BC0F974) Data [69 215 80 0]`

Ebenso wird der Hausverbrauch falsch dargestellt, Wert "0,00 kW". 
Ich denke, Ursache ist immer noch die gleiche... 
Habe Intervall mal auf 30 Sekunden hochgesetzt, bringt aber auch nichts.
Version: `0.118.11 (dbe0f2f5)`

[image]
[image]

/cc @mlnoga
/cc @andig

andig, 2023-08-12:
Nimm bitte die nightly version...