evcc-io/evcc #9150: Update not possible - Repository 402 - Payment Required

Hoelli4C, 2023-07-28:
### Describe the bug

It looks like there is a problem with the repository?

`E: Failed to fetch https://dl.cloudsmith.io/public/evcc/stable/deb/debian/dists/any-version/InRelease  402  Payment Required [IP: 143.204.98.14 443]
E: The repository 'https://dl.cloudsmith.io/public/evcc/stable/deb/debian any-version InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.`

### Steps to reproduce

1. sudo apt-get update on Raspberry Pi 

### Configuration details

```yaml
not applicable
```

### Log details

```text
not applicable
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.7

StefanSchoof, 2023-07-28:
https://status.cloudsmith.io/ says "Partially Degraded Service". This might be related.

roennieinator, 2023-07-28:
I get the same error while installing for the first time

PolyGnom123, 2023-07-28:
Die Meldung beim Update des Repositories kann ich ebenfalls bestätigen.

TobiasSt78, 2023-07-28:
same problem. evcc ist zu beliebt und so wurde wohl das Datenvolumen von Cloudsmith aufgebracht?

andig, 2023-07-28:
Es scheint einen Fehler bei Cloudsmith zu geben. Müssen wir abwarten...

StefanSchoof, 2023-07-28:
[Status](https://status.cloudsmith.io/) ist wieder grün, Problem noch da.

Laut https://github.com/balena-io/etcher/issues/3672#issuecomment-1005482209 kann es auch das monthly quota sein.

andig, 2023-07-28:
Guter Hinweis. Quota exceeded. Da hilft wohl nur weniger Releases pro Monat :O. Evtl. geht es jetzt wieder?

roennieinator, 2023-07-28:
it works again :)

Hoelli4C, 2023-07-28:
Works again, issue can be closed.

closed 2023-07-28 by Hoelli4C

andig, 2023-08-08:
We've upped the quota limit.

mstormi, 2024-01-29:
I'm getting hit by this today, could you please check ?

andig, 2024-01-29:
Thanks for the hint. Please retry.

mstormi, 2024-01-29:
LGTM now, thanks for your prompt service

boris-b3d, 2024-03-27:
Hi evcc team,

seems that the quota is reached again? Cannot update or install my evcc installation.

402  Payment Required [IP: 81.28.12.12 443]
Reading package lists...
E: Failed to fetch https://dl.evcc.io/public/evcc/stable/deb/ubuntu/dists/jammy/InRelease  402  Payment Required [IP: 81.28.12.12 443]
E: The repository 'https://dl.evcc.io/public/evcc/stable/deb/ubuntu jammy InRelease' is not signed.

StefanSchoof, 2024-03-28:
You should update the Package Repository https://github.com/evcc-io/evcc/discussions/12902

andig, 2024-03-28:
We still hae people accessing the old repo and need everyone to switch.

goebelmeier, 2024-03-28:
@andig aber https://dl.evcc.io/ ist doch das neue Repo oder irre ich mich?

andig, 2024-03-28:
Ja, aber das geht (seltener) auch auf das gleiche Backend.