evcc-io/evcc #9157: evcc 0.118.11 hyundai unauthorized

FuR1u5, 2023-07-29:
### Describe the bug

Hallo,

ich hatte bis gestern die normale stable Version installiert, tesla und hyundai wurde auch sauber abgerufen, seit heute klappt der Abruf von dem hyundai leider nicht mehr. Bin daraufhin auf die nightly gewechselt, dort ist der Fehler aber leider immer noch. Das Auto wird als offline angezeigt und wenn ich mit evcc vehicle den Status überprüfen will kommt der unten benannte Fehler.

Die Bluelink App funktioniert einwandfrei, ist der Fehler bekannt?

### Steps to reproduce

Fehler besteht durchgehend.

### Configuration details

```yaml
type: template
  template: hyundai
  title: Blue
  user: *****
  password: *****
  language: de
  name: ev3
```

### Log details

```text
[main  ] ERROR 2023/07/29 10:59:59 creating vehicle ev3 failed: cannot create vehicle 'template': cannot create vehicle 'hyundai': cannot get vehicles: unexpected status: 401 (Unauthorized)
```

### What type of operating system are you running?

Linux

### Version

0.118.11

VolkerK62, 2023-07-29:
hat Hyundai wieder an der API gedreht?
https://github.com/Hacksore/bluelinky/issues/259

mclane, 2023-07-29:
same here with Kia and version 0.118.10

emcepe, 2023-07-29:
so langsam geht mir Hyundai mit ihren Spielchen richtig auf die Nerven :-(

pencoe, 2023-07-29:
Hier mir einem Kia ebenfalls. Hatte (nach der 118.11) schon 118.9 und 118.10 ausprobiert, bis ich diesen Thread gesehen habe...

Griepen, 2023-07-29:
Ich habe gestern den EVCC von einem anderen Raspi mit auf den OpenHab Raspi gepackt und dabei von 114 geupdated. Nun ist der Zoe seit dem nur noch als Offline-Fahrzeug verfügbar. 
Hängt das vielleicht an etwas anderem als an Hundai-Kia?

VolkerK62, 2023-07-29:
@andig  scheint gefixed zu sein https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/issues/370#issuecomment-1656768843
Musst du noch etwas machen, oder ist das Problem damit behoben?

FuR1u5, 2023-07-29:
> @andig scheint gefixed zu sein [Hyundai-Kia-Connect/hyundai_kia_connect_api#370 (comment)](https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/issues/370#issuecomment-1656768843) Musst du noch etwas machen, oder ist das Problem damit behoben?

Bei mir besteht das Problem weiterhin, denke das muss etwas im Code angepasst werden.

Lex318, 2023-07-30:
Bei mir ist seit vorgestern (Update auf 0.118.11 und gleichzeitig im Fahrzeug Bestätigung der neuen Datenschutzbestimmungen) auch mein Ioniq5 offline:
cannot get vehicles: unexpected status: 401 (Unauthorized)
Sollte ich die aktuelle Nightly probieren?

VolkerK62, 2023-07-30:
das Problem ist noch nicht gelöst (sonst wäre dieses Issue geschlossen)

FuR1u5, 2023-07-30:
Ich habe eben auf die Version 0.118.11+1690683508 aktualisiert, jetzt klappt die Verbindung zu Hyundai wieder ohne Probleme :)

andig, 2023-07-30:
> Vielen Dank für den schnellen fix 

Da wurde nichts gefixt...

closed 2023-07-30 by andig

FuR1u5, 2023-07-30:
Sehr cool, selten so eine schnelle Reaktion bei einer kostenlosen Software gesehen 🥇

andig, 2023-07-30:
Wir freuen uns über Unterstützung in Form von Sponsoring :)

d-feulner, 2023-07-30:
Muss man jetzt noch irgendwas machen damit die Verbindung wieder klappt? Habe die v0.118.11 installiert, gestern hat es nicht geklappt.

VolkerK62, 2023-07-30:
ist morgen im Nighly vorhanden

emcepe, 2023-07-30:
> Muss man jetzt noch irgendwas machen damit die Verbindung wieder klappt? Habe die v0.118.11 installiert, gestern hat es nicht geklappt.

Ich habe evcc vorhin neu gestartet, als die Info kam, dass es wieder funktionieren sollte, und es klappt wieder einwandfrei.

Ebenso klappt Bluelinky wieder ohne Probleme, ohne Code Update.

VolkerK62, 2023-07-30:
ja, weil mittlerweile auch das externe Problem gefixed wurde https://github.com/neoPix/bluelinky-stamps/issues/11