evcc-io/evcc #9161: easee wallbox zeigt keine Werte an

manny4566, 2023-07-29:
### Describe the bug

Hi,

ich musste gerade bei meiner aktiven Ladung feststellen, dass keine werte im evcc angezeigt werden.
Gibt es ein Probem mit easee aktuell?

### Steps to reproduce

root@evcc-srv:~# evcc -l debug charger
[main  ] INFO 2023/07/29 14:38:48 evcc 0.118.11
[main  ] INFO 2023/07/29 14:38:48 using config file: /etc/evcc.yaml
[db    ] INFO 2023/07/29 14:38:48 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/07/29 14:38:48 connecting evcc-1409056543 at tcp://192.168.0.201:1883
[mqtt  ] DEBUG 2023/07/29 14:38:48 tcp://192.168.0.201:1883 connected
Power:          0W
Energy:         0.0kWh
Current L1..L3: 0A 0A 0A
Charge status:  C
Enabled:        false
Charged:        0.0kWh
Identifier:     <none>

### Configuration details

```yaml
chargers:
- type: template
  template: easee
  name: wallbox
  user: XXX
  password: XXX
  charger: XXX
```

### Log details

```text
root@evcc-srv:~# evcc --log debug
[main  ] INFO 2023/07/29 14:39:50 evcc 0.118.11
[main  ] INFO 2023/07/29 14:39:50 using config file: /etc/evcc.yaml
[main  ] INFO 2023/07/29 14:39:50 starting ui and api at :7070
[db    ] INFO 2023/07/29 14:39:50 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/07/29 14:39:50 connecting evcc-858580337 at tcp://192.168.0.201:1883
[mqtt  ] DEBUG 2023/07/29 14:39:50 tcp://192.168.0.201:1883 connected
[main  ] ERROR 2023/07/29 14:39:56 creating vehicle ev2 failed: cannot create vehicle 'template': cannot create vehicle 'bmw': Post "https://customer.bmwgroup.com/gcdm/oauth/authenticate": dial tcp: lookup customer.bmwgroup.com: i/o timeout
[lp-1  ] WARN 2023/07/29 14:39:56 locking phase config to 3p for switchable charger
[main  ] WARN 2023/07/29 14:39:56 tariff configuration changed, use co2 instead of planner for co2 tariff
[site  ] WARN 2023/07/29 14:39:56 deprecated: use 'pv' instead of 'pvs'
[site  ] WARN 2023/07/29 14:39:56 deprecated: use 'battery' instead of 'batteries'
[site  ] WARN 2023/07/29 14:39:56 battery configured but residualPower is missing (add residualPower: 100 to site)
[site  ] INFO 2023/07/29 14:39:56 site config:
[site  ] INFO 2023/07/29 14:39:56   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/07/29 14:39:56     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/07/29 14:39:56     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/07/29 14:39:56     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/07/29 14:39:56   vehicles:
[site  ] INFO 2023/07/29 14:39:56     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/07/29 14:39:56 loadpoint 1:
[lp-1  ] INFO 2023/07/29 14:39:56   mode:        minpv
[lp-1  ] INFO 2023/07/29 14:39:56   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/07/29 14:39:56   meters:      charge ✓
[lp-1  ] INFO 2023/07/29 14:39:56     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/07/29 14:39:56 phase timer inactive
[lp-1  ] DEBUG 2023/07/29 14:39:56 pv timer inactive
[lp-1  ] DEBUG 2023/07/29 14:39:56 guard timer inactive
[lp-1  ] DEBUG 2023/07/29 14:39:56 max charge current: 6A
[main  ] FATAL 2023/07/29 14:39:56 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
```

### What type of operating system are you running?

Linux

### Version

0.118.11

ministryofsillywalks, 2023-07-29:
Meine beiden Easee boxen werden mir online auf easee.cloud und in der easee app als offline angezeigt.
Evtl haben die mal wieder Probleme mit deren cloud Lösung?
Siehst du in der easee app irgendwelche Werte?

manny4566, 2023-07-29:
> Meine beiden Easee boxen werden mir online auf easee.cloud und in der easee app als offline angezeigt. Evtl haben die mal wieder Probleme mit deren cloud Lösung? Siehst du in der easee app irgendwelche Werte?

Beide sind online,. sowohl in der App als auch der Webseite. 
Die Status Seite von Easee zeigt auch aktuell keine Störung an.