evcc-io/evcc #9164: Ladepunkt geht auf Default auch beim Anstecken eines Fahrzeugs

pencoe, 2023-07-29:
### Describe the bug

Wenn in der evcc.yaml für den Loadpoint "resetOnDisconnect: true" eingetragen ist, dann sollte nach meinem Verständnis nur beim Abstecken eines Fahrzeugs der Switch erfolgen, nicht jedoch beim Anstecken eines neuen Fahrzeugs.

Ich meine, dass das in 118.9 noch korrekt funktioniert hat, bin aber nicht sicher. In 118.11 passiert jedenfalls der Reset auch beim Anstecken des Fahrzeugs, weshalb man nicht mehr im Vorhinein ein anderes Fahrzeug wählen kann.

Zeile 1 und 2 des Log-Ausschnitts sind noch ok, vor Zeile 3 wurde das andere Fz gewählt, in Zeile 3 wird es angesteckt und in Zeile 4 fälschlich wieder auf Default zurückgeschaltet, was dann (natürlich) zu den Fehlermeldungen in den letzten beiden Zeilen führt (der Citigo lädt 2-phasig, der Niro 3-phasig)

### Steps to reproduce

1. Fahrzeug abstecken
2. anderes Fahrzeug als das Default-KFZ im Dropdown wählen
3. Fahrzeug anstecken

### Configuration details

```yaml
vehicles:
- type: template
  template: dacia
  name: Spring
  title: Spring
  capacity: 26.8
  user: **********************
  password: ********************
  phases: 1
  targetSoC: 80
- type: template
  template: skoda
  title: Citigo
  user: *******************
  password: *******************
  vin: **********************
  capacity: 32.3
  name: Citigo
  phases: 2
  targetSoC: 90
- type: template
  template: kia
  name: NiroEV
  title: Niro EV 
  user: **************** 
  password: ************************* 
  vin: **************************
  capacity: 64 # Akkukapazität in kWh (Optional)
  targetSoC: 80
  language: de 

chargers:
- type: template
  template: ellipro
  ski: ******************************
  ip: 192.168.48.66
  name: wallbox1

loadpoints:
- title: Carport
  charger: wallbox1
  mode: pv 
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  vehicle: Citigo
  disable:
    threshold: 0
    delay: 5m
  enable:
    threshold: 0
    delay: 30s
  guardduration: 5m
  soc:
    poll:
      mode: always 
      interval: 2h
    estimate: true
  resetOnDisconnect: true
```

### Log details

```text
Jul 28 14:15:58 RPNagel3 evcc[3476720]: [lp-1  ] INFO 2023/07/28 14:15:58 car disconnected
Jul 28 14:16:02 RPNagel3 evcc[3476720]: [lp-1  ] INFO 2023/07/28 14:16:02 vehicle updated: Citigo -> Niro EV
Jul 28 15:34:58 RPNagel3 evcc[3476720]: [lp-1  ] INFO 2023/07/28 15:34:58 car connected
Jul 28 15:34:58 RPNagel3 evcc[3476720]: [lp-1  ] INFO 2023/07/28 15:34:58 vehicle updated: Niro EV -> Citigo
Jul 28 15:34:58 RPNagel3 evcc[3476720]: [lp-1  ] INFO 2023/07/28 15:34:58 start charging ->
Jul 28 15:34:59 RPNagel3 evcc[3476720]: [lp-1  ] WARN 2023/07/28 15:34:59 charger out of sync: expected disabled, got enabled
Jul 28 15:35:28 RPNagel3 evcc[3476720]: [lp-1  ] WARN 2023/07/28 15:35:28 phase mismatch between 3p measured for 2p vehicle and 3p charger
Jul 28 15:35:58 RPNagel3 evcc[3476720]: [lp-1  ] WARN 2023/07/28 15:35:58 phase mismatch between 3p measured for 2p vehicle and 3p charger
```

### What type of operating system are you running?

Linux

### Version

0.118.11

andig, 2023-07-29:
`vehicle updated` kommt immer wenn das Fahrzeug gewechselt wird. In Deinem Log ist nicht erkennbar, worin die Ursache für die Änderung besteht, vllt. sogar durch den Anwender. Da ist leider nichts zu diagnostizieren.

closed 2023-07-29 by andig

pencoe, 2023-07-29:
Die Log-Zeilen sind ohne Lücke, ich selbst war der Anwender. Ich habe vor dem Anstecken im Dropdown den Niro gewählt (Log-Zeile 2) und später bei genug Sonne angesteckt (Zeile 3), was ohne weitere Intervention von mir zum Wechsel des Fz geführt hat. Sieht man auch an den identischen Timestamps der Zeilen 3-5.

Reicht es, bestimmte Bereiche des Log auf trace zu stellen, bevor ich das das nächste Mal probiere oder muss ich global trace einstellen?