evcc-io/evcc #9165: Nochmal: Ladepunkt geht auf Default auch beim Anstecken eines Fahrzeugs

pencoe, 2023-07-29:
`vehicle updated` kommt immer wenn das Fahrzeug gewechselt wird. In Deinem Log ist nicht erkennbar, worin die Ursache für die Änderung besteht, vllt. sogar durch den Anwender. Da ist leider nichts zu diagnostizieren.

_Originally posted by @andig in https://github.com/evcc-io/evcc/issues/9164#issuecomment-1656762949_

pencoe, 2023-07-29:
In dem Log steht NIE warum der Wechsel erfolgt, auch nicht bei Level trace...

Also nochmal:

Wenn in der evcc.yaml für den Loadpoint "resetOnDisconnect: true" eingetragen ist, dann sollte nach meinem Verständnis nur beim Abstecken eines Fahrzeugs der Switch erfolgen, nicht jedoch beim Anstecken eines neuen Fahrzeugs.

Ich meine, dass das in 118.9 noch korrekt funktioniert hat, bin aber nicht sicher. In 118.11 passiert jedenfalls der Reset auch beim Anstecken des Fahrzeugs, weshalb man nicht mehr im Vorhinein ein anderes Fahrzeug wählen kann.

Zeile 1 und 2 des Log-Ausschnitts sind noch ok, vor Zeile 3 wurde das andere Fz gewählt, in Zeile 3 wird es angesteckt und in Zeile 4 fälschlich wieder auf Default zurückgeschaltet, was dann (natürlich) zu den Fehlermeldungen in den letzten beiden Zeilen führt (der Citigo lädt 2-phasig, der Niro 3-phasig)

Steps to reproduce

    Fahrzeug abstecken
    anderes Fahrzeug als das Default-KFZ im Dropdown wählen
    Fahrzeug anstecken

Das lässt sich jederzeit einfach nachvollziehen.

Im Log:
Jul 29 19:03:47 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:03:47 ----
Jul 29 19:03:47 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:03:47 charge power: 0W
Jul 29 19:03:47 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:03:47 pv power: 756W
Jul 29 19:03:48 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:03:48 battery soc: 95%
Jul 29 19:03:48 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:03:48 battery power: 1862W
Jul 29 19:03:48 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:03:48 grid power: 12W
Jul 29 19:03:48 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:03:48 site power: 1974W
Jul 29 19:03:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:03:48 charge currents: [0 0 0]A
Jul 29 19:03:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:03:48 charger status: A
Jul 29 19:03:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:03:48 next soc poll remaining time: 1h58m0s
Jul 29 19:04:13 RPNagel3 evcc[3718015]: [lp-1  ] INFO 2023/07/29 19:04:13 vehicle updated: Citigo -> Niro EV (offline)
Jul 29 19:04:17 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:17 ----
Jul 29 19:04:17 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:17 charge power: 0W
Jul 29 19:04:17 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:17 pv power: 702W
Jul 29 19:04:18 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:18 battery soc: 95%
Jul 29 19:04:18 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:18 battery power: 1916W
Jul 29 19:04:18 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:18 grid power: -5W
Jul 29 19:04:18 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:18 site power: 2011W
Jul 29 19:04:18 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:18 charge currents: [0 0 0]A
Jul 29 19:04:18 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:18 charger status: A
Jul 29 19:04:47 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:47 ----
Jul 29 19:04:47 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:47 charge power: 2857W
Jul 29 19:04:47 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:47 pv power: 700W
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:48 battery soc: 95%
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:48 battery power: 4698W
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:48 grid power: -1W
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:48 site power: 4797W
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:48 charge currents: [0.65 6 5.77]A
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:48 charger status: C
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] INFO 2023/07/29 19:04:48 car connected
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] INFO 2023/07/29 19:04:48 vehicle updated: Niro EV (offline) -> Citigo
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:48 set target soc: 90
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:48 pv timer elapse
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:48 pv timer inactive
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:48 charger: guard elapse
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:48 guard timer inactive
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] INFO 2023/07/29 19:04:48 start charging ->
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:48 wake-up timer: stop
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [db    ] TRACE 2023/07/29 19:04:48 INSERT INTO `sessions` (`created`,`finished`,`loadpoint`,`identifier`,`vehicle`,`odometer`,`meter_start_kwh`,`meter_end_kwh`,`charged_kwh`,`solar_percentage`,`price`,`price_per_kwh`,`co2_per_kwh`) VALUES ("2023-07-29 19:04:48.018","0000-00-00 00:00:00","Carport","","Citigo",NULL,NULL,NULL,0.000000,NULL,NULL,NULL,NULL) RETURNING `id` 1 <nil>
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:48 vehicle status: A
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:48 vehicle soc: 78%
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:48 vehicle range: 257km
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] WARN 2023/07/29 19:04:48 charger out of sync: expected disabled, got enabled
[... cut ...]
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [site  ] DEBUG 2023/07/29 19:04:48 ----

Interessant sind (wieder) diese Zeilen:
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] DEBUG 2023/07/29 19:04:48 charger status: C
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] INFO 2023/07/29 19:04:48 car connected
Jul 29 19:04:48 RPNagel3 evcc[3718015]: [lp-1  ] INFO 2023/07/29 19:04:48 vehicle updated: Niro EV (offline) -> Citigo

Mit dem Connect wird das Fahrzeug gewechselt. Da ist kein mysteriöser Anwender beteiligt, und ich bin auch nicht zu doof auf einen Bildschirm zu gucken.

VolkerK62, 2023-07-29:
in der Config #9164  ist der Citigo als Standardfahrzeug konfiguriert.
Zitat Doku: `Beim Anschluss eines Fahrzeugs an den Ladepunkt wird damit immer davon ausgegangen dass dieses Fahrzeug angeschlossen wurde.`
Insofern nützt die VORHERIGE Fahrzeugauswahl gar nichts.

andig, 2023-07-29:
> passiert jedenfalls der Reset auch beim Anstecken des Fahrzeugs

woran machst du das fest?

VolkerK62, 2023-07-29:
Das ist kein Reset auf Loadpoint Defaults, das sind die `onIdentify` Parameter des Citigo.

pencoe, 2023-07-29:
> Das ist kein Reset auf Loadpoint Defaults, das sind die `onIdentify` Parameter des Citigo.

Da kann nix identifiziert werden (die Elli-Box liefert ja eben keine ID-Daten, leider), es wird ja lediglich festgestellt, dass (irgendein) Fahrzeug angesteckt wurde...

Aber wenn ich das hier 

> Zitat Doku: Beim Anschluss eines Fahrzeugs an den Ladepunkt wird damit immer davon ausgegangen dass dieses Fahrzeug angeschlossen wurde.
> Insofern nützt die VORHERIGE Fahrzeugauswahl gar nichts.

richtig verstehe, muss ich also darauf verzichten ein Default-Fahrzeug einzurichten, wenn ich mit mehreren Fahrzeugen hantiere(n muss), weil das das implementierte Verhalten ist. Das ist schade, ich hatte das anders verstanden: nach meinem Verständnis war resetOnDisconnect so, dass bei $true die Einstellung _auch des Fahrzeugs_ im Moment des Absteckens erfolgt und nicht auch beim Anstecken. Aber da werde ich dann mit leben müssen.

Mal sehen, evtl. mache ich dann einen Feature Request, dass das fummelige Dropdown vielleicht durch besser bedienbare große Knöpfe ersetzt wird :-)

VolkerK62, 2023-07-30:
>Da kann nix identifiziert werden (die Elli-Box liefert ja eben keine ID-Daten, leider), es wird ja lediglich festgestellt, dass (irgendein) Fahrzeug angesteckt wurde...

Das hat nichts mit der Wallbox zu tun. https://docs.evcc.io/docs/reference/configuration/vehicles#onidentify
Das ist der `onIdentify` Parameter vom Citigo:
`   targetSoC: 90`

> wenn ich mit mehreren Fahrzeugen hantiere(n muss), 

du hast für alle Fahrzeuge Login-Daten hinterlegt, also sind die doch alle "automatisch erkennbar"?
Warum nutzt du dieses feature nicht?

>nach meinem Verständnis war resetOnDisconnect so, dass bei $true die Einstellung auch des Fahrzeugs im Moment des Absteckens erfolgt und nicht auch beim Anstecken. 

Das passiert auch, wird aber durch das konfigurieren eines Standardfahrzeugs "ausgehebelt".

andig, 2023-07-30:
Wenn ein Defaultfahrzeug konfiguriert ist wird dieses beim Abstecken aktiviert. Nichts anderes passiert hier.

closed 2023-07-30 by andig