evcc-io/evcc #9183: Modbusproxy unreliable

CNCKitchen, 2023-07-31:
### Describe the bug

I've been very successfully using EVCC for a bunch of weeks now and passed along my Huawei Smart Meter readings to Home Assistant using the ModbusProxy. Unfortunately, after a recent update of EVCC and HA, Home Assistant is barely able to receive readings through the modbusproxy anymore and I get debug errors in the EVCC log.
Not sure if this is a problem of EVCC or HA.

### Steps to reproduce

Home Assistant 2023.7.3
Operating System 10.3

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xx

interval: 30s # control cycle interval

sponsortoken: xx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1
  host: 192.168.178.85  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
  timeout: 45s
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.85  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
  timeout: 45s
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.85  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3
  timeout: 45s
- name: my_pv
  type: template
  template: solarman
  usage: pv
  host: 192.168.178.88 # IP-Adresse oder Hostname
  user: xx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.) (Optional)
  password: xx  # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (Optional) 
- name: my_pv2
  type: template
  template: solarman
  usage: pv
  host: 192.168.178.94 # IP-Adresse oder Hostname
  user: xx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.) (Optional)
  password: xx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (Optional) 

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.178.97  
  name: wallbox4
  
vehicles:
- type: template
  template: tesla 
  title: Tesla Model Y LR  
  accessToken: xx
  refreshToken: xx
  capacity: 73  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  vehicle: ev4
  resetOnDisconnect: true

site:
  title: My home
  meters:
    grid: grid1
    pv:
    - pv2
    - my_pv
    - my_pv2
    battery:
    - battery3
  prioritySoc: 50 # Hausbatterie bekommt bis zum Soc 50% Priorität beim laden
  bufferSoc: 60 # Hausbatterie wird oberhalb Soc 60% als Puffer genutzt
  bufferStartSoc: 80 # hat die Hausbatterie Soc 80% erreicht startet der Ladevorgang
  residualPower: 100
    
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.40 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.08 # [currency]/kWh
    
modbusproxy:
  - port: 5021
    uri: 192.168.178.85:502
```

### Log details

```text
[lp-1  ] DEBUG 2023/07/31 10:01:46 charger status: B
[lp-1  ] DEBUG 2023/07/31 10:01:46 pv charge current: 0A = 0A + -0.47A (108W @ 1p)
[site  ] DEBUG 2023/07/31 10:02:07 ----
[lp-1  ] DEBUG 2023/07/31 10:02:07 charge power: 0W
[site  ] DEBUG 2023/07/31 10:02:09 pv power: 1786W
[site  ] DEBUG 2023/07/31 10:02:10 battery soc: 26%
[site  ] DEBUG 2023/07/31 10:02:10 battery power: -1472W
[proxy-5021] DEBUG 2023/07/31 10:02:12 failed to write response: write tcp 192.168.178.91:5021->192.168.178.91:40354: i/o timeout
[site  ] DEBUG 2023/07/31 10:02:12 grid power: 10W
[site  ] DEBUG 2023/07/31 10:02:16 grid currents: [0.56 -0.98 1.31]A
[site  ] DEBUG 2023/07/31 10:02:17 giving priority to battery charging at soc: 26%
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.11

andig, 2023-07-31:
Nothing buggy with an i/o timeout. Which version did work last?

andig, 2023-08-01:
Please also set `proxy-5021: trace` for log level.

andig, 2023-08-01:
Closing until more diagnosis available. Please update this issue when done.

closed 2023-08-01 by andig