evcc-io/evcc #9184: ABL : "cannot create charger i/o timeout" in evcc > 0.118.1

domi27, 2023-07-31:
### Describe the bug

Hallo,

bin jetzt da ich endlich mein E-Auto habe - vorher hat immer mein Nachbar getankt ;-) - über einen Bug (?) gestolpert, als ich meine EVCC Installation von 0.118.1 auf die aktuelle 0.118.11 upgraden wollte. 

Nach dem Upgrade funktioniert die Kommunikation mit meiner ABL Wallbox (hinter einem Modbus TCP Adapter) nicht mehr.

`FATAL 2023/07/17 17:40:41 cannot create charger 'wallbox4': cannot create charger 'template': cannot create charger 'abl': read tcp 192.168.178.21:50226->192.168.178.180:8899: i/o timeout`

Habe dann auch verschiedene Zwischenversionen ausprobiert, der Fehler ist auch in 0.118.2 und 0.118.8.

Konfiguration von EVCC und Modbus Adapter habe ich jeweils nicht geändert bei den Versionssprüngen.

Nach dem Downgrade auf 0.118.1 läuft die Kommunikation mit der Wallbox wieder sauber und auch das Laden. Leider brauche ich für meinen iX3 den Fix (#8713 ) aus 0.118.2 .... aber das Laden ist mir erstmal wichtiger ;-)

Laut Github ist ja im Code vom ABL Charger seit Monaten nichts geändert worden - falls ich das korrekt interpretiere.
Könnt ihr bitte mal schauen woran das liegen könnte ? 
Danke & Gruß
Michael

### Steps to reproduce

1. Upgrade EVCC auf Version > 0.118.1
2. Restart EVCC

### Configuration details

```yaml
chargers:
- type: template
  template: abl
  id: 1
  host: 192.168.178.180
  port: 8899
  modbus: rs485tcpip
  name: wallbox4
```

### Log details

```text
[main  ] INFO 2023/07/31 10:18:43 evcc 0.118.1
[main  ] INFO 2023/07/31 10:18:43 using config file: /home/michael/evcc/evcc.yaml
[db    ] INFO 2023/07/31 10:18:43 using sqlite database: /home/michael/.evcc/evcc.db
[db    ] TRACE 2023/07/31 10:18:43 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/07/31 10:18:43 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/07/31 10:18:43 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/07/31 10:18:43 SELECT * FROM `settings` 0 <nil>
[abl   ] TRACE 2023/07/31 10:18:43 modbus: send ":010300010002F9\r\n"
[abl   ] TRACE 2023/07/31 10:18:44 modbus: send ":010300010002F9\r\n"
[abl   ] TRACE 2023/07/31 10:18:44 modbus: recv ">010304014141363F\r\n"
[abl   ] TRACE 2023/07/31 10:18:44 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2023/07/31 10:18:44 modbus: recv ">01030A2EA1121503E803E803E83B\r\n"
[abl   ] TRACE 2023/07/31 10:18:44 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2023/07/31 10:18:44 modbus: recv ">01030A2EA1121503E803E803E83B\r\n"
[abl   ] TRACE 2023/07/31 10:18:44 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2023/07/31 10:18:45 modbus: recv ">01030A2EA1121503E803E803E83B\r\n"

===========

michael@raspi-knx:~/evcc $ evcc charger --log trace
[main  ] INFO 2023/07/31 10:17:06 evcc 0.118.2
[main  ] INFO 2023/07/31 10:17:06 using config file: /home/michael/evcc/evcc.yaml
[db    ] INFO 2023/07/31 10:17:17 using sqlite database: /home/michael/.evcc/evcc.db
[db    ] TRACE 2023/07/31 10:17:17 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/07/31 10:17:17 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/07/31 10:17:17 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/07/31 10:17:17 SELECT * FROM `settings` 0 <nil>
[abl   ] TRACE 2023/07/31 10:17:17 modbus: send ":010300010002F9\r\n"
[abl   ] TRACE 2023/07/31 10:17:27 modbus: send ":010300010002F9\r\n"
[main  ] FATAL 2023/07/31 10:17:37 cannot create charger 'wallbox4': cannot create charger 'template': cannot create charger 'abl': read tcp 192.168.178.21:43244->192.168.178.180:8899: i/o timeout

===========

michael@raspi-knx:~ $ evcc charger --log trace
[main  ] INFO 2023/07/17 17:40:21 evcc 0.118.8
[main  ] INFO 2023/07/17 17:40:21 using config file: /etc/evcc.yaml
[db    ] INFO 2023/07/17 17:40:21 using sqlite database: /home/michael/.evcc/evcc.db
[db    ] TRACE 2023/07/17 17:40:21 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/07/17 17:40:21 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/07/17 17:40:21 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/07/17 17:40:21 SELECT * FROM `settings` 0 <nil>
[abl   ] TRACE 2023/07/17 17:40:21 modbus: send ":010300010002F9\r\n"
[abl   ] TRACE 2023/07/17 17:40:31 modbus: send ":010300010002F9\r\n"
[main  ] FATAL 2023/07/17 17:40:41 cannot create charger 'wallbox4': cannot create charger 'template': cannot create charger 'abl': read tcp 192.168.178.21:50226->192.168.178.180:8899: i/o timeout	

===============

[main  ] INFO 2023/07/31 10:15:30 evcc 0.118.11
[main  ] INFO 2023/07/31 10:15:30 using config file: /home/michael/evcc/evcc.yaml
[db    ] INFO 2023/07/31 10:15:30 using sqlite database: /home/michael/.evcc/evcc.db
[db    ] TRACE 2023/07/31 10:15:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/07/31 10:15:30 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/07/31 10:15:30 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/07/31 10:15:30 SELECT * FROM `settings` 0 <nil>
[abl   ] TRACE 2023/07/31 10:15:30 modbus: send ":010300010002F9\r\n"
[abl   ] TRACE 2023/07/31 10:15:40 modbus: send ":010300010002F9\r\n"
[main  ] FATAL 2023/07/31 10:15:50 cannot create charger 'wallbox4': cannot create charger 'template': cannot create charger 'abl': read tcp 192.168.178.21:58388->192.168.178.180:8899: i/o timeout
```

### What type of operating system are you running?

Linux

andig, 2023-07-31:
Probier mal bitte

    timeout: 1s

oder grösser.

domi27, 2023-08-01:
Hi,
danke für die Antwort - wo muss das in die Config eingetragen werden ?
Wenn ich es in die Charger Sektion eintrage kommt ein Fehler :

`[main  ] FATAL 2023/08/01 13:16:41 cannot create charger 'wallbox4': cannot create charger 'template': invalid key: timeout`

danny2630, 2023-08-01:
Ich habe den gleichen Fehler allerdings mit einer Warp2 pro habe dann auf 0.111.0 ein downgrade gemacht jetzt läuft es wider.!

[image](https://github.com/evcc-io/evcc/assets/127605125/8746748e-cbbd-4287-b289-f4ab946c5958)

andig, 2023-08-01:
@danny2630 auch wenn völlig off-topic: schau Dir bitte mal Deine Fehlermeldung *GENAU* an. Du hast einen ganz anderen Fehler, eine Box mit ganz anderer Technologie. Entferne das `fw2` und schon läuft es...

danny2630, 2023-08-02:
> @danny2630 auch wenn völlig off-topic: schau Dir bitte mal Deine Fehlermeldung *GENAU* an. Du hast einen ganz anderen Fehler, eine Box mit ganz anderer Technologie. Entferne das `fw2` und schon läuft es...

Ok vielen Dank für die Info.
Irgendwie seltsam das dies nach dem Update passiert ist.

domi27, 2023-08-03:
Hi,

@andig nochmal die Frage wo in der "evcc.yaml" muss ich den timeout eintragen ?

Weil hier "https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml" finde ich den Eintrag nicht und wenn ich den zum Charger geben, dann bekomme ich einen FATAL error.

Danke

Michael

andig, 2023-08-04:
Ganz einfach (nur ohne template):

```yaml
chargers:
- type: abl
  id: 1
  uri: 192.168.178.180:8899
  timeout: 1s
  name: wallbox4

```

andig, 2023-08-04:
Kommando zurück. Geht nicht. Ich müsste Dir einen eigenen Build dafür machen, wäre kein großes Problem. Für welche Plattform würdest Du das brauchen?

domi27, 2023-08-04:
EVCC läuft bei mir auf dem Raspberry (glaube v3) :

`michael@raspi-knx:~/evcc $ uname -a
Linux raspi-knx 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux`

domi27, 2023-08-04:
Sieht - von meiner Einschätzung gut aus :

`[main  ] INFO 2023/08/04 11:43:51 evcc 0.118.11 (3df67db16)
[main  ] INFO 2023/08/04 11:43:51 using config file: /home/michael/evcc/evcc.yaml
[db    ] INFO 2023/08/04 11:43:51 using sqlite database: /home/michael/.evcc/evcc.db
[db    ] TRACE 2023/08/04 11:43:51 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2023/08/04 11:43:51 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2023/08/04 11:43:51 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2023/08/04 11:43:51 SELECT * FROM `settings` 0 <nil>
[abl   ] TRACE 2023/08/04 11:43:51 modbus: send ":010300010002F9\r\n"
[abl   ] TRACE 2023/08/04 11:43:52 modbus: send ":010300010002F9\r\n"
[abl   ] TRACE 2023/08/04 11:43:52 modbus: recv ">010304014141363F\r\n"
[abl   ] TRACE 2023/08/04 11:43:52 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2023/08/04 11:43:52 modbus: recv ">01030A2EA1121503E803E803E83B\r\n"
[abl   ] TRACE 2023/08/04 11:43:52 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: recv ">01030A2EA1121503E803E803E83B\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: recv ">01030A2EA1121503E803E803E83B\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: send ":0103002E0005C9\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: recv ">01030A2EA1121503E803E803E83B\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: send ":010300040001F7\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: recv ">01030204A155\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: send ":010300040001F7\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: recv ">01030204A155\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: send ":0103000F0005E8\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: recv ">01030A0F3F0215021503E8021574\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: send ":0103000F0005E8\r\n"
[abl   ] TRACE 2023/08/04 11:43:53 modbus: recv ">01030A0F3F0215021503E8021574\r\n"
Power:          0W
Current L1..L3: 0A 0A 0A
Charge status:  A
Enabled:        false`

Heißt in der "evcc.yaml" brauch ich gar nicht mehr die Optionen "modbus" und "template" ?
Habe ich da eine Änderung verpasst ? Dann sorry, habe aber eigentlich eifrig hier mit gelesen.

Gruß
Michael

andig, 2023-08-04:
Siehe https://github.com/evcc-io/evcc/pull/9241

closed 2023-08-04 by andig