evcc-io/evcc #9186: Unable to change months in sessions

RTTTC, 2023-07-31:
### Describe the bug

Clicking on the left or on the right (different months) does not do anything.

### Steps to reproduce

1. in EVCC ui select charging sessions
2. try to change months

### Configuration details

```yaml
v 0.118.11
```

### Log details

```text
Same using windows / edge and mac/safari
```

### What type of operating system are you running?

Linux

RTTTC, 2023-07-31:
There was an earlier issue already.. https://github.com/evcc-io/evcc/issues/9182

BTW if I change address link manually to month "6" (instead of 7) it opens up.

andig, 2023-08-01:
WFM. Could you please check for errors in the browser console or maybe record a video?

RTTTC, 2023-08-01:
Works for me too now.
What changed? New month is the culprit perhaps?
Did not work on 3 different devices yesterday.

andig, 2023-08-01:
I think this is a duplicate or rather different effect of underlying same problem.