evcc-io/evcc #9191: fehler beim starten sponsortoken

TobiasHuber1980, 2023-07-31:
### Describe the bug

sponsortoken
`rpc error: code = Unknown desc = non-200 OK status code: 403 Forbidden body: "{\n \"documentation_url\": \"https://docs.github.com/en/free-pro-team@latest/rest/overview/resources-in-the-rest-api#secondary-rate-limits\",\n \"message\": \"You have exceeded a secondary rate limit. Please wait a few minutes before you try again.\"\n}\n"`

### Steps to reproduce

1. evcc neu starten
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: info  
levels:  
  cache: error  
  
# unique installation id  
plant: *****  
  
interval: 15s # control cycle interval  
  
sponsortoken: *****  
  
# sponsors can set telemetry: true to enable anonymous data aggregation  
# see https://github.com/evcc-io/evcc/discussions/4554  
telemetry: false  

hier kommt der rest....
```

### Log details

```text
[main  ] INFO 2023/07/31 20:07:45 evcc 0.118.11 (e02f689e)
[main  ] INFO 2023/07/31 20:07:45 using config file: /etc/evcc.yaml
[main  ] INFO 2023/07/31 20:07:45 starting ui and api at :7070
[main  ] FATAL 2023/07/31 20:07:46 sponsortoken: rpc error: code = Unknown desc = non-200 OK status code: 403 Forbidden body: "{\n  \"documentation_url\": \"https://docs.github.com/en/free-pro-team@latest/rest/overview/resources-in-the-rest-api#secondary-rate-limits\",\n  \"message\": \"You have exceeded a secondary rate limit. Please wait a few minutes before you try again.\"\n}\n"
[main  ] FATAL 2023/07/31 20:07:46 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

0.118.11 (e02f689e)

Ahhpaddy, 2023-07-31:
gleicher Fehler auch bei mir nach Update auf 0.118.11??

TobiasHuber1980, 2023-07-31:
sponsortoken
`rpc error: code = Unknown desc = API rate limit exceeded for user IDxxxxxx`

Bullischorsch, 2023-07-31:
seems to work now (I created a new sponsor token and put it to the config, afterwards evcc started).

closed 2023-07-31 by andig

Derjensus, 2024-03-24:
Hallo,
muss mich hier mal kurz mit anhängen...
Versuche seit etlichen Wochen EVCC erfolgreich zum Laufen zu kriegen, scheitere dabei aber kläglich.
Habe mit einer Entratek Power Dot Pro 2 angefangen und Sie mittels OCPP und Clever-PV zum Laufen bekommen.
Aber da man ja unabhängiger sein möchte (und einige Dinge nicht so rund liefen), wurde schnell der Wunsch nach Änderung Gross.
Da ich beruflich teilweise mit Installationen/Inbetriebnahmen von Mennekes-Wallboxen zu tun habe, war mir bekannt, dass dort Bender-Laderegler verbaut sind. Kurz...als Elektrofachkraft alle benötigten Einzelteile besorgt inkl. EVCC-Sponsortoken und ... krieg es einfach nicht hin. Der Ladevorgang am EV geht, aber der Betrieb mit EVCC...neeeeee...

Die ganze Sache läuft auf einem Promox-System, wo Homeassistent und (zu Testzwecken) EVCC  installiert sind.
Alle Einstellungen im Laderegler sind nach Vorgabe EVCC.
Leider kommen dann beim Test Fehlermeldungen zum Sponsortoken, egal, wo ich teste (impl. im HASS oder als sep. EVCC)
siehe:

"Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/03/24 21:09:48 evcc 0.124.10
[main  ] INFO 2024/03/24 21:09:48 using config file: /config/evcc.yaml
[main  ] INFO 2024/03/24 21:09:48 starting ui and api at :7070
[main  ] FATAL 2024/03/24 21:09:48 sponsortoken: rpc error: code = Unknown desc = token is malformed: could not base64 decode claim: illegal base64 data at input byte 104
[main  ] FATAL 2024/03/24 21:09:48 will attempt restart in: 5m0s"

Hab den Token wohl schon 3 mal neu erstellt aber keine Änderung des Ergebnisses.

... Help ...
Vielen Dank im Voraus

MfG J.M.

flumb, 2025-03-05:
Hallo zusammen, 

ich habe das Problem auch. Ich versuche alles rein über die Oberfläche zu konfigurieren (leere evcc.yaml). Und erhalte die folgende Fehlermeldung: 
sponsortoken: rpc error: code = Unknown desc = token is malformed: could not JSON decode header: invalid character 'Ò' after object key