evcc-io/evcc #9194: Error when creating vehicle VW ID.4: invalid character 'd'

bolisaf, 2023-07-31:
### Describe the bug

worked for a while, but doesn't anymore for some reason.

### Steps to reproduce

1. copy the template from here: https://docs.evcc.io/docs/devices/vehicles/#we-connect-id
2. doesn't matter what you fill in for the user, password and vin, error stays the same
3. fire up evcc -l debug vehicle
...

### Configuration details

```yaml
sponsortoken: xxxxxxxxxxxxx

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
#modbusproxy:
  #  - port: 5200æ
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true

# meter definitions
[... 273 lines cut ...]
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
C:\Users\stere\Downloads\evcc_0.118.1_windows-amd64>evcc.exe -l debug vehicle
[main  ] INFO 2023/07/31 21:46:55 evcc 0.118.1
[main  ] INFO 2023/07/31 21:46:55 using config file: C:\Users\stere\Downloads\evcc_0.118.1_windows-amd64\evcc.yaml
[main  ] ERROR 2023/07/31 21:46:56 creating vehicle GTX failed: cannot create vehicle 'template': cannot create vehicle 'id': invalid character 'd' looking for beginning of object key string
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'id': invalid character 'd' looking for beginning of object key string
Capacity: 50.0kWh
Features: [Offline]
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.118.11

premultiply, 2023-07-31:
C:\Users\stere\Downloads\evcc_0.118.1_windows-amd64>evcc.exe -l debug vehicle
[main  ] INFO 2023/07/31 21:46:55 **evcc 0.118.1**

bolisaf, 2023-07-31:
when testing with 0.118.11

`C:\Users\stere\Downloads\evcc_0.118.11_windows-amd64>evcc.exe -l debug vehicle
[main  ] INFO 2023/07/31 22:40:52 evcc 0.118.11
[main  ] INFO 2023/07/31 22:40:52 using config file: C:\Users\stere\Downloads\evcc_0.118.11_windows-amd64\evcc.yaml
[main  ] ERROR 2023/07/31 22:40:58 creating vehicle GTX failed: cannot create vehicle 'template': cannot create vehicle 'id': missing state
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'id': missing state
Capacity: 77.0kWh
Features: [Offline]`

andig, 2023-08-01:
Terms&Conditions akzeptieren. Ich sehe lokal, dass Audi und ID3 funktionieren.

closed 2023-08-01 by andig

bolisaf, 2023-08-01:
> Terms&Conditions akzeptieren. Ich sehe lokal, dass Audi und ID3 funktionieren.

I’m sorry, what? Where do I accept which t&c?

thanks for your answer :)

andig, 2023-08-01:
VW App and/or website.

bolisaf, 2023-08-01:
this is the weirdest thing. I just logged into volkswagen.com, didn't get a prompt to accept t&c's or anything, didn't do anything on the driver portal and logged back out. now it's working again...

[image]

bolisaf, 2024-05-31:
Hi, I'm having the same issue again, but now I'm not able to get it to work by logging into volkswagen.com or vwgroup.io. what else can I try apart from this to get it to work again?

andig, 2024-06-01:
VW und Audi funktionieren soweit ich sehe.

bolisaf, 2024-06-01:
Yesterday evening when i wrote this it wasnt working, im checking now and its fine again… this happens every few months. im sure its on the end of volkswagen and not evcc. Are you seeing the same or is it stable for you?Thanks!L