evcc-io/evcc #9196: MQTT evcc/status is offline, even evcc updates other mqtt topics

jowi24, 2023-08-01:
### Describe the bug

- evcc is running without obvious problems
- evcc is connected to mqtt broker, updating topics like `evcc/site/gridPower` or `evcc/site/tariffEffectivePrice`
- however, topic `evcc/status` is set to `offline`

### Steps to reproduce

1. start evcc with mqtt configuration enabled
2. network connection to internet is lost for some time

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

mqtt:
  broker: openhab:1883
  topic: evcc

log: info
levels:
  cache: error
#  pulse: trace

# unique installation id
plant: xxx

interval: 10s # control cycle interval

sponsortoken: xxx
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
# telemetry: false

meters:
  - name: netz
    type: template
    template: solaredge-inverter
    usage: grid
    modbus: tcpip
    id: 1
    host: 192.168.178.99
    port: 1502
  - name: solaredge
    type: template
    template: solaredge-inverter
    usage: pv
    modbus: tcpip
    id: 1
    host: 192.168.178.99
    port: 1502

modbusproxy:
  - port: 1502
    uri: 192.168.178.99:1502
    rtu: false

vehicles:
- name: ev1
  type: template
  template: id
  title: ID.3
  user: xxx
  password: xxx
  vin: xxx
[... 198 lines cut ...]
[site  ] ERROR 2023/08/01 05:23:12 tariff: outdated
[telegram] ERROR 2023/08/01 05:23:16 Post "https://api.telegram.org/bot***/getUpdates": dial tcp: lookup api.telegram.org on 127.0.0.11:53: server misbehaving
[telegram] ERROR 2023/08/01 05:23:16 Failed to get updates, retrying in 3 seconds...
[site  ] ERROR 2023/08/01 05:23:22 tariff: outdated
[telegram] ERROR 2023/08/01 05:23:27 Post "https://api.telegram.org/bot***/getUpdates": dial tcp: lookup api.telegram.org on 127.0.0.11:53: server misbehaving
[telegram] ERROR 2023/08/01 05:23:27 Failed to get updates, retrying in 3 seconds...
[site  ] ERROR 2023/08/01 05:23:32 tariff: outdated
[telegram] ERROR 2023/08/01 05:23:38 Post "https://api.telegram.org/bot***/getUpdates": dial tcp: lookup api.telegram.org on 127.0.0.11:53: server misbehaving
[telegram] ERROR 2023/08/01 05:23:38 Failed to get updates, retrying in 3 seconds...
[site  ] ERROR 2023/08/01 05:23:42 tariff: outdated
[telegram] ERROR 2023/08/01 05:23:49 Post "https://api.telegram.org/bot***/getUpdates": dial tcp: lookup api.telegram.org on 127.0.0.11:53: server misbehaving
[telegram] ERROR 2023/08/01 05:23:49 Failed to get updates, retrying in 3 seconds...
[site  ] ERROR 2023/08/01 05:23:52 tariff: outdated
[site  ] ERROR 2023/08/01 05:24:02 tariff: outdated
[site  ] ERROR 2023/08/01 05:24:12 tariff: outdated
[site  ] ERROR 2023/08/01 05:24:22 tariff: outdated
```

### What type of operating system are you running?

Linux

### Version

0.118.10

andig, 2023-08-01:
Bitte erstmal Fehler beheben- in Deinem Fall funktioniert DNS nicht. Verhalten damit undefiniert.

jowi24, 2023-08-01:
Hallo andig, ich habe den evcc container neugestartet, jetzt ist der mqtt status wieder "online". DNS funktioniert auf dem Host und im Container, zumindest funktioniert ein `dig api.telegram.org` problemlos.

jowi24, 2023-08-01:
@andig der DNS Fehler steht in keinem Zusammenhang mit dem geschilderten Problem, da die Verbindung zum MQTT etabliert ist, ist es irrelevant ob DNS nun funktioniert oder nicht.

tobox, 2023-08-14:
Ich kann den Fehler bei mir bestätigen:

Ich nutze in Home Assistant "status = online" als availability topic und habe oft das Problem, dass EVCC in Home Assistant nicht funktioniert, obwohl EVCC selbst ganz normal läuft. 

Der Fehler ist bei mir reproduzierbar, wenn ich den MQTT Server neu starte!

EVCC verbindet sich dann wieder mit dem MQTT server, setzt aber status NICHT auf ONLINE!

andig, 2023-08-14:
@tobox wenn Dein System sich so verhält brauchen wir bitte- wie bei jedem anderen Fehler auch- das Logfile. "Bei mir auch" ist keine verwertbare Fehlermeldung :(. Wenn Du das hast bitte neue Fehlermeldung- hier war das mit anderen Themen überlagert.

tobox, 2023-08-14:
Ich kann gerne morgen ein neues issue aufmachen...

Aber hier nochmal im Schnelldurchgang die Möglichkeit den Fehler zu reprodzieren:

- evcc.dist.yaml so vereinfachen, dass  evcc sich starten lässt
- dabei sah der MQTT-Teil bei mir so aus:
```
mqtt:
  broker: mosquitto.fritz.box:1883
  topic: evcctest
  user: evcc
  password: geheim
```
- man sieht im MQTT-Explorer dann auch, wie evcctest/status auf online geht
- wenn ich dann den MQTT Server neu starte, geht evcctest/status auf offline
- EVCC verbindet sich sofort neu mit dem Server (korrekt) und schickt weiterhin Änderungen ins MQTT
- status bleibt auf offline

Hier ein Logfile, falls das hilft:

```
root@evcc:/etc# /usr/bin/evcc 
[main  ] INFO 2023/08/14 23:10:56 evcc 0.118.11 (224c7d2d)
[main  ] INFO 2023/08/14 23:10:56 using config file: /etc/evcc.yaml
[main  ] INFO 2023/08/14 23:10:56 starting ui and api at :7070
[db    ] INFO 2023/08/14 23:10:56 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2023/08/14 23:10:56 connecting evcc-176797813 at tcp://mosquitto.fritz.box:1883
[mqtt  ] DEBUG 2023/08/14 23:10:56 tcp://mosquitto.fritz.box:1883 connected
[site  ] INFO 2023/08/14 23:10:56 site config:
[site  ] INFO 2023/08/14 23:10:56   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2023/08/14 23:10:56     grid:      power ✓ energy ✓ currents ✗
[lp-1  ] INFO 2023/08/14 23:10:56 loadpoint 1:
[lp-1  ] INFO 2023/08/14 23:10:56   mode:        off
[lp-1  ] INFO 2023/08/14 23:10:56   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/08/14 23:10:56   meters:      charge ✗
[lp-1  ] DEBUG 2023/08/14 23:10:56 phase timer inactive
[lp-1  ] DEBUG 2023/08/14 23:10:56 pv timer inactive
[lp-1  ] DEBUG 2023/08/14 23:10:56 guard timer inactive
[lp-1  ] ERROR 2023/08/14 23:10:56 charger: dial tcp 192.168.0.8:502: connect: network is unreachable
[site  ] DEBUG 2023/08/14 23:10:56 ----
[lp-1  ] DEBUG 2023/08/14 23:10:56 charge power: 0W
[site  ] ERROR 2023/08/14 23:10:56 grid meter: read failed: dial tcp: lookup rs485.fritz.box on [fd00::7642:7fff:fe40:b887]:53: no such host
[site  ] ERROR 2023/08/14 23:10:56 grid meter: read failed: dial tcp: lookup rs485.fritz.box on [fd00::7642:7fff:fe40:b887]:53: no such host

Hier wird jetzt der MQTT Server neu gestartet

[mqtt  ] ERROR 2023/08/14 23:11:01 tcp://mosquitto.fritz.box:1883 connection lost: EOF
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 connected
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/loadpoints/1/vehicle/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/loadpoints/1/minSoc/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/loadpoints/1/targetSoc/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/loadpoints/1/phases/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/site/residualPower/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/site/smartCostLimit/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/loadpoints/1/disableThreshold/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/site/bufferSoc/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/loadpoints/1/mode/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/loadpoints/1/enableThreshold/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/loadpoints/1/targetTime/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/loadpoints/1/minCurrent/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/loadpoints/1/maxCurrent/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/site/prioritySoc/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/site/bufferStartSoc/set
[mqtt  ] DEBUG 2023/08/14 23:11:02 tcp://mosquitto.fritz.box:1883 subscribe evcctest/loadpoints/1/targetEnergy/set

```

tobox, 2023-08-14:
In provider/mqtt/client.go wird folgendes gesetzt:

`        options.SetAutoReconnect(true)
`

Der status wird nur in Run in server/mqtt.go gesetzt, welches vermutlich nur einmal beim Starten in runRoot aus cmd/root.go aufgerufen wird. 

Ich denke, das erklärt das Verhalten einigermaßen.

jowi24, 2023-08-15:
@tobox  Danke für Deine Mühe, das Vorgehen ausführlich zu dokumentieren!

andig, 2023-08-15:
V.a. zeigt das Logfile jetzt überhaupt das Mqtt Verhalten anstatt nur irrelevante Fehlermeldungen 😬

andig, 2023-08-15:
Test wäre gut :)

tobox, 2023-08-16:
```
evcc/status online

---> hier MQTT Server neu gestartet

evcc/status offline
evcc/status online
```

Sieht schon mal nicht ganz falsch aus!

closed 2023-08-16 by andig