evcc-io/evcc #9204: EVCC Minimalkonfig startet nicht

lmuser22, 2023-08-01:
Bin nun ganz neu dabei und möchte nur den SOC von meinem Hyundai (Genesis) über EVCC auslesen damit ich den Wert an den MQTT Server übermitteln kann.

Leider startet der EVCC Server nicht, ich habe diese Konfig:

network:
  
  # schema is the HTTP schema
  
  # setting to `https` does not enable https, it only changes the way URLs are generated
  
  schema: http
  
  # host is the hostname or IP address
  
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  
  host: 192.168.1.240
  
  # port is the listening port for UI and api
  
  # evcc will listen on all available interfaces
  
  port: 7070
  

  
interval: 10s # control cycle interval
  

  

  
# log settings
  
log: info
  
levels:
  
  site: debug
  
  lp-1: debug
  
  lp-2: debug
  

  

  
vehicles:
  
- name: GV60Sport
  
  type: template
  
  template: hyundai
  
  title: # GV60
  
  user: *****  
  password: *****  
  vin: *****  
  capacity: 77 # Akkukapazität in kWh (Optional)
  
  language: en

VolkerK62, 2023-08-01:
Das ist keine Konfig. Das ist weniger als nichts.

Du brauchst mindestens 
- ein meter (grid oder pv). Geht auch als Dummy.
- einen charger. Geht auch als Dummy
- einen Loadpoint mit diesem charger
Ob das mit dem Hyundai Template funktioniert, ist zweifelhaft.

Wenn es dir nur um den Soc geht, gibt es auch andere Möglichkeiten. Z.B. NodeRed

lmuser22, 2023-08-01:
hmmm verstehe, dann muss ich zumindest Dummy eintragen haben. 

Was muss ich einstellen damit der "Dummy Meter" funktioniert?

meters:
- name: gridmeter
  type: custom
  power:
    source: script
    cmd: /bin/sh -c 'echo -1000'
- name: pvmeter
  type: custom
  power:
    source: script
    cmd: /bin/sh -c 'echo 9000'

VolkerK62, 2023-08-01:
schau mal hier, hilft evtl weiter https://docs.evcc.io/docs/guides/meters#ich-bekomme-meine-anlageauto-in-einiger-zeit-kann-ich-schon-evcc-irgendwie-ohne-echten-z%C3%A4hlerauto-ausprobieren

lmuser22, 2023-08-01:
Genesis nutzt offenbar eine eigene API :/
https://developers.genesis.com/web/v1/genesis/main#a