evcc-io/evcc #9209: Guest Vehicle vs API

weymann, 2023-08-01:
### Describe the bug

I'm using evcc only using API. When starting from scratch every time _guest vehicle_ is selected and is reporting demo values. There's no way to figure out which vehicle is currently selected 

Response from api/state with one configured vehicle **_EQA_**
`{
    "result": {
        "auth": {
            "vehicles": {
                "EQA": {
                    "uri": "oauth/vehicles/1",
                    "authenticated": true
                }
            }
        },
        "availableVersion": "0.118.11",
        "battery": [
            {
                "power": -3009,
                "soc": 48,
                "capacity": 0
            }
        ],
        "batteryCapacity": 0,
        "batteryConfigured": true,
        "batteryPower": -3009,
        "batterySoc": 48,
        "bufferSoc": 66,
        "currency": "EUR",
        "gridConfigured": true,
        "gridPower": -3067,
        "homePower": 0,
        "loadpoints": [
            {
                "chargeCurrent": 6,
                "chargeDuration": 340000000000,
                "chargePower": 0,
                "chargeRemainingDuration": 0,
                "chargeRemainingEnergy": 0,
                "chargeTotalImport": 2.6178000000000003,
                "chargedEnergy": 4.650000000000265,
                "charging": false,
                "connected": true,
                "connectedDuration": 0,
                "enabled": true,
                "maxCurrent": 16,
                "minCurrent": 6,
                "minSoc": 0,
                "mode": "minpv",
                "phaseAction": "inactive",
                "phaseRemaining": 0,
                "phasesActive": 3,
                "phasesConfigured": null,
                "phasesEnabled": 3,
                "planActive": false,
                "pvAction": "inactive",
                "pvRemaining": 0,
                "targetEnergy": 0,
                "targetSoc": 100,
                "targetTimeActive": false,
                "title": "Wallbox",
                "vehicleCapacity": 0,
                "vehicleDetectionActive": false,
                "vehicleFeatureOffline": false,
                "vehicleIcon": "",
                "vehicleOdometer": 0,
                "vehiclePresent": false,
                "vehicleRange": 0,
                "vehicleSoc": 0,
                "vehicleTargetSoc": 0,
                "vehicleTitle": ""
            }
        ],
        "prioritySoc": 50,
        "pv": [
            {
                "power": 4283
            }
        ],
        "pvConfigured": true,
        "pvPower": 4283,
        "residualPower": 400,
        "savingsAmount": 0.07971264730182069,
        "savingsEffectivePrice": 0.084632111770681,
        "savingsGridCharged": 0.0020921257655440715,
        "savingsSelfConsumptionCharged": 0.2506686982656515,
        "savingsSelfConsumptionPercent": 99.17229033669955,
        "savingsSince": 1679956217,
        "savingsTotalCharged": 0.2527608240311956,
        "siteTitle": "Zuhause",
        "tariffFeedIn": 0.081999995,
        "tariffGrid": 0.4,
        "vehicles": [
            "EQA"
        ],
        "version": "0.111.1"
    }
}`

### Steps to reproduce

1. Start evvc from scratch
2. How to evaluate which car is currently selected via API - **_not UI_**

### Configuration details

```yaml
chargers:
- type: template
  template: shelly
  host: 192.168.1.106
  name: Wallbox

vehicles:
- name: merc
  type: mercedes
  title: EQA
  capacity: 66
  clientId: 6935a6bd-8712-4703-be10-43f963044423
  clientSecret: ryRxQxTxWzgUnZHxnOQsTJVuaKpqaevYnwDkNNtnDLMAPdpezwMMdVBBbnZJaCUF
  vin: W1N2437011J016433

loadpoints:
- title: Wallbox
  charger: Wallbox
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.111.1 ff

StefanSchoof, 2023-08-01:
Is the vehicle selection working and you see the right car in the UI?

Can you use a default vehicle? https://docs.evcc.io/docs/reference/configuration/loadpoints#vehicle

`"vehicleTitle": ""` sounds like your selection is not working. Take a look at https://docs.evcc.io/docs/guides/vehicles/#erkennung-des-fahrzeugs-an-der-wallbox

andig, 2023-08-01:
Version severely outdated.

closed 2023-08-01 by andig