evcc-io/evcc #9224: Fehlerhafte Währung in Ladevorgang

nelli81, 2023-08-03:
### Describe the bug

Obwohl als Währung in evcc.yaml CHF hinterlegt ist, wird in den Details eines Ladevorgangs € / ct angegeben.

In der Hauptansicht und der Übersicht der Ladevorgänge ist die Währung korrekt.

### Steps to reproduce

1. Anpassen der Währung > CHF

### Configuration details

```yaml
tariffs:
  currency: CHF
  grid:
    type: fixed
    price: 0.26
    zones:
      - days: Mo-Sa
        hours: 6:00-22:00
        price: 0.29
  feedin:
    type: fixed
    price: 0.13
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.118.11

nelli81, 2023-08-03:
Übersicht Ladevorgänge:
[image]

Detailansicht Ladevorgang:
[image]

naltatis, 2023-08-03:
@nelli81 Danke für den Issue. Ist eine Kleinigkeit.

PR #9230 (2023-08-03): Fix: Session details currency formatting
fixes #9224

Hier wurde die Währung nicht korrekt durchgereicht.

[image]
closed 2023-08-03 by naltatis