evcc-io/evcc #9226: Ladevorgänge fehlende Werte ab August

nuiler, 2023-08-03:
### Describe the bug

Es fehlen die Werte Ladepunkt und Fahrzeug in der Übersicht. 

### Steps to reproduce

1.
[image]

2.
[image]

3.
...

### Configuration details

```yaml
**
```

### Log details

```text
**
```

### What type of operating system are you running?

Linux

### Version

0.118.11

andig, 2023-08-03:
Bei mir fehlen die Spalten wo alle Werte identisch sind- i.S. der Vereinfachung.

naltatis, 2023-08-03:
Ja, so ist das. Für Nutzer die nur ein Auto und/oder einen Ladepunkt haben ist diese Spalte nicht wichtig. Daher haben wir entschieden sie nur anzuzeigen wenns auch unterschiedliche Werte gibt. Das wird "pro Monat" entschieden damit Fahrzeugwechsel oder Umbenennung von Ladepunkten nicht Auswirkung auf alle zukünftigen Ansichten hat.

Ist also erwartetes Verhalten. In der Detailansicht eines Ladevorgangs werden Ladepunkt und Fahrzeug übrigens immer angezeigt.

closed 2023-08-03 by naltatis