evcc-io/evcc #9238: Fehlermeldung beim Installieren auf Windows

myfirstPV, 2023-08-04:
### Describe the bug

Es wurde das File "evcc_0.118.11_windows-amd64" heruntergeladen entpackt und evcc.exe Anwendung gestartet.

Kurz nach dem Start kommt eine Fehlermeldung:

This is a command line tool.

You need to open cmd.exe and run it from there.

Ich habe bis dato keine Installation von evcc getätigt. 

### Steps to reproduce

1

### Configuration details

```yaml
1
```

### Log details

```text
1
```

### What type of operating system are you running?

Windows

### Version

evcc_0.118.11_windows-amd64

TobiasHuber1980, 2023-08-04:
Na du musst die evcc.exe ja auch im Terminal/Eingeabeaufforderung starten. Also nicht doppelklicken, sondern im Terminal/Powershell/cmd.exe den Befehl eintippen.

StefanSchoof, 2023-08-04:
Am besten folgst du der Anleitung unter https://docs.evcc.io/docs/installation/manual