evcc-io/evcc #9242: "phases: 0" Konfiguration für "loadpoints" wird in Kombination mit der Keba Wallbox nicht berücksichtigt. --> "phases not configured, assuming 3p"

Farrell89, 2023-08-04:
### Describe the bug

Die "phases: 0" Konfiguration im Abschnitt "loadpoints" wird in Verbindung mit einer Keba P30 mit S10 Phasenumschalter (Eigenbau) nicht berücksichtigt.

[lp-1  ] WARN 2023/08/04 12:48:18 phases not configured, assuming 3p

Die Konfiguration mit "phases: 1" oder "phases: 3" führt nicht zu dem Eintrag im Log.
Ebenfalls wird im UI auch das Menü für die Phasenumschaltung neben dem Loadpoint nicht angezeigt.

### Steps to reproduce

1. Keba P30 C-Series und S10 (Eigenbau) verbaut, DIP-Schalter entsprechend gesetzt.
2. Firmwareupdate auf Version P30 v 3.10.42 durchgeführt.
3. EVCC Konfiguration wie unten vorgenommen und neu gestartet. 
4. Log anschauen
...

### Configuration details

```yaml
chargers:
- name: Keba
  type: template
  template: keba-modbus
  modbus: tcpip
  id: 255
  host: 192.168.X.X
  port: 502

loadpoints:
- title: Garage
  charger: Keba
  mode: pv
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
  phases: 0
  enable:
    delay: 5m
    threshold: 0
  disable:
    delay: 5m
    threshold: 0
  guardduration: 15m
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
```

### Log details

```text
2023/08/04 12:49:29,stdout,[lp-1  ] DEBUG 2023/08/04 12:49:29 pv charge current: 0A = 6A + -15.4A (3540W @ 1p)

2023/08/04 12:49:29,stdout,[lp-1  ] DEBUG 2023/08/04 12:49:29 vehicle range: 267km

2023/08/04 12:49:29,stdout,[lp-1  ] DEBUG 2023/08/04 12:49:29 vehicle soc limit: 70%

2023/08/04 12:49:29,stdout,[lp-1  ] DEBUG 2023/08/04 12:49:29 vehicle soc: 52%

2023/08/04 12:49:29,stdout,[lp-1  ] DEBUG 2023/08/04 12:49:29 soc estimated: 52.02% (vehicle: 52.00%)

2023/08/04 12:49:29,stdout,[lp-1  ] DEBUG 2023/08/04 12:49:29 charger status: C

2023/08/04 12:49:29,stdout,[lp-1  ] DEBUG 2023/08/04 12:49:29 charge total import: 2511.308kWh

2023/08/04 12:49:29,stdout,[lp-1  ] DEBUG 2023/08/04 12:49:29 detected active phases: 1p

[... 325 lines cut ...]

2023/08/04 12:48:18,stdout,[site  ] INFO 2023/08/04 12:48:18     grid:      power ✓ energy ✗ currents ✗

2023/08/04 12:48:18,stdout,[site  ] INFO 2023/08/04 12:48:18   meters:      grid ✓ pv ✓ battery ✓

2023/08/04 12:48:18,stdout,[site  ] INFO 2023/08/04 12:48:18 site config:

2023/08/04 12:48:18,stdout,"[lp-1  ] WARN 2023/08/04 12:48:18 phases not configured, assuming 3p
"
2023/08/04 12:48:15,stdout,[db    ] INFO 2023/08/04 12:48:15 using sqlite database: /.evcc/evcc.db

2023/08/04 12:48:14,stdout,[main  ] INFO 2023/08/04 12:48:14 starting ui and api at :7070

2023/08/04 12:48:14,stdout,[main  ] INFO 2023/08/04 12:48:14 using config file: /etc/evcc.yaml

2023/08/04 12:48:14,stdout,[main  ] INFO 2023/08/04 12:48:14 evcc 0.118.11
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.118.11

andig, 2023-08-04:
> charger:     power ✓ energy ✓ currents ✓ phases ✗

Da wird keine Phasenumschaltung erkannt, vmtl. ist sie in der Keba nicht eingerichtet.

closed 2023-08-04 by andig

Farrell89, 2023-08-04:
Bei der Keba P30 c-series wird die Konfiguration über die DIP-Switches vorgenommen, wurde wie im Handbuch beschrieben durchgeführt. Das ist leider nicht wie bei der P30 x-series wo das über's Webinterface möglich ist.
Wie wird die Phasenumschaltung bei der Keba P30 c-series mit installiertem S10 in EVCC erkannt? Bzw. ist es möglich die Phasenumschaltung per Kommando für Testzwecke manuell auszulösen?

andig, 2023-08-04:
Erkannt wird es über die phasesource.

0x3d13f, 2023-08-04:
Ich habe das mit meiner c-Series noch nicht mit dem Nachbau-S10 am Laufen (wg. Lieferzeit der Komponenten), aber habe durch Tests nachgestellt, dass das Häkchen bei "phases" kommt, sobald man gemäss dem UDP Programming Guide "x2src" auf 3 (Modbus) stellt. X2 ist ja dann die Klemme ab der P30, wo das Signalkabel zum Phasentrennschalter weg geht.
Siehe https://www.keba.com/download/x/4a925c4c61/kecontactp30udp_pgen.pdf Abschnitt 3.2.14.

0x3d13f, 2023-08-04:
die UDP Kommandos kann man absetzen, wenn man sich mittels socat verbindet:
`socat - udp:192.168.x.x:7090,sourceport=7090,reuseaddr`
mit dem "x2" Befehl müsste sich auch die Phasenumschaltung auslösen lassen, allerdings muss dafür dann x2src auf "4" (UDP) gestellt werden

Farrell89, 2023-08-04:
@0x3d13f, tausend Dank, das wars!
Ich habe jetzt mittels dem Programm Packet Sender "x2src 3" per UDP an die Keba gesendet und nach einem Restart von EVCC sieht es jetzt aus wie es soll.
Da soll mal einer drauf kommen. Weder im aktuellen Handbuch der Wallbox noch des Phasenumschalters habe ich ein Hinweis dazu gesehen.

hardl1, 2024-04-13:
@Farrell89 , Könntest du mir bitte erklären, was ich im Packet Sender eingeben muß um "x2src 3" an die Keba zu senden?
Davon habe ich keine Ahnung und möchte nichts verkehrt machen.

snapscanii, 2025-07-19:
Hallo zusammen, habe eine Keba P30c mit einem S10 installiert. Bekomme aber leider die Umstellung mit UDP nicht hin. Kann mir einer für einen Laien erklären, mit welchem Programm ich welche Befehle senden muss. Werde leider aus allen Beschreibungen nicht schlau.

Vielen Dank dafür.