evcc-io/evcc #9245: Remove yaml settings adjustable in UI

andig, 2023-08-04:
- phases
- minsoc
- targetsoc
- prioritySoc
- bufferSoc
- bufferStartSoc

VolkerK62, 2023-08-04:
phases?
Die Phases-Anzeige im UI hat man mWn nur, wenn der Charger 1p3p kann.
Wenn die Wallbox nicht die Phasenspannungen ausgibt, wird das auch nicht automatisch eingestellt bzw. erst beim Laden, durch die Ströme.

naltatis, 2023-08-04:
Bei `targetsoc` bin ich mir noch nicht 100%ig sicher ob der schon gehen kann. @VolkerK62 hatte das in der Doku auch schon angemerkt.

Die UI speichert momentan immer das zuletzt eingestellte SoC Limit. So wie ein Tesla das bspw. auch macht (via App / im Fahrzeug). Mit `onidentify.targetsoc` ist es heute möglich das Limit beim Anstecken immer wieder auf einen festen Wert zurückzusetzen. Das geht über die UI nicht. Gerade vor dem Hintergrund von #9235 und #5492 fänd ich aber gut, wenn wir nicht zu viele SoCs einstellbar machen und dadurch zusätzliche Komplexität schaffen.

Vorschlag: wir werfen `onidentify.targetsoc` erstmal raus. Wenn wir überzeugendes Feedback bekommen, dass es zwingend notwendig ist diese Zurücksetzen-Funktion weiter zu erhalten und auch die geplanten Ladeplanungsfeature (s.o.) nicht ausreichend sind. Dann bauen wir das Setting in die UI ein (Ladeplanung > Ankunft > unter Min. Ladung).

VolkerK62, 2023-08-04:
Vom Prinzip her (nur umgekehrt) habe ich es heute "leidvoll" erfahren. Ich habe vor kurzem in der yaml `targetsoc: 80` eingestellt. Zuletzt habe ich bis 90% geladen (im UI eingestellt). Heute wollte ich voll laden und habe mich gewundert, dass nicht geladen wird.
Ein Blick in UI zeige sofort, warum. Beim anstecken wurde natürlich 80% eingestellt. Hab nicht mehr an diese Grenze gedacht, die stand vorher immer auf 100.
Will sagen:
Wenn immer der zuletzt eingestellte Wert gültig ist, geht das fire (anstecken) and forget verloren.
Insofern wäre so ein Default-Wert bestimmt nicht schlecht. Den kennt man eher, als den zuletzt eingestellten.

andig, 2023-08-04:
> Vorschlag: wir werfen onidentify.targetsoc erstmal raus. 

Den im Loadpoint würde ich aber schon auch rauswerfen!

naltatis, 2023-08-04:
Ja, der im Loadpoint kann auf jeden Fall weg.

naltatis, 2023-09-17:
Doku für prioritysoc, buffersoc, bufferstartsoc angepasst und bebildert:
https://github.com/evcc-io/docs/pull/444

[image]

micklafisch, 2023-09-28:
wird die Änderung über die UI sofort und dauerhaft wirksam oder wird ein reboot benötigt?

VolkerK62, 2023-09-28:
wirkt sofort und dauerhaft

andig, 2023-10-23:
First part done, what about

- residualPower
- maxGridSupplyWhileBatteryCharging
- smartCostLimit

closed 2023-12-02 by andig