evcc-io/evcc #9265: Fix FoxESS RS485 templates

premultiply, 2023-08-06:
H1 oder H3?
Vermutlich passt das Template nicht.
Hat wohl noch nie einer mit deiner Variante getestet und die vorhandene Fremd-Doku ist auch sehr uneindeutig hinsichtlich des Zielsystems.

_Originally posted by @premultiply in https://github.com/evcc-io/evcc/discussions/9255#discussioncomment-6647541_

premultiply, 2023-08-06:
https://github.com/StealthChesnut/HA-FoxESS-Modbus/blob/main/custom_components/HA-FoxESS-Modbus/modbus-H3-USB.yaml

premultiply, 2023-08-26:
https://github.com/evcc-io/evcc/discussions/7699#discussioncomment-6830976

FerdiM83, 2023-08-27:
Danke hab ich gesehen aber ich hab ja keine Ethernet Adapter sondern usb... muss dann warten bis die Template gefixt sind

premultiply, 2023-08-27:
https://github.com/nathanmarlor/foxess_modbus/blob/main/custom_components/foxess_modbus/entities/entity_descriptions.py

FerdiM83, 2023-09-14:
Also der H3 funktioniert noch immer nicht per USB RS485...

Hab als Template fox-ess-h3 hinterlegt 

[image]

premultiply, 2023-09-14:
Hast du dir das selbst compiliert?
Ansonsten wäre es hilfreich erstmal mindestens auf das nächste Nightly Build zu warten.

FerdiM83, 2023-09-14:
Mein Fehler... dachte das ist dann sofort online...

Auf welches Template muss ich dann setzen? fox-ess-rs485 oder fox-ess-h3?

Zweite Frage... modbus: rs485 oder rs485serial?

Ich warte dann mal auf die nächste Version und solange Entschuldigung für meine Ungeduld bin ganz neu in dem Thema hier... top Arbeit die ihr bisher gemacht habt 👍👍

FerdiM83, 2023-09-17:
So update installiert und mit dem h3 Template sieht es gut aus! 
Saubere Arbeit!!
Werd gleich mal nen Ladevorgang starten sollte aber funktionieren.

PR #9561 (2023-09-14): FoxESS: cleanup templates
Fixes #9265