evcc-io/evcc #9272: Fix Enphase templates

premultiply, 2023-08-06:
Mittlerweile habe ich vom Enphase eine Antwort auf meine Anfrage, wie man die Batterie Leistung (Laden/Entladen), einbinden kann.
Bei der Antwort habe ich dann lediglich einen Verweis auf den link: https://enphase.com/download/iq-gateway-access-using-local-apis-or-local-ui-token-based-authentication-tech-brief bekommen. 

Ich habe dann alle api Abfragen ausprobiert und nach meiner Interpretation bekommt man alle benötigen Informationen aus der https://{IQ Gateway_ip}/ivp/livedata/status / [192.168.178.133_ivp_livedata_status.pdf](https://github.com/evcc-io/evcc/files/12269198/192.168.178.133_ivp_livedata_status.pdf)

"soc": 43,                                 SoC of all batteries in %

"pv": {
"agg_p_mw": 1720049,            solar production in milliwatt

"storage": {
"agg_p_mw": -768001,            batterie power in milliwatt (minus means charging/plus means discharging)

"grid": {
"agg_p_mw": 53184,               grid power in milliwatt (minus means grid feeding/plus grid consumption)

"load": {
"agg_p_mw": 1005232,           house consumption in milliwatt

Kann das jemand bestätigen? 
Beste Grüße!

_Originally posted by @JAK0721 in https://github.com/evcc-io/evcc/discussions/8462#discussioncomment-6649944_

JAK0721, 2023-08-08:
@premultiply, danke dass Du meine Recherche aufgegriffen hast und für die Implementierung eingeplant hast. Soll ich die Implementierung via einer Nightly Variante ausprobieren bevor es dann in offizielles Release einfließt. 
Danke und beste Grüße!

premultiply, 2023-08-10:
```json
{
  "connection": {
    "mqtt_state": "connected",
    "prov_state": "configured",
    "auth_state": "ok",
    "sc_stream": "enabled",
    "sc_debug": "disabled"
  },
  "meters": {
    "last_update": 1691316998,
    "soc": 43,
    "main_relay_state": 1,
    "gen_relay_state": 5,
    "backup_bat_mode": 1,
    "backup_soc": 0,
    "is_split_phase": 0,
    "phase_count": 3,
    "enc_agg_soc": 43,
    "enc_agg_energy": 6055,
    "acb_agg_soc": 0,
    "acb_agg_energy": 0,
    "pv": {
      "agg_p_mw": 1720049,
      "agg_s_mva": 1747399,
      "agg_p_ph_a_mw": 568391,
      "agg_p_ph_b_mw": 573709,
      "agg_p_ph_c_mw": 577948,
      "agg_s_ph_a_mva": 573909,
      "agg_s_ph_b_mva": 584637,
      "agg_s_ph_c_mva": 588852
    },
    "storage": {
      "agg_p_mw": -768001,
      "agg_s_mva": -768000,
      "agg_p_ph_a_mw": -202000,
      "agg_p_ph_b_mw": -285000,
      "agg_p_ph_c_mw": -281000,
      "agg_s_ph_a_mva": -202000,
      "agg_s_ph_b_mva": -285000,
      "agg_s_ph_c_mva": -281000
    },
    "grid": {
      "agg_p_mw": 53184,
      "agg_s_mva": 272152,
      "agg_p_ph_a_mw": 73271,
      "agg_p_ph_b_mw": -130567,
      "agg_p_ph_c_mw": 110479,
      "agg_s_ph_a_mva": 208541,
      "agg_s_ph_b_mva": -184970,
      "agg_s_ph_c_mva": 248579
    },
    "load": {
      "agg_p_mw": 1005232,
      "agg_s_mva": 1251551,
      "agg_p_ph_a_mw": 439662,
      "agg_p_ph_b_mw": 158142,
      "agg_p_ph_c_mw": 407427,
      "agg_s_ph_a_mva": 580450,
      "agg_s_ph_b_mva": 114667,
      "agg_s_ph_c_mva": 556431
    },
    "generator": {
      "agg_p_mw": 0,
      "agg_s_mva": 0,
      "agg_p_ph_a_mw": 0,
      "agg_p_ph_b_mw": 0,
      "agg_p_ph_c_mw": 0,
      "agg_s_ph_a_mva": 0,
      "agg_s_ph_b_mva": 0,
      "agg_s_ph_c_mva": 0
    }
  },
  "tasks": {
    "task_id": -1984413882,
    "timestamp": 1691008746
  },
  "counters": {
    "main_CfgLoad": 4,
    "main_CfgChanged": 4,
    "main_CfgNotFound": 28,
    "main_taskUpdate": 535,
    "MqttClient_publish": 243824,
    "MqttClient_respond": 2780,
    "MqttClient_msgarrvd": 1391,
    "MqttClient_reconnect": 2,
    "MqttClient_create": 6733,
    "MqttClient_setCallbacks": 6733,
    "MqttClient_connect": 6733,
    "MqttClient_connect_err": 6629,
    "MqttClient_connect_Err": 6629,
    "MqttClient_subscribe": 104,
    "SSL_Keys_Create": 6733,
    "sc_hdlDataPub": 244540,
    "sc_SendStreamCtrl": 1045,
    "sc_SendDemandRspCtrl": 1,
    "rest_Status": 36986
  },
  "dry_contacts": {
    " \u0011": {
      "dry_contact_id": " \u0011",
      "dry_contact_type": " \u0011",
      "dry_contact_load_name": "\u0006",
      "dry_contact_status": 3050996
    },
    "": {
      "dry_contact_id": "",
      "dry_contact_type": "",
      "dry_contact_load_name": "",
      "dry_contact_status": 3050996
    }
  }
}
```

PR #9332 (2023-08-11): Enphase: fix soc and power
Fixes #9272

@JAK0721 Bitte testen.
closed 2023-08-11 by premultiply