evcc-io/evcc #9274: vehicle soc: get \"..." unknown or invalid refresh  token.\

Hartung42, 2023-08-06:
### Describe the bug

vehicle soc: get \"..." unknown or invalid refresh  token.\

Diese Fehlermeldung bekomme ich beim Laden eines Taycans. Bei anderen Fahrzeugen funktioniert alles. Das Merkwürdige daran: auch beim Taycan funktioniert es nach einem Neustart des EVCC für einige Tage, dann plötzlich nicht mehr. Hat jemand ähnliche Erfahrung und ev. eine Lösung? 
PS: ich verwende die V0.118.11 auf einem Blackberry V3.

### Steps to reproduce

[image]

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 5s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3NTU4NzQ3ODQsImlhdCI6MTY2MTI2Njc4NCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IkhhcnR1bmc0MiJ9.4cMSvqNVMc3fD70MDVEMSqJ0DRFAGB3CF93VFJX5_xk

# log settings
log: info
levels:
  site: debug
  lp-1: debug

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: solaredge-pv-meter
    type: template
    template: solaredge-inverter
    usage: pv      
    modbus: tcpip
    id: 1
    host: 192.168.178.46 # Hostname
    port: 1502 # Port
    timeout: 10s # Optional 
  - name: sonnenbatterie-grid-meter
    type: template
    template: sonnenbatterie-eco10
    usage: grid
    host: 192.168.178.48 # IP-Adresse oder Hostname
    port: 8080 # Port # Optional 
  - name: sonnenbatterie-battery-meter
    type: template
    template: sonnenbatterie-eco10
    usage: battery
    host: 192.168.178.48 # IP-Adresse oder Hostname
    port: 8080 # Port # Optional 

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers
chargers:
  - name: webasto
    type: template
    template: webasto-next
    host: 192.168.178.44 # IP-Adresse oder Hostname
    port: 502 # Port # Optional 

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: porsche
    title: Taycan # Optional
    type: template
    template: porsche #porsche
    user: xxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: xxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 90 # Akku-Kapazität in kWh # Optional 
    minSoC: 20 # charge to at least 20% independent of charge mode
    targetSoC: 90 # limit charge to 90%
  - name: mini
    title: Mini # Optional
    type: template
    template: mini
    user: xxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: xxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 30 # Akku-Kapazität in kWh # Optional 
    minSoC: 20 # charge to at least 20% independent of charge mode
    targetSoC: 90 # limit charge to 90%
    cache: 1m
  - name: fiat
    title: Knubbel # Wird in der Benutzeroberfläche angezeigt # Optional
    # type: custom
    type: template
    template: offline
    capacity: 20 # Akku-Kapazität in kWh # Optional 
    # soc:
    #   source: js
    #   script: 0;

# site describes the EVU connection, PV and home battery
site:
  title: Wilstermann # display name for UI
  meters:
    grid: sonnenbatterie-grid-meter # grid meter
    pvs: solaredge-pv-meter # list of pv inverters/ meters
    battery: sonnenbatterie-battery-meter # battery meter
  residualPower: 250 # 250W Hausverbrauch
  prioritySoC: 50 # give home battery priority up to this soc (empty to disable)
  bufferSoC: 80 # ignore home battery discharge above soc (empty to disable)

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: Ladeverhalten # display name for UI
    charger: webasto # charger
    # meter: charge # charge meter # optional
    # vehicle: porsche # standard vehicle
    vehicles: # use if multiple vehicles allowed to charge on this loadpoint
     - porsche
     - mini
     - fiat
    mode: minpv # default charge mode (off, now, minpv, pv)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    soc:
      # polling defines usage of the vehicle APIs
      # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
      # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
      poll:
        # poll mode defines under which condition the vehicle API is called:
        #   charging: update vehicle ONLY when charging (this is the recommended default)
        #   connected: update vehicle when connected (not only charging), interval defines how often
        #   always: always update vehicle regardless of connection state, interval defines how often (only supported for single vehicle)
        mode: charging
        # poll interval defines how often the vehicle API may be polled if NOT charging
        interval: 60m
      min: 0 # immediately charge to 0% regardless of mode unless "off" (disabled)
      target: 90 # always charge to 90%
      estimate: false # set true to interpolate between api updates
    phases: 3 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    enable: # pv mode enable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 250 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: 250 # maximum import power (W)
    guardDuration: 1m # switch charger contactor not more often than this (default 5m)
    minCurrent: 8 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)

# tariffs are the fixed or variable tariffs
# cheap (tibber/awattar) can be used to define a tariff rate considered cheap enough for charging
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: fixed
    price: 0.37 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # EUR/kWh

# mqtt message broker
mqtt:
  # broker: localhost:1883
  # topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# eebus credentials
eebus:
  # uri: # :4712
  # interfaces: # limit eebus to specific network interfaces
  # - en0
  # certificate: # local signed certificate, required, can be generated via `evcc eebus-cert`
  #   public: # public key
  #   private: # private key

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: SoC updated
      msg: Battery charged to ${vehicleSoC:%.0f}%
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
```

### Log details

```text
siehe steps to Reproduce
```

### What type of operating system are you running?

Linux

### Version

V0.118.11

gengenbe, 2023-08-06:
Habe momentan das selbe Problem. Nach einem restart von evcc werden die Daten des Taycan wieder ca. 1 Tag erfolgreich ausgelesen.

andig, 2023-08-07:
Bekanntes Problem mit dem Porsche IDP- das Refresh Token lässt sich nur 1x verwenden, Grund unklar.

Sonnensucher, 2023-08-09:
habe das gleiche Problem, aber auch Neustart von EVCC hilft nicht. Weiterhin wird beim anstecken das Fahrzeug erkannt aber der SOC nicht ausgelesen aufgrund des Fehler. Wenn ich aber über die Komandozeile `evcc vehicle` aufrufe spukt er mir alle Daten korrekt aus, und erst dann nach `stop` und `start` wird der SOC richtig übernommen.

Hartung42, 2023-08-09:
Hallo Andi,Danke für die Info. Gibt es irgendeine Idee wie wir hier weiter kommen? Mit was müsste ich ev auf Porsche zugehen um das Problem zu lösen? Mit freundlichen Grüßen - Hartung Wilstermann_________________________________________________Hartung WilstermannHöhenweg 2672631 ***@***.***: +49-7127-953854Mobil: +49-172-3574150

MarkusGH, 2023-08-10:
Es wäre schön, wenn bis zur Behebung des Problems nicht versucht würde das ursprüngliche Token zu aktualisieren. 
Dann hielte das Token wenigstens 6-12 Stunden. Man könnte einfach evcc neu starten und hätte dann 6 Stunden Ruhe.

andig, 2023-09-09:
Ich hoffe es passt jetzt. Am Ende war die Lösung doch überraschend einfach. Abgucken konnte ich bei der Easee wo @GrimmiMeloni das Problem analog gelöst hatte.

closed 2023-09-09 by andig

Hartung42, 2023-09-09:
😳 🎊🤩🙏Woran lag es?Mit freundlichen Grüßen - Hartung Wilstermann_________________________________________________Hartung WilstermannHöhenweg 2672631 ***@***.***: +49-7127-953854Mobil: +49-172-3574150

andig, 2023-09-09:
Fehlende Inspiration ;)

Hartung42, 2023-09-09:
Na Hauptsache es tut jetzt ☺️Danke nochmal für Deine Inspiration und vor allem den super Support 🎊Mit freundlichen Grüßen - Hartung Wilstermann_________________________________________________Hartung WilstermannHöhenweg 2672631 ***@***.***: +49-7127-953854Mobil: +49-172-3574150

hegi2k, 2023-09-09:
Also ich hatte vor 2 Wochen die version geupdatet und es ging trotzdem nur paar stunden.

gengenbe, 2023-09-13:
Funktioniert das Thema bei euch? Ich habe immer noch das Problem, dass die Fahrzeugdaten nur eine begrenzte Zeit ausgelesen werden. Dann bekomm ich wieder die Fehlermeldung wie oben beschrieben. Nach einem EVCC Neustart funktioniert es dann wieder für begrenzte Zeit.

MarkusGH, 2023-09-13:
Funktioniert bei mir mit Nightly. Ein Release mit der Änderung gibt es ja noch nicht.

Sonnensucher, 2023-09-13:
Denke es liegt hier an Porsche, dass sie es nicht hinbekommen oder hinbekommen wollen ... aber Software und Porsche kennt man ja ... Schlüssel im Fahrzeug nicht erkannt, Automatisches anmelden mit der Porsche-ID bzgl. Fahrzeugeinstellungen klappt nicht, persönliche Einstellungen weg ... die Liste ist lang. Warum sollte es hier dann klappen ;-(

MarkusGH, 2023-09-13:
> Denke es liegt hier an Porsche, dass sie es nicht hinbekommen oder hinbekommen wollen

Ja - die können das nicht und haben auch generell keine Lust. Bei mir haben die verbrannte Erde hinterlassen.

Das Problem hier allerdings wurde von andig gelöst - bei mir funktioniert es jedenfalls.