evcc-io/evcc #9282: cFos Power Brain Wallbox Solar (inklusive 1P/3P)

aBua-MUC, 2023-08-07:
CFos hat neben der normalen Power Brain Wallbox, auch die Power Brain **Solar** rausgebracht. 
Diese unterstütz auch 1P/3P. Darf ich euch bitten, diese Funktion zu implementieren?

Siehe auch [Produktbeschreibung Website](https://shop.cfos-emobility.de/products/cfos-power-brain-wallbox-solar-22kw)
und in der [Dokumentation](https://www.cfos-emobility.de/de/cfos-charging-manager/documentation/phase-switching.htm).

andig, 2023-08-08:
Vor dem Hintergrund macht es keinen Sinn da etwas zu implementieren. Schade.

closed 2023-08-08 by andig

aBua-MUC, 2023-08-10:
Die WB unterstützt von der Hardware sogar automatisch umschalten. Alles eine Sache der Firmware, in der anktuelle Nightly ist die umschaltung implementiert. Die Info habe ich vom Support:

"...die "Solar"-Wallboxen ([11kW](https://shop.cfos-emobility.de/products/cfos-power-brain-wallbox-solar)/[22kw](https://shop.cfos-emobility.de/products/cfos-power-brain-wallbox-solar-22kw)) haben die für Phasenumschaltung erforderliche Ausstattung.
Im Moment wird mit der "normalen" Firmware die Phasenumschaltung nur im nicht eingesteckten Zustand durchgeführt. Mit der aktuellen "nightly"-Version der Firmware ist auch eine manuelle Umschaltung im eingesteckten Zustand möglich. Eine automatische Umschaltung befindet sich in Vorbereitung."

andig, 2023-08-10:
Das klingt doch gut. Ich vermute, das wäre laut https://www.cfos-emobility.de/en/cfos-power-brain/modbus-registers.htm vmtl.

relay_select | 8087
-- | --

Fehlt noch eine Idee, wie man rausfinden kann ob 1p3p verbaut ist.

aBua-MUC, 2023-08-11:
Ich würde gerne zur Ideenfindung beitragen. Allerdings verstehe ich nicht den Kontext zu 1p3p, da die Hardware in der "Solarvariante" verbaut ist. Darf ich dich bitten, dass du mir deinen Gedanken kurz erklärst...

damienmc3, 2024-03-05:
Hallo zusammen,

kann mir jemand Screenshots aus cFos und evcc erstellen, wie die Konfiguration aussehen muss, damit die automatische Umschaltung funktioniert?

Besten Dank schon mal.