evcc-io/evcc #9284: Evcc or charger doesn't adjust the power according to the comsumption

Jo-sebove, 2023-08-07:
### Describe the bug

Hello, even if the Evcc get data from the solar array and the meter it doesn't adjust the power to reduce the charging power.

### Steps to reproduce

plug in the car for charging

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xxxxxxxxxxxx

interval: 10s # control cycle interval

sponsortoken: xxxxxxxxxxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.1.86  
  name: grid1
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.1.86  
  name: pv2
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.1.85  
  name: pv3

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.1.13  
  name: wallbox5

vehicles:
- type: template
  template: bmw 
  title: i3  
  user: joseph.fochesato@hotmail.fr  
  password: jujpy8-dykbeD-turseg  
  vin: Wby7z41040vd82368  
  capacity: 27  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: pv  
  minSoc: 100  
  targetSoc: 100  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev4

loadpoints:
- title: go-eCharger
  charger: wallbox5
  vehicle: ev4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    - pv3
```

### Log details

```text
==> Successfully stopped `evcc` (label: homebrew.mxcl.evcc)
josephfochesato@Mac-mini-de-Joseph ~ % evcc --log debug
[main  ] INFO 2023/08/07 18:57:56 evcc 0.118.11
[main  ] INFO 2023/08/07 18:57:56 using config file: /Users/josephfochesato/evcc.yaml
[main  ] INFO 2023/08/07 18:57:56 starting ui and api at :7070
[db    ] INFO 2023/08/07 18:57:57 using sqlite database: /Users/josephfochesato/.evcc/evcc.db
[lp-1  ] WARN 2023/08/07 18:57:58 locking phase config to 3p for switchable charger
[site  ] INFO 2023/08/07 18:57:58 site config:
[site  ] INFO 2023/08/07 18:57:58   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/08/07 18:57:58     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/08/07 18:57:58     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/08/07 18:57:58     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/08/07 18:57:58   vehicles:
[site  ] INFO 2023/08/07 18:57:58     vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/08/07 18:57:58 loadpoint 1:
[lp-1  ] INFO 2023/08/07 18:57:58   mode:        pv
[lp-1  ] INFO 2023/08/07 18:57:58   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/08/07 18:57:58   meters:      charge ✓
[lp-1  ] INFO 2023/08/07 18:57:58     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/08/07 18:57:58 phase timer inactive
[lp-1  ] DEBUG 2023/08/07 18:57:58 pv timer inactive
[lp-1  ] DEBUG 2023/08/07 18:57:58 guard timer inactive
[lp-1  ] INFO 2023/08/07 18:57:58 vehicle updated: unknown -> i3
[lp-1  ] DEBUG 2023/08/07 18:57:58 set charge mode: pv
[lp-1  ] DEBUG 2023/08/07 18:57:58 set min current: 6
[lp-1  ] DEBUG 2023/08/07 18:57:58 set max current: 16
[lp-1  ] DEBUG 2023/08/07 18:57:58 set min soc: 100
[lp-1  ] DEBUG 2023/08/07 18:57:58 set target soc: 100
[lp-1  ] DEBUG 2023/08/07 18:57:59 max charge current: 6A
[site  ] WARN 2023/08/07 18:57:59 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/08/07 18:57:59 ----
[lp-1  ] DEBUG 2023/08/07 18:57:59 charge power: 10620W
[site  ] DEBUG 2023/08/07 18:57:59 pv power: 6583W
[site  ] DEBUG 2023/08/07 18:57:59 grid power: 5821W
[site  ] DEBUG 2023/08/07 18:57:59 site power: 5821W
[lp-1  ] DEBUG 2023/08/07 18:57:59 vehicle odometer: 52120km
[lp-1  ] DEBUG 2023/08/07 18:57:59 charge currents: [15.4 15.4 14.5]A
[lp-1  ] DEBUG 2023/08/07 18:57:59 charge total import: 2639.605kWh
[lp-1  ] DEBUG 2023/08/07 18:57:59 charger status: C
[lp-1  ] INFO 2023/08/07 18:57:59 car connected
[lp-1  ] DEBUG 2023/08/07 18:57:59 pv timer elapse
[lp-1  ] DEBUG 2023/08/07 18:57:59 pv timer inactive
[lp-1  ] DEBUG 2023/08/07 18:57:59 charger: guard elapse
[lp-1  ] DEBUG 2023/08/07 18:57:59 guard timer inactive
[lp-1  ] DEBUG 2023/08/07 18:57:59 charge total import: 2639.605kWh
[lp-1  ] INFO 2023/08/07 18:57:59 start charging ->
[lp-1  ] DEBUG 2023/08/07 18:57:59 wake-up timer: stop
[lp-1  ] DEBUG 2023/08/07 18:57:59 vehicle status: C
[lp-1  ] DEBUG 2023/08/07 18:57:59 vehicle soc: 88%
[lp-1  ] DEBUG 2023/08/07 18:57:59 vehicle range: 189km
[lp-1  ] DEBUG 2023/08/07 18:58:00 switched phases: 3p
[site  ] DEBUG 2023/08/07 18:58:00 ----
[lp-1  ] DEBUG 2023/08/07 18:58:00 charge power: 10620W
[site  ] DEBUG 2023/08/07 18:58:00 pv power: 6543W
[site  ] DEBUG 2023/08/07 18:58:00 grid power: 5853W
[site  ] DEBUG 2023/08/07 18:58:00 site power: 5853W
[lp-1  ] DEBUG 2023/08/07 18:58:00 vehicle odometer: 52120km
[lp-1  ] DEBUG 2023/08/07 18:58:00 charge currents: [15.5 15.4 14.5]A
[lp-1  ] DEBUG 2023/08/07 18:58:00 charge total import: 2639.608kWh
[lp-1  ] DEBUG 2023/08/07 18:58:00 charger status: C
[lp-1  ] DEBUG 2023/08/07 18:58:00 soc estimated: 88.01% (vehicle: 88.00%)
[lp-1  ] DEBUG 2023/08/07 18:58:00 vehicle soc: 88%
[lp-1  ] DEBUG 2023/08/07 18:58:00 vehicle range: 189km
[lp-1  ] DEBUG 2023/08/07 18:58:00 max charge current: 16A
```

### What type of operating system are you running?

Docker container

### Version

0.118.11

StefanSchoof, 2023-08-07:
Is that your real password for your bmw account? If so you should change it now.

andig, 2023-08-07:
>  WARN 2023/08/07 18:57:59 interval <30s can lead to unexpected behavior

Bitte erstmal die Warnungen beseitigen!

closed 2023-08-07 by andig