evcc-io/evcc #9285: Sponsor token not accepted

juehv, 2023-08-07:
### Describe the bug

I use evcc as docker image with the Heidelberg EC Wallbox (so sponsor token required). I had the system running for around half a year and restarts, updates never was an issue. This weekend I had a resetart again and the system is telling me on boot time that I need a sponsor token, even though I didn't change anything (sponsor token is still in place in the config file). As a consequence, the system does not create the wallbox and loadpoint. I tried to stop my sponsorship and recreate it, but thats not possible instantly. Is the issue on my side, or is the issue on your side?

Also: My wallbox is basically useless now. It would be great if EVCC sets the wallbox in some mode where it is allowed to charge. I'm now fiddeling around with the Brick-Loader as I don't want to call an electrician to reconfigure the wallbox :/

Thanks in advance for any hints!

### Steps to reproduce

1. Restart docker container

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: my id

interval: 30s # control cycle interval

sponsortoken: my token

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

tariffs:
  grid:
    type: tibber
    token: "my tibber token" # access token
    homeid: "my home id"
  feedin:
    type: fixed
    price: 0.08
  co2:
    type: grünstromindex
    zip: my zip code

meters:
- type: template
  template: kostal-plenticore
  id: 71
  host: 192.168.100.81
  port: 1502
  usage: pv
  modbus: tcpip
  name: pv1
- type: template
  template: kostal-ksem-inverter
  id: 71
  host: 192.168.100.81
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid1

chargers:
- type: template
  template: heidelberg
  id: 1
  device: /dev/ttyUSB0
  baudrate: 19200
  comset: 8E1
  modbus: rs485serial
  name: wallbox1

loadpoints:
- title: WB Sued
  charger: wallbox1
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
 #enable:
    #threshold: -1000
    #delay: 5m

site:
  title: Home
  meters:
    grid: grid1
    pv: pv1
  residualPower: 0
  smartCostLimit: 0.10
```

### Log details

```text
[main  ] INFO 2023/08/07 19:40:46 evcc 0.118.11
[main  ] INFO 2023/08/07 19:40:46 using config file: /etc/evcc.yaml
[main  ] INFO 2023/08/07 19:40:46 starting ui and api at :7070
[db    ] INFO 2023/08/07 19:40:46 using sqlite database: /root/.evcc/evcc.db
[main  ] FATAL 2023/08/07 19:40:47 cannot create charger 'wallbox1': cannot create charger 'template': cannot create charger 'heidelberg': sponsorship required, see https://github.com/evcc-io/evcc#sponsorship
[main  ] FATAL 2023/08/07 19:40:47 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.118.11

StefanSchoof, 2023-08-07:
Is it working after you create a new sponsor token at https://sponsor.evcc.io/ and put it into your config?

andig, 2023-08-07:
@juehv please contact info@evcc.io

juehv, 2023-08-07:
> Is it working after you create a new sponsor token at https://sponsor.evcc.io/ and put it into your config?

No, I can't create a new one. The page shows the same symptome, stating that I'm not sponsor :(

andig, 2023-08-08:
Fixed- thanks for your patience!

closed 2023-08-08 by andig

andig, 2023-10-12:
Danke für den Hinweis- sollte wieder behoben sein!