evcc-io/evcc #9297: Heidelberg: Add async watchdog timer

premultiply, 2023-08-08:
Was ist denn als `interval` in evcc eingestellt?
Wenn mehr als 15 Sekunden ist klar dass die Boxen ständig die Verbindung verlieren.

_Originally posted by @premultiply in https://github.com/evcc-io/evcc/discussions/3596#discussioncomment-6667429_

premultiply, 2023-08-08:
Wenn das Default interval nicht mehr 10 Sekunden ist passt dies nicht mehr zur Standardkonfiguration der Box mit 15 Sekunden.
Hier ist daher ein asynchroner Watchdog Handler erforderlich.

panzerdev, 2023-08-08:
Wenn sowas im Charger reicht würde ich das in nen PR machen.

```
	go func() {
		ticker := time.NewTicker(5 * time.Second)
		for {
			select {
			case <-ticker.C:
				wb.Status()
			}
		}
	}()
```

andig, 2023-08-08:
for time.Tick() ....

bitte analog der anderen heartbeats- nix neues erfinden ;)

panzerdev, 2023-08-08:
Habe zumindest noch diese Schreibweise entdeckt

`for ; true; <-time.Tick(time.Hour) `

In der Doku steht man solle aufpassen wegen garbage collection da der Ticker drunter nicht abschaltbar ist. Werden also bei den Tick Aufruf im for nicht dauernd neue Ticker ins Nichts gestartet? In der Doku wird der Channel vor dem for initialisiert.

https://pkg.go.dev/time#Tick

andig, 2023-08-08:
Nein…

andig, 2023-08-08:
Jetzt gibts einen weniger: https://github.com/evcc-io/evcc/commit/7a314c79ee6d493049606a8debd992c023567e2f

closed 2023-08-10 by andig