evcc-io/evcc #9302: Easee: Enabled() should consider REASON_FOR_NO_CURRENT

GrimmiMeloni, 2023-08-08:
Ah, ja ok, stimmt. Aber dann müsste (sollte ;)) der Wert von REASON_FOR_NO_CURRENT mit in Betracht gezogen werden.
Ein OpMode von 2 sagt alleine nichts über enabled oder nicht im hier benötigten Sinne aus: OpMode 2 teilt sich in mehrere Zustände auf, z.B. pausiert durch App oder Api (welcher Wert auch immer dann vorliegt, müsstem an rausfinden) - oder aber gestartet, aber noch wartend auf das Auto oder Lastausgleich.

Gemäß [https://developer.easee.cloud/docs/enumerations](https://developer.easee.cloud/docs/enumerations) bedeutet ein Wert von  REASON_NO_FOR_CURRENT zwischen 1 und 6 ein Warten auf das Fahrzeug/Lastausgleich, mit OpMode 2 also klar enabled, die Box würde bzw. wird automatisch wenn es vorbei ist laden.
51, 52, 54, 75 bis 81 sehe ich dann ebenfalls als enabled.
7 bis 30 als Error kann man wohl eher als !enabled betrachten, 50, 53, 55. 56, 57 und 100 ebenfalls.
Wobei wohl kaum alle ReasonForNoCurrent im OpMode 2 vorkommen können.

Evcc müht sich hier redlich ab den Ladevorgang in Schwung zu bringen, das führt aber zu nichts außer einem wiederholten resume_charging mit bekanntem Reset von DYNAMIC_CHARGER_CURRENT auf 32 mit jedem resume_charging.

```
[easee ] TRACE 2023/08/08 10:45:15 POST https://api.easee.com/api/chargers/********/commands/start_charging
[easee ] TRACE 2023/08/08 10:45:16 {"device":"********","commandId":25,"ticks":638270811160060285}
[easee ] TRACE 2023/08/08 10:45:17 ProductUpdate ********: (2023-08-08 08:45:15 +0000 UTC) ObservationID(223) {"Id":330,"MeterValue":2201.984033,"Start":"2023-08-08T08:45:15.000Z","Auth":"userid_xxx","AuthReason":6}
[easee ] TRACE 2023/08/08 10:45:17 ProductUpdate ********: (2023-08-08 08:45:15 +0000 UTC) REASON_FOR_NO_CURRENT 8
[easee ] TRACE 2023/08/08 10:45:17 ProductUpdate ********: (2023-08-08 08:45:15 +0000 UTC) NUMBER_OF_CARS_CONNECTED 1
[easee ] TRACE 2023/08/08 10:45:17 ProductUpdate ********: (2023-08-08 08:45:15 +0000 UTC) NUMBER_OF_CARS_FULLY_CHARGED 1
[easee ] TRACE 2023/08/08 10:45:17 CommandResponse ********: {SerialNumber:******** ID:25 Timestamp:2023-08-08 08:45:16.006028 +0000 UTC DeliveredAt:2023-08-08 08:45:15.476 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638270811160060285}
[easee ] TRACE 2023/08/08 10:45:18 ProductUpdate ********: (2023-08-08 08:45:16 +0000 UTC) REASON_FOR_NO_CURRENT 100
[easee ] TRACE 2023/08/08 10:45:18 ProductUpdate ********: (2023-08-08 08:45:16 +0000 UTC) LEDMODE 21
[easee ] TRACE 2023/08/08 10:45:18 ProductUpdate ********: (2023-08-08 08:45:16 +0000 UTC) REASON_FOR_NO_CURRENT 6
[easee ] TRACE 2023/08/08 10:45:19 ProductUpdate ********: (2023-08-08 08:45:17 +0000 UTC) CHARGER_OP_MODE 2
[lp-1  ] ERROR 2023/08/08 10:45:27 charger enable: timeout

**********************
EINSCHUB: Genau hier sollte idealerweise das hier stehen (von weiter unten kopiert nach dem CHARGER_OP_MODE auf 6 wechselt): EVCC erkennt: Easee ist "enabled" im OpMode 2 mit ReasonForNoCurrent 6 (oder wie eins untendrunter 5) und setzt dynamicChargerCurrent auf den korrekten Wert:

[easee ] TRACE 2023/08/08 10:46:33 POST https://api.easee.com/api/chargers/********/settings
[easee ] TRACE 2023/08/08 10:46:33 {"dynamicChargerCurrent":6}
**********************

[easee ] TRACE 2023/08/08 10:45:27 POST https://api.easee.com/api/chargers/********/commands/resume_charging
[easee ] TRACE 2023/08/08 10:45:28 {"device":"********","commandId":48,"ticks":638270811279507580}
[easee ] TRACE 2023/08/08 10:45:28 CommandResponse ********: {SerialNumber:******** ID:48 Timestamp:2023-08-08 08:45:27.950758 +0000 UTC DeliveredAt:2023-08-08 08:45:27.211 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638270811279507580}
[easee ] TRACE 2023/08/08 10:45:29 ProductUpdate ********: (2023-08-08 08:45:27 +0000 UTC) DYNAMIC_CHARGER_CURRENT 32
[easee ] TRACE 2023/08/08 10:45:29 ProductUpdate ********: (2023-08-08 08:45:27 +0000 UTC) NUMBER_OF_CARS_IN_QUEUE 1
[easee ] TRACE 2023/08/08 10:45:29 ProductUpdate ********: (2023-08-08 08:45:27 +0000 UTC) NUMBER_OF_CARS_FULLY_CHARGED 0
[easee ] TRACE 2023/08/08 10:45:30 ProductUpdate ********: (2023-08-08 08:45:28 +0000 UTC) REASON_FOR_NO_CURRENT 5
[lp-1  ] ERROR 2023/08/08 10:45:38 charger enable: timeout

[easee ] TRACE 2023/08/08 10:45:38 POST https://api.easee.com/api/chargers/********/commands/resume_charging
[easee ] TRACE 2023/08/08 10:45:38 {"device":"********","commandId":48,"ticks":638270811387218590}
[easee ] TRACE 2023/08/08 10:45:39 CommandResponse ********: {SerialNumber:******** ID:48 Timestamp:2023-08-08 08:45:38.721859 +0000 UTC DeliveredAt:2023-08-08 08:45:38.448 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638270811387218590}
[easee ] TRACE 2023/08/08 10:45:40 ProductUpdate ********: (2023-08-08 08:45:38 +0000 UTC) DYNAMIC_CHARGER_CURRENT 32
[lp-1  ] ERROR 2023/08/08 10:45:49 charger enable: timeout

[easee ] TRACE 2023/08/08 10:45:50 POST https://api.easee.com/api/chargers/********/commands/resume_charging
[easee ] TRACE 2023/08/08 10:45:50 {"device":"********","commandId":48,"ticks":638270811504656590}
[easee ] TRACE 2023/08/08 10:45:51 CommandResponse ********: {SerialNumber:******** ID:48 Timestamp:2023-08-08 08:45:50.465659 +0000 UTC DeliveredAt:2023-08-08 08:45:49.781 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638270811504656590}
[easee ] TRACE 2023/08/08 10:45:51 ProductUpdate ********: (2023-08-08 08:45:49 +0000 UTC) DYNAMIC_CHARGER_CURRENT 32
[lp-1  ] ERROR 2023/08/08 10:46:01 charger enable: timeout

[easee ] TRACE 2023/08/08 10:46:02 POST https://api.easee.com/api/chargers/********/commands/resume_charging
[easee ] TRACE 2023/08/08 10:46:02 {"device":"********","commandId":48,"ticks":638270811625963050}
[easee ] TRACE 2023/08/08 10:46:03 CommandResponse ********: {SerialNumber:******** ID:48 Timestamp:2023-08-08 08:46:02.596305 +0000 UTC DeliveredAt:2023-08-08 08:46:01.98 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638270811625963050}
[easee ] TRACE 2023/08/08 10:46:04 ProductUpdate ********: (2023-08-08 08:46:01 +0000 UTC) DYNAMIC_CHARGER_CURRENT 32
[lp-1  ] ERROR 2023/08/08 10:46:13 charger enable: timeout

[easee ] TRACE 2023/08/08 10:46:14 POST https://api.easee.com/api/chargers/********/commands/resume_charging
[easee ] TRACE 2023/08/08 10:46:14 {"device":"********","commandId":48,"ticks":638270811741190580}
[easee ] TRACE 2023/08/08 10:46:14 CommandResponse ********: {SerialNumber:******** ID:48 Timestamp:2023-08-08 08:46:14.119058 +0000 UTC DeliveredAt:2023-08-08 08:46:13.398 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638270811741190580}
[easee ] TRACE 2023/08/08 10:46:15 ProductUpdate ********: (2023-08-08 08:46:13 +0000 UTC) DYNAMIC_CHARGER_CURRENT 32
[lp-1  ] ERROR 2023/08/08 10:46:24 charger enable: timeout

[easee ] TRACE 2023/08/08 10:46:25 POST https://api.easee.com/api/chargers/********/commands/resume_charging
[easee ] TRACE 2023/08/08 10:46:25 {"device":"********","commandId":48,"ticks":638270811854435390}
[easee ] TRACE 2023/08/08 10:46:26 CommandResponse ********: {SerialNumber:******** ID:48 Timestamp:2023-08-08 08:46:25.443539 +0000 UTC DeliveredAt:2023-08-08 08:46:24.749 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638270811854435390}
[easee ] TRACE 2023/08/08 10:46:26 ProductUpdate ********: (2023-08-08 08:46:24 +0000 UTC) DYNAMIC_CHARGER_CURRENT 32
[easee ] TRACE 2023/08/08 10:46:32 ProductUpdate ********: (2023-08-08 08:46:30 +0000 UTC) CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L1 0
[easee ] TRACE 2023/08/08 10:46:32 ProductUpdate ********: (2023-08-08 08:46:30 +0000 UTC) CIRCUIT_TOTAL_ALLOCATED_PHASE_CONDUCTOR_CURRENT_L2 0
[... 3 lines cut ...]
[easee ] TRACE 2023/08/08 10:46:32 ProductUpdate ********: (2023-08-08 08:46:30 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L3 0.014000000432133675
[easee ] TRACE 2023/08/08 10:46:32 ProductUpdate ********: (2023-08-08 08:46:30 +0000 UTC) NUMBER_OF_CARS_CHARGING 1
[easee ] TRACE 2023/08/08 10:46:32 ProductUpdate ********: (2023-08-08 08:46:30 +0000 UTC) NUMBER_OF_CARS_IN_QUEUE 0
[easee ] TRACE 2023/08/08 10:46:32 ProductUpdate ********: (2023-08-08 08:46:31 +0000 UTC) REASON_FOR_NO_CURRENT 77
[easee ] TRACE 2023/08/08 10:46:33 ProductUpdate ********: (2023-08-08 08:46:31 +0000 UTC) CHARGER_OP_MODE 6
[easee ] TRACE 2023/08/08 10:46:33 ProductUpdate ********: (2023-08-08 08:46:31 +0000 UTC) OUTPUT_PHASE 30
[easee ] TRACE 2023/08/08 10:46:33 ProductUpdate ********: (2023-08-08 08:46:31 +0000 UTC) OUTPUT_CURRENT 32
[... cut ...]
[easee ] TRACE 2023/08/08 10:46:34 CommandResponse ********: {SerialNumber:******** ID:48 Timestamp:2023-08-08 08:46:33.798424 +0000 UTC DeliveredAt:2023-08-08 08:46:33.158 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:638270811937984240}
[easee ] TRACE 2023/08/08 10:46:35 ProductUpdate ********: (2023-08-08 08:46:33 +0000 UTC) DYNAMIC_CHARGER_CURRENT 6
[lp-1  ] DEBUG 2023/08/08 10:46:35 charger enable
[lp-1  ] DEBUG 2023/08/08 10:46:35 wake-up timer: start
...

```

_Originally posted by @allcoolusernamesaregone in https://github.com/evcc-io/evcc/discussions/9228#discussioncomment-6669700_

GrimmiMeloni, 2023-08-08:
@allcoolusernamesaregone lass uns das Thema Enabled() / Op_Mode / Reason for no Current getrennt diskutieren.
Ich denke Du hast Recht, allerdings hat es nicht direkt mit der start_charge vs. resume_charge Thematik zu tun.

GrimmiMeloni, 2023-08-08:
@allcoolusernamesaregone Bonus Frage. Im Log Snippet ist leider nicht zu erkennen auf welchem Wert Dynamic Charger Current steht bevor der Start kommt. Start selbst scheint es nicht zu verändern, jedenfalls sehe ich da kein passendes ProductUpdate reinkommen.
Hintergrund der Frage ist, das Enabled() aktuell auch auf diesen Wert schaut, und diesen >0 erwartet. Wäre jetzt gut zu wissen wie sich dieser Wert verhalten hat, als das Auto nicht wollte.

allcoolusernamesaregone, 2023-08-08:
Ich antworte mal auf beide Fragen in einem...

In dieser Minute stand die Easee App auf "In der Warteschlange-Lastausgleich". Warum das dieses mal so lange war - ich weiß es nicht. Ich habe das bei meinen Tests (mit dem Master, meinen eigenen Veränderungen) aber wiederholt gesehen, dass ganz kurz in der App diese Meldung mit Lastausgleich erscheint, bevor der eigentliche Ladevorgang ("3, Charging") mit kurzem Zwischenschritt ("6, Ready to charge") startet. Ist auch m.M.n. ein bekanntes "Problem" (eher Phänomen) der Easee, quer durch's Netz bekannt, z.B. hier https://github.com/evcc-io/evcc/discussions/6160 (ganz anderes Thema und Problem, aber gleicher Status. Hat sich mit Firmwareupdates bei der Easee auch verbessert)

Die Schuld liegt NICHT beim Auto. Das Auto würde gerne schon laden, darf aber noch nicht, bzw. noch kein Saft auf der Leitung - die Relais in der Easee klicken dann erst nach dem Ende dieses Lastausgleichs (stehe bei den Tests an der Box mit Handy und EVCC + Easeee App im Zugriff).
Nur die Easee hat einen früheren Start verhindert bzw. verzögert. 

Da evcc im aktuellen Code Stand in diesem Fall ja annimmt dass der Charger !enabled ist, wird wiederholt "resumed", so weit so klar.

Am Ende hat die Box dann das gemacht, was evcc derzeit unmittelbar erwartet, einen Wechsel in OpMode 3. Die ganzen Versuchen mit resume_charging haben daran nichts verbessert, geändert oder bewirkt. 
start_charging allein reichte hier schon um den ganzen Prozess anzutriggern.

Ich ergänze hier das Log VOR dem start_charging, also seit letzten Ab- und wieder Einstecken des Autos. Es gab kein Update von dynamicChargerCurrent zwischen Anstecken und start_charging - nur das setzen auf 6 durch evcc, welches durch start_charging  in der Easee mit auf 32 setzen per Default ja wieder überschrieben wird.

Nach dem Abstecken wurde um 10:45:03 der Wert 32 - der Default auf dem mit dem Abstecken die Box intern sich ja selbst versieht.

```
[lp-1  ] INFO 2023/08/08 10:45:01 car disconnected
[lp-1  ] INFO 2023/08/08 10:45:01 vehicle updated: e-Golf -> unknown
[lp-1  ] DEBUG 2023/08/08 10:45:01 set charge mode: pv
[lp-1  ] DEBUG 2023/08/08 10:45:01 set min current: 6
[lp-1  ] DEBUG 2023/08/08 10:45:01 set max current: 32
[lp-1  ] DEBUG 2023/08/08 10:45:01 set min soc: 0
[lp-1  ] DEBUG 2023/08/08 10:45:01 set target soc: 100
[lp-1  ] DEBUG 2023/08/08 10:45:01 set priority: 0
[easee ] TRACE 2023/08/08 10:45:02 ProductUpdate EH5JHSMP: (2023-08-08 08:44:59 +0000 UTC) SMART_CHARGING false
[easee ] TRACE 2023/08/08 10:45:02 ProductUpdate EH5JHSMP: (2023-08-08 08:44:59 +0000 UTC) SMART_CHARGING false
[easee ] TRACE 2023/08/08 10:45:03 ProductUpdate EH5JHSMP: (2023-08-08 08:45:01 +0000 UTC) SESSION_ENERGY 0
[easee ] TRACE 2023/08/08 10:45:03 ProductUpdate EH5JHSMP: (2023-08-08 08:45:01 +0000 UTC) LEDMODE 25
[easee ] TRACE 2023/08/08 10:45:03 ProductUpdate EH5JHSMP: (2023-08-08 08:45:01 +0000 UTC) DYNAMIC_CHARGER_CURRENT 32
[easee ] TRACE 2023/08/08 10:45:03 ProductUpdate EH5JHSMP: (2023-08-08 08:45:01 +0000 UTC) PILOT_MODE B
[easee ] TRACE 2023/08/08 10:45:03 ProductUpdate EH5JHSMP: (2023-08-08 08:45:01 +0000 UTC) CHARGER_OP_MODE 7
[easee ] TRACE 2023/08/08 10:45:04 ProductUpdate EH5JHSMP: (2023-08-08 08:45:02 +0000 UTC) REASON_FOR_NO_CURRENT 55
[easee ] TRACE 2023/08/08 10:45:07 ProductUpdate EH5JHSMP: (2023-08-08 08:45:05 +0000 UTC) LOCAL_TX_POWER 20
[easee ] TRACE 2023/08/08 10:45:07 ProductUpdate EH5JHSMP: (2023-08-08 08:45:05 +0000 UTC) LOCAL_NODE_TYPE 1
[easee ] TRACE 2023/08/08 10:45:07 ProductUpdate EH5JHSMP: (2023-08-08 08:45:05 +0000 UTC) LOCAL_RADIO_CHANNEL 0
[easee ] TRACE 2023/08/08 10:45:07 ProductUpdate EH5JHSMP: (2023-08-08 08:45:05 +0000 UTC) LOCAL_SHORT_ADDRESS 0
[easee ] TRACE 2023/08/08 10:45:07 ProductUpdate EH5JHSMP: (2023-08-08 08:45:05 +0000 UTC) LOCAL_PARENT_ADDR_OR_NUM_OF_NODES 0
[site  ] DEBUG 2023/08/08 10:45:12 ----
[lp-1  ] DEBUG 2023/08/08 10:45:12 charge power: 0W
[site  ] DEBUG 2023/08/08 10:45:12 pv power: 9603W
[site  ] DEBUG 2023/08/08 10:45:12 grid power: -9081W
[site  ] DEBUG 2023/08/08 10:45:12 grid powers: [-3009 -3013 -3059]W
[site  ] DEBUG 2023/08/08 10:45:12 grid currents: [-12.9 -12.7 -12.9]A
[site  ] DEBUG 2023/08/08 10:45:12 site power: -8881W
[lp-1  ] DEBUG 2023/08/08 10:45:12 charge currents: [0.011 0.01 0.014]A
[lp-1  ] DEBUG 2023/08/08 10:45:12 charge total import: 2201.920kWh
[easee ] TRACE 2023/08/08 10:45:12 POST https://api.easee.com/api/chargers/********/settings
[easee ] TRACE 2023/08/08 10:45:12 {"smartCharging":true}
--
[]
[lp-1  ] DEBUG 2023/08/08 10:45:12 charger status: B
[lp-1  ] INFO 2023/08/08 10:45:12 car connected
[lp-1  ] DEBUG 2023/08/08 10:45:12 vehicle api refresh
[lp-1  ] DEBUG 2023/08/08 10:45:12 pv timer elapse
[lp-1  ] DEBUG 2023/08/08 10:45:12 pv timer inactive
[lp-1  ] DEBUG 2023/08/08 10:45:12 charger: guard elapse
[lp-1  ] DEBUG 2023/08/08 10:45:12 guard timer inactive
[lp-1  ] DEBUG 2023/08/08 10:45:12 charge total import: 2201.920kWh
[main  ] DEBUG 2023/08/08 10:45:14 vehicle status: B (e-Golf)
[lp-1  ] INFO 2023/08/08 10:45:14 vehicle updated: unknown -> e-Golf
[lp-1  ] DEBUG 2023/08/08 10:45:14 set min soc: 30
[lp-1  ] DEBUG 2023/08/08 10:45:14 set target soc: 85
[lp-1  ] DEBUG 2023/08/08 10:45:14 vehicle soc: 71%
[lp-1  ] DEBUG 2023/08/08 10:45:14 vehicle range: 184km
[lp-1  ] DEBUG 2023/08/08 10:45:14 pv charge current: 38.6A = 0A + 38.6A (-8881W @ 1p)
[lp-1  ] DEBUG 2023/08/08 10:45:14 site power -8881W <= 0W enable threshold
[... cut ...]
[easee ] TRACE 2023/08/08 10:45:15 CommandResponse ********: {SerialNumber:******** ID:48 Timestamp:2023-08-08 08:45:14.229957 +0000 UTC DeliveredAt:2023-08-08 08:45:13.517 +0000 UTC WasAccepted:true ResultCode:0 Comment: Ticks:}
[easee ] TRACE 2023/08/08 10:45:15 ProductUpdate ********: (2023-08-08 08:45:13 +0000 UTC) DYNAMIC_CHARGER_CURRENT 6
[lp-1  ] DEBUG 2023/08/08 10:45:15 max charge current: 6A
[easee ] TRACE 2023/08/08 10:45:15 POST https://api.easee.com/api/chargers/********/settings
[easee ] TRACE 2023/08/08 10:45:15 {"enabled":true}
--
```

GrimmiMeloni, 2023-08-08:
OK. Dann nehmen wir jetzt also erstmal an, daß in CHARGER_OP_MODE 2 auch 32A gesetzt sind, und wir an der Logik bzgl. dieses Werts nichts verändern müssen. 

Es bleibt also der Change hier jetzt `Enabled()` zu verändern, so daß 
`CHARGER_OP_MODE == 2 && REASON_NO_FOR_CURRENT in (0-6, 51, 52, 54, 75-81)` ebenfalls `true` liefern.

PR kommt.

allcoolusernamesaregone, 2023-08-08:
Genau, sehe ich grundsätzlich genau so. 
Wobei die Reasons...Current gerne mal noch von anderen validiert werden sollten - das war jetzt nur meine Einschätzung, und die könnte falsch sein, wie ich nachfolgend ausführe.

Welche Codes überhaupt gut und welche schlecht sind ist die große und gute Frage.

Wobei, ich habe mal mein Log ausgewertet bzgl. Meldungen von OpMode 2 ja schon vom Namen her einfach schon mal mit enabled gleichgesetzt habe, wäre es vielleicht richtig, einfach den Zustand "Easee ist im aktiven Ladevorgang, aber pausiert" = für evcc  !enabled als DIE Ausnahme von OpMode 2 = enabled  zu betrachten:

```
(hoffe die Syntax passt ungefähr..)
func (c *Easee) inEnabledOpMode() bool {
        c.mux.Lock()
        defer c.mux.Unlock()
        return c.opMode == easee.ModeCharging ||
                c.opMode == easee.ModeCompleted ||
-->          (c.opMode == easee.ModeAwaitingStart && !c.reasonForNoCurrent != 52) ||
                c.opMode == easee.ModeReadyToCharge
}

```

Bei den Ladestarts folgen mehrere ReaseonForNoCurrent Codes in wenigen oder gar der gleichen Sekunde: 5, 6, 8 55, 77, 78 und 100. 
Nach meinem Verständnis vom Code bzgl. der Auswertung dieser Updates gewinnt der letzte Code, das wären  5, 6, 77, 78 und 100,
Ggf. muss man sich aber gedanklich davon lösen, dass ein als Error betitelter ReasonForNoCurrent Code ein Fehler ist, bzw. im Sinn der Erkennung von enabled oder nicht im OpCode es nicht relevant ist, dass das als Fehler in der Easee Codeliste bezeichnet wird.
Konkret z.B. 8 "Primary unit has not received current request from secondary unit (car)" - das mag für die Easee ein Fehler zu diesem Zeitpunkt sein, im Rahmen eines Startvorgangs ist das aber nicht relevant bzw. der Code sagt uns eigentlich, dass die Easee eine Anforderung erwartet, und damit sagt die Easee uns eigentlich auch: sie ist enabled.

```
Ladepause:
[easee ] TRACE 2023/08/01 21:04:48 ProductUpdate ******: (2023-08-01 17:15:02 +0000 UTC) CHARGER_OP_MODE 2
[easee ] TRACE 2023/08/01 21:04:48 ProductUpdate ******: (2023-08-01 17:15:00 +0000 UTC) REASON_FOR_NO_CURRENT 52
ODER auch umgekehrt, beides sehr oft vorhanden (Wolken....)
[easee ] TRACE 2023/08/02 17:58:12 ProductUpdate ******: (2023-08-02 15:58:11 +0000 UTC) REASON_FOR_NO_CURRENT 52
[easee ] TRACE 2023/08/02 17:58:15 ProductUpdate ******: (2023-08-02 15:58:14 +0000 UTC) CHARGER_OP_MODE 2

OpMode 2 und ReasonForNoCurrent kann man also sicher mit !enabled gleichsetzen.
Da ich "OpMode2

Ladestarts:

[easee ] TRACE 2023/08/02 17:13:44 ProductUpdate ******: (2023-08-02 15:13:42 +0000 UTC) REASON_FOR_NO_CURRENT 8
[easee ] TRACE 2023/08/02 17:13:45 ProductUpdate ******: (2023-08-02 15:13:43 +0000 UTC) REASON_FOR_NO_CURRENT 6
[easee ] TRACE 2023/08/02 17:13:45 ProductUpdate ******: (2023-08-02 15:13:43 +0000 UTC) REASON_FOR_NO_CURRENT 100
[easee ] TRACE 2023/08/02 17:13:47 ProductUpdate ******: (2023-08-02 15:13:45 +0000 UTC) CHARGER_OP_MODE 2

[easee ] TRACE 2023/08/03 09:20:33 ProductUpdate ******: (2023-08-03 07:20:31 +0000 UTC) REASON_FOR_NO_CURRENT 55
[easee ] TRACE 2023/08/03 09:20:43 ProductUpdate ******: (2023-08-03 07:20:41 +0000 UTC) REASON_FOR_NO_CURRENT 8
[easee ] TRACE 2023/08/03 09:20:43 ProductUpdate ******: (2023-08-03 07:20:41 +0000 UTC) REASON_FOR_NO_CURRENT 100
[easee ] TRACE 2023/08/03 09:20:44 ProductUpdate ******: (2023-08-03 07:20:42 +0000 UTC) REASON_FOR_NO_CURRENT 5
[easee ] TRACE 2023/08/03 09:20:44 ProductUpdate ******: (2023-08-03 07:20:42 +0000 UTC) CHARGER_OP_MODE 2
[easee ] TRACE 2023/08/03 09:20:44 ProductUpdate ******: (2023-08-03 07:20:42 +0000 UTC) REASON_FOR_NO_CURRENT 77

[easee ] TRACE 2023/08/03 14:18:21 ProductUpdate ******: (2023-08-03 12:18:20 +0000 UTC) REASON_FOR_NO_CURRENT 55
[easee ] TRACE 2023/08/03 14:18:28 ProductUpdate ******: (2023-08-03 12:18:26 +0000 UTC) REASON_FOR_NO_CURRENT 8
[easee ] TRACE 2023/08/03 14:18:28 ProductUpdate ******: (2023-08-03 12:18:26 +0000 UTC) REASON_FOR_NO_CURRENT 100
[easee ] TRACE 2023/08/03 14:18:28 ProductUpdate ******: (2023-08-03 12:18:27 +0000 UTC) REASON_FOR_NO_CURRENT 5
[easee ] TRACE 2023/08/03 14:18:28 ProductUpdate ******: (2023-08-03 12:18:27 +0000 UTC) CHARGER_OP_MODE 2

[easee ] TRACE 2023/08/03 14:41:28 ProductUpdate ******: (2023-08-03 12:40:59 +0000 UTC) REASON_FOR_NO_CURRENT 55
[easee ] TRACE 2023/08/03 14:41:58 ProductUpdate ******: (2023-08-03 12:41:56 +0000 UTC) REASON_FOR_NO_CURRENT 8
[easee ] TRACE 2023/08/03 14:41:58 ProductUpdate ******: (2023-08-03 12:41:57 +0000 UTC) REASON_FOR_NO_CURRENT 100
[easee ] TRACE 2023/08/03 14:41:59 ProductUpdate ******: (2023-08-03 12:41:57 +0000 UTC) REASON_FOR_NO_CURRENT 5
[easee ] TRACE 2023/08/03 14:41:59 ProductUpdate ******: (2023-08-03 12:41:58 +0000 UTC) CHARGER_OP_MODE 2

[easee ] TRACE 2023/08/03 16:13:34 ProductUpdate ******: (2023-08-03 14:13:32 +0000 UTC) REASON_FOR_NO_CURRENT 55
[easee ] TRACE 2023/08/03 16:13:41 ProductUpdate ******: (2023-08-03 14:13:39 +0000 UTC) REASON_FOR_NO_CURRENT 8
[easee ] TRACE 2023/08/03 16:13:41 ProductUpdate ******: (2023-08-03 14:13:40 +0000 UTC) REASON_FOR_NO_CURRENT 100
[easee ] TRACE 2023/08/03 16:13:42 ProductUpdate ******: (2023-08-03 14:13:40 +0000 UTC) REASON_FOR_NO_CURRENT 6
[easee ] TRACE 2023/08/03 16:13:42 ProductUpdate ******: (2023-08-03 14:13:41 +0000 UTC) CHARGER_OP_MODE 2

[easee ] TRACE 2023/08/03 16:24:53 ProductUpdate ******: (2023-08-03 14:24:51 +0000 UTC) REASON_FOR_NO_CURRENT 8
[... cut ...]
[easee ] TRACE 2023/08/05 08:44:00 ProductUpdate ******: (2023-08-05 06:43:59 +0000 UTC) CHARGER_OP_MODE 2

[easee ] TRACE 2023/08/05 11:17:16 ProductUpdate ******: (2023-08-05 09:17:15 +0000 UTC) REASON_FOR_NO_CURRENT 8
[easee ] TRACE 2023/08/05 11:17:17 ProductUpdate ******: (2023-08-05 09:17:15 +0000 UTC) REASON_FOR_NO_CURRENT 100
[easee ] TRACE 2023/08/05 11:17:17 ProductUpdate ******: (2023-08-05 09:17:16 +0000 UTC) REASON_FOR_NO_CURRENT 6
[easee ] TRACE 2023/08/05 11:17:19 ProductUpdate ******: (2023-08-05 09:17:17 +0000 UTC) CHARGER_OP_MODE 2

```

GrimmiMeloni, 2023-08-08:
> Also eher umgekehrt die Codes ausschließen, die auf eine gewünschte Pause (und Erwartung !enabled) hindeuten.
> Außer 52 habe ich da keine gesehen. Die anderen, die man mit !enabled verbinden würde (z.B. 55 - Waiting Pending Authorization) haben ein anderen OpMode.

Auch ok. Können wir mal so probieren. Ich habe mir gerade nochmal gezielt das Log vom Lauf angeschaut, der damals zu #9100 geführt hat. Dort sehe ich zuletzt REASON 100, das Problem würde dieser Change also auch sauber abfangen.

GrimmiMeloni, 2023-08-08:
Gerade nochmal drüber nachgedacht. Ich glaube wir können das doch nicht auf "alles außer 52" reduzieren. Mein Beispiel aus der #9100 zeigt ja genau das. Der Charger bleibt mit 100 (also !=52) hängen. Somit müßten wir die 100 zumindest auch ausschließen. Ich glaube Dein initialer Vorschlag alle Reasons die auf Loadbalancing hindeuten zu übernehmen macht mehr Sinn.

allcoolusernamesaregone, 2023-08-08:
Hmm, wo in der #9100 taucht denn der Reason 100 auf?

GrimmiMeloni, 2023-08-09:
Das Log habe ich dort damals nicht gepostet. Da war ein pause_charge in ein Timeout gelaufen, im Anschluß kam Easee komplett durcheinander. :( Am Ende dann dieser Zustand.

```
[easee ] TRACE 2023/07/23 12:40:45 ProductUpdate XXXXXXXX: (2023-07-23 10:40:41 +0000 UTC) DYNAMIC_CHARGER_CURRENT 13
[easee ] TRACE 2023/07/23 12:40:45 ProductUpdate XXXXXXXX: (2023-07-23 10:40:42 +0000 UTC) CHARGER_OP_MODE 2
[easee ] TRACE 2023/07/23 12:40:46 ProductUpdate XXXXXXXX: (2023-07-23 10:40:42 +0000 UTC) REASON_FOR_NO_CURRENT 100
```

Ohne den Change in #9100, ist dies ein Zustand der als Enabled() angesehen wird (opmode 2 + DCC > 0), obwohl die Box ganz klar nicht lädt. Ich habe damals erstmal nur den OP_MODE 2 an sich als Grund ausgemacht, weil dieser für "EVSE suspended" steht. Ich hatte angenommen, daß OP_MODE 2 nur für explizites Suspend (wie ein pause_charge) steht. Das es auch externe Auslöser wie Loadbalancing gibt hatte ich nicht auf der Rechnung.

allcoolusernamesaregone, 2023-08-09:
Ah ok Ja aber das meinte ich mit "Welche Codes überhaupt gut und welche schlecht sind ist die große und gute Frage."

Ich hatte die 100 beim Ladestart.
```
[easee ] TRACE 2023/08/02 17:13:44 ProductUpdate ******: (2023-08-02 15:13:42 +0000 UTC) REASON_FOR_NO_CURRENT 8
[easee ] TRACE 2023/08/02 17:13:45 ProductUpdate ******: (2023-08-02 15:13:43 +0000 UTC) REASON_FOR_NO_CURRENT 6
[easee ] TRACE 2023/08/02 17:13:45 ProductUpdate ******: (2023-08-02 15:13:43 +0000 UTC) REASON_FOR_NO_CURRENT 100
[easee ] TRACE 2023/08/02 17:13:47 ProductUpdate ******: (2023-08-02 15:13:45 +0000 UTC) CHARGER_OP_MODE 2
```
Wie man's hier macht ist es falsch 🙄

Was man aber ggf. auch überlegen könnte: solche Zwitter Codes, die so oder so vorkommen können, einfach zu ignorieren, also den Wert vom Reason in evcc nicht zu aktualisieren. Damit hätte bei mir die 6 "gewonnen" als letzter Wert vor 100.
Bei Dir?

GrimmiMeloni, 2023-08-09:
> Ich hatte die 100 beim Ladestart.

Es könnte auf dem Charger auch die 6 gewesen sein. Beide ProductUpdates tragen den gleichen Timestamp. Leider haben die ProductUpdates nur eine Auflösung von Sekunden, so daß nicht klar ist welcher dieser beiden Werte aktuell ist. evcc verwirft nur **_veraltete_** Nachrichten. Bei Gleichstand wird der interne State überschrieben.

allcoolusernamesaregone, 2023-08-09:
Vielleicht sollte man dann bei Gleichstand die 100 völlig ignorieren als neuen Wert (und ggf. nur loggen, bis man schlauer ist, ob, wie, wann das vorkommt und was man draus machen könnte. Wenn es tatsächlich auch in der Easee der letzte Stand ist, also die 100 ist und bleibt, dann sollte wohl auch früher oder später der Fehler und Schiefstand in EVCC ankommen, durch andere Probleme im Nachgang. Da ist dann eigentlich auch nichts mehr zu retten ;)

GrimmiMeloni, 2023-08-09:
Naja, es passiert ja erstmal nichts "schlimmes" wenn es mal nicht passt. Im Worst Case hätten wir einen Fall ähnlich deinem Log oben. Wir würden dann einen "unnötigen" Resume Aufruf im nächsten Intervall durchführen. Wenn dadurch dann der Reason Code geradegebogen wird passt es ja wieder.

Ich habe einen entsprechenden PR geöffnet (https://github.com/evcc-io/evcc/pull/9314). Ist aber mangels Auto gerade noch nicht getestet. Hinweis: Ich teste das auf Basis von Master - ohne die Changes für den Autostart bei Authentifizierung. Nach Deinem Hinweis ist zumindest klar, daß #9100 über's Ziel hinausgeschossen ist. Daher würde ich dies gerne isoliert anpassen und mit reinnehmen.

allcoolusernamesaregone, 2023-08-09:
Alles klar, ich teste Deine Änderungen aus #9314 mal mit meinem Master (https://github.com/evcc-io/evcc/tree/e6b88f7b5839428272d306b75235fedf19ed6b56) von letzter Woche (ich merge das mal bei mir rein) und teste mit dem Autostart bei Authentifizierung.. :)

andig, 2023-08-09:
Ich habe das Gefühl, dass wir uns hier völlig im Wald verrennen. evcc erwartet, eine  Box ein/aus schalten zu können. Das klappt bei der Easee nur bedingt weil sie ein Eigenleben hat.

Der Ansatz hier scheint zu sein, immer mehr Parameter hinzu zu ziehen um zu ermitteln, ob die Easee eigentlich abgeschaltet sein müsste. Hilft uns das? Wäre es nicht alternativ einfacher, uns

- zu merken ob sie an oder aus ist (und falls kein Match nachzusteuern) oder
- konsequent die Authentifizierung zum abschalten zu nutzen

allcoolusernamesaregone, 2023-08-09:
Also das Ein- und Ausschalten, egal ob start/resume oder pause/stop klappt ziemlich gut, eigentlich kann man sich auf die Umsetzung entsprechender Befehle an die API verlassen, auch wenn es synchron kein Feedback gibt, und asynchron die 
Statusmeldungen etwas schwieriger zu verarbeiten sind wie man sieht. 

Evcc versucht aber derzeit aus den Productupdates eine (Erfolgs-)Rückmeldung und einen Status abzuleiten, und das passt soweit auch, bis auf den OpMode 2, der alleine nicht aussagekräftig ist. 
Aber wenn man einen Status ableiten will, dann müsste man es zur Vermeidung von Warnungen und Fehlern in der Oberfläche aber auch bestmöglich vollständig machen. Und ich behaupte, mit der Prüfung des Reasons im OpMode 2 ist die Sache dann an der Stelle auch sauber.

Alternativ könnte auch nur mit start_charging und stop_charging (hebt die Autorisierung auf) gearbeitet werden, und sich darauf verlassen, dass die API das schon richtig macht.
Und ggf. über den tatsächlichen Strom auf den 3 Phasen ableiten, ob oder ob nicht geladen wird, anstatt auf Productupdates zu warten. Wobei hier eher Minuten statt Sekunden anzusetzen sind, gerade beim Ladestart bzw. nach einer Pause.
Eben wegen OpMode 2 und dem Loadbalancing. 

Bei meiner Easee im Leerlauf, kein Fahrzeug angeschlossen - == 0 wäre hier nicht erfolgreich...
   "circuitTotalPhaseConductorCurrentL1":0.010999999940395355,
   "circuitTotalPhaseConductorCurrentL2":0.008999999612569809,
   "circuitTotalPhaseConductorCurrentL3":0.014000000432133675,

PS: start/stop führt allerdings zu einem deutlich hörbaren beepbeep Ton bei jedem Ladestart nach Unterbrechnung durch start_charging ;)

andig, 2023-08-09:
> Evcc versucht aber derzeit aus den Productupdates eine (Erfolgs-)Rückmeldung und einen Status abzuleiten

Geht es hier um `Status()` oder- wie der Titel sagt- um `Enabled()`?

allcoolusernamesaregone, 2023-08-09:
> Geht es hier um `Status()` oder- wie der Titel sagt- um `Enabled()`?

sorry, ich wusste nicht, dass Status anderweitig belegt ist in evcc.. ich meinte den "Enabled Status". Es geht um die Frage (wirksam, effektiv) enabled oder nicht, das wird derzeit abgeleitet aus den Productupdates (soweit ich den Code und Architektur von evcc verstehe)..

allcoolusernamesaregone, 2023-08-09:
> Also das Ein- und Ausschalten, egal ob start/resume oder pause/stop klappt ziemlich gut, eigentlich kann man sich auf die Umsetzung entsprechender Befehle an die API verlassen, auch wenn es synchron [...]

Ich hatte auch mal vor längerer Zeit und vor einer PV Anlage aus Spaß an der Freude ein wenig rumprogrammiert um die Easee zu steuern - und bin da diesen Weg gegangen, Vertrauen in die Umsetzung der Befehle, und lediglich Auswerten der direkten synchronen Rückmeldungen / Antworten der API bzw. HTTP Codes.

Das ist wie die ganze Easee Api jetzt nicht so ganz konsistent und einheitlich, mal gibt's einen 200 oder aber 202, mal ein Json Array, mal nur ein Json Object. Mal gibt's alle Infos direkt, mal muss man die "Command Info" abrufen mit Command-ID und Tick. Mal ist ein boolean im Json, mal der HTTP Code.
Aber wenn man alle Varianten mal durch hat geht's eigentlich 😂 - und dann man weiß eigentlich genau, ob der Befehl an die Api ok war und angenommen wurde, oder nicht. 

[Easee.java.txt](https://github.com/evcc-io/evcc/files/12306222/Easee.java.txt)

GrimmiMeloni, 2023-08-10:
Die Implementierung Enabled() ist seit jeher nicht korrekt. Mein vermeintlicher Fix in #9100 ist aber falsch. Egal was wir hier sonst noch diskutieren, zumindest **_irgendeine_** Korrektur an Enabled() müssen wir vornehmen.

Zum Rest bzgl. der gesteigerten Komplexität durch die Einbeziehung der asynchronen Antworten, nochmal kurz zusammen gefasst warum wir die Änderungen / Ergänzungen der letzten Zeit eingebaut haben.

- Dynamic Charger Update direkt nach pause/resume, damit nicht im minA Szenario (6A) mit 32A geladen wird. Fehlerbild: PV Überschuss wird ignoriert
- Warten auf Product Update für OP_MODE nach pause/resume, weil sonst eine Race Condition zwischen pause/resume Auftritt und ein deaktivierter Charger als aktiv angesehen wird. Fehlerbild: Ladung startet nicht. 
- Warten auf Product Update für Dynamic Charger Upgrade nach pause/resume, weil sonst eine Race Condition zwischen dem folgenden evcc Update auf 6A und dem Easee internen Update auf 32A auftritt. Wenn 32A gewinnt Fehlerbild: PV Überschuss wird ignoriert
- Warten auf Product Update nach Dynamic Charger Upgrade, weil sonst nicht garantiert ist, das die 6A angenommen wurden. Bei Timeout oder sonstigem Fehler, Fehlerbild: PV Überschuss wird ignoriert. 

Der ignorierte PV Überschuss eine ganze blöde Sache. Der Strom bleibt so lange auf maxA kleben, bis sich der PV Überschuss im Vergleich zum Wert der beim Ladestart gemessenen wurde veränderte. 
Dieses Problem ist mit der Einführung von Charger.CurrentGetter() und der Anpassung am Loadpoint (siehe nightly) vom Tisch. Man könnte hier sicherlich nochmal optimieren, und vereinfachen. Aber das möchte ich ehrlich gesagt erst machen, wenn wir die Changes im nightly released haben. Da die waits für die Product Updates auch bereits gebaut sind und funktionieren sehe ich keinen akuten Bedarf das zu verändern.

Lasst uns doch daher bitte erstmal bei Enabled() bleiben. Danke! 🙏🏻

allcoolusernamesaregone, 2023-08-10:
> * Warten auf Product Update für Dynamic Charger Upgrade nach pause/resume, weil sonst eine Race Condition zwischen dem folgenden evcc Update auf 6A und dem Easee internen Update auf 32A auftritt. Wenn 32A gewinnt Fehlerbild: PV Überschuss wird ignoriert

Aber könnte man an dieser Stelle nicht bei einem empfangenen Update auf 32A den von evcc bestimmten Wert drüber bügeln, also der Easee sagen "nein, bitte X A"? Also immer, wenn ein Productupdate einen abweichenden Wert mitteilt, den Sollwert senden?
Dann wird zwar kurz dem (nur ersten?) Start mit 32 A geladen, aber das macht die Easee ja sowieso?

Und: sind die letzten beiden der 4 Punkte nicht im Ergebnis das gleiche., bzw. wie vorstehend geschrieben zu erkennen und lösen`?

andig, 2023-08-10:
> Die Implementierung in Enabled() ist seit jeher nicht korrekt. 

Meine Frage ist ganz einfach: macht die Semantik bei Easee irgendeinen Sinn? Kann man die ein- und aus schalten? Falls nein können wir uns auch einfach den internen Status von evcc merken und gut ist. Falls Status=C ist "disabled" offensichtlich fehlgeschlagen und muss erneut vorgenommen werden.

Und fertig ists?

GrimmiMeloni, 2023-08-10:
> Aber könnte man an dieser Stelle nicht bei einem empfangenen Update auf 32A den von evcc bestimmten Wert drüber bügeln, also der Easee sagen "nein, bitte X A"? Also immer, wenn ein Productupdate einen abweichenden Wert mitteilt, den Sollwert senden?

Ja, genau das macht der angesprochene Fix im Nightly.

> Dann wird zwar kurz dem (nur ersten?) Start mit 32 A geladen, aber das macht die Easee ja sowieso?

Jepp. Das lässt sich nicht verhindern.

>Falls Status=C ist "disabled" offensichtlich fehlgeschlagen und muss erneut vorgenommen werden.
>
>Und fertig ists?

Ja - oder mal nochmal ganz anders gefragt. Wir haben dieses Issue eröffnet, weil @allcoolusernamesaregone beobachtet hat, wie wiederholt `resume_charge` aufgerufen wurde - bis es dann letztendlich geklappt hat. Das ist meiner Ansicht nach eigentlich das gleiche was Andig vorschlägt - wir probieren es halt in jedem Intervall, so lange bis es klappt. 

@allcoolusernamesaregone gab es bei diesem ursprünglichen Fall eigentlich einen "Nachteil"? Mal abgesehen von den wiederkehrenden Aufrufen von `resume_charge` hat doch letztendlich alles funktioniert?

allcoolusernamesaregone, 2023-08-10:
Ja der Nachteil hier war: alle 10 Sekunden eine Error Meldung in der GUI mit Charger enable timeout, und danach dann glaube ich auch ein charger out of sync exp. disabled got enabled (oder umgekehrt). Vielleicht war's auch die Meldung charger logic error: disabled but charging.

Grundsätzlich bin ich schon auch der Meinung, dass man einfach die API aufrufen könnte im Vertrauen, dass die schon macht, was ihr gesagt wurde. Und nicht weiter auf Antworten warten bzw. diese auswerten. 

Allerdings müsste man hier trotzdem dann idealerweise die Besonderheiten der Easee berücksichtigen wie Ladestart mit maximaler Leistung, Ignorieren von entsprechenden Runterstufungen von dynamicChargerCurrent solange im Lastausgleich, außerdem die automatische Freischaltung von Boxen mit Autorisierung unterstützen bzw. Usern als Empfehlung dies zu nutzen, um Netzbezug (und weitere Folgen, Stichwort Sunny Home und Abschaltung von Überschussverbrauchern) zu vermeiden bzw. für einen allgemein geregelteren Ladestart. Analog Clever PV.

Denke man könnte auch folgendes Vorgehen anwenden.

Enable(true):
- Falls überhaupt Fahrzeug verbunden
- API start_charging falls Autorisierung nötig (in OpMode 7), sonst API resume_charging (falls in OpMode 2 (ist er nach pause_charging))
- falls API Aufruf ok (Rückmeldung HTTP Code oder im Antwort Json oder in dem Tick):
   - Variable c.isEnabled = true 
   - (asynchron? Geht das?) warten bis Productupdate auf OpMode 3 wechselt, wenn OpMode nicht schon 3 ist, dann max current setzen auf den richtigen Wert (falls nötig bzw. nicht Mode Max gewählt). Timeout hier mind. 1 Minute statt 10s, Lastausgleich kann dauern...  Nach Ablauf Timeout Fehlermeldung und c.isEnabled = false. Falls OpMode von 7
   - return "ok"
- falls API Aufruf nicht ok  
   - return "err"

Enable(false):
- API pause_charging
- falls API Aufruf ok (wie oben)
     c.isEnabled = false
     return "ok"
- sonst
    return "err"

enabled()
- return c.isEnabled

ProductUpdate von 
- maxChargerCurrent: wenn der Wert lt. dem Update abweicht von dem Soll wert: neu setzen
- OpMode: auf 
  - 1: Disconnected -> c.isEnabled = false
  - 2: Awaiting Start -> nicht eindeutig, nichts tun
  - 3: Charging -> c.isEnabled = true - auch wenn das unerwartet kommt (da Auto neu angeschlossen und bei deaktivierter Autorisierung gleich das Laden gestartet wird, KEINEN charger out of sync melden sondern die Tatsache antizipieren. Kenne mich mit dem Rest in evcc aber so gar nicht aus (Loadpoint, Site, ...). Stattdessen: max current neu setzen auf richtigen Wert nach Modus
  - 4: Completed -> c.isEnabled = true
  - 5: Error -> c.isEnbaled = false
  - 6: Ready to charge: c.isEnabled = true. Dürfte bei Boxen ohne Autorisierung ja immer unmittelbar vor 3 kommen
  - 7; Awaiting Authentication: max current setzen auf richtigen Wert
  - 8: De-authenticatin: c.isEnbaled = false

Weitere Methoden?

Wobei ich jetzt mal noch ein paar Tage meine Anpassungen am Master bzw. dem PR #9314 testen werde.

GrimmiMeloni, 2023-08-10:
> Denke man könnte auch folgendes Vorgehen anwenden. [...]

Das beschriebene Verhalten entspricht dem aktuellen Stand - mit Ausnahme des 60s (statt 10s) Timeout und der Unterscheidung start/resume_charge - nur das halt ohne einen fixen state wie `c.isEnabled` gearbeitet wird. 

Ich habe heute bei einem Start mit dem unveränderten Nightly übrigens laut Easee App auch im "Lastausgleich" gehangen. Soweit ich es überblicken konnte war dies der Zeitraum, während Charge mode noch off war und ich auf die Identifizierung des Fahrzeugs gewartet habe (also Dynamic Charger Current 0 war).

> Ja der Nachteil hier war: alle 10 Sekunden eine Error Meldung in der GUI mit Charger enable timeout, und danach dann glaube ich auch ein charger out of sync exp. disabled got enabled (oder umgekehrt). Vielleicht war's auch die Meldung charger logic error: disabled but charging.

OK, das mit den Fehlern in der GUI ist sicherlich nicht so schön - zumindest nicht wenn die immer vorkommen. Soweit ich es gerade überblicke hat der aktuelle Nightly zumindest noch diesen Schönheitsfehler, dass immer 1 mal der Fehler geworfen wird. Ich schaue nochmal genau drauf.

GrimmiMeloni, 2023-08-10:
> Meine Frage ist ganz einfach: macht die Semantik bei Easee irgendeinen Sinn? Kann man die ein- und aus schalten?

Oh, ich glaube da hast Du jetzt was gesagt! Das Setting gibt es, und ich habe den Code Teil auch schon 100 mal gesehen. Ich glaube aber, da ist ein Bug wie wir das benutzen. Wir nutzen es nur zum (einmaligen) Enablen. Der Code liesst sich, als sollte auch ein disable() darüber abgewickelt werden, aber das kann nie vorkommen.

Hier:

https://github.com/evcc-io/evcc/blob/682ab053adefc3f9e0cc136db28cafbe81bd35ad/charger/easee.go#L390

enable ist der gewünschte Zustand in den geschaltet werden soll. Wenn dieser false ist, wird auch enableRequired immer false. Somit wird der darauffolgende Code Block für im Falle von false niemals durchlaufen. Wenn überhaupt setzt evcc nur einmal am Charger den Wert auf true, und fasst Ihn danach nie wieder an.

Wenn wir das verändern und auch über diesen Weg disablen hätten wir zumindest einen enabled/disabled state, der ohne OP_CODE auskommt, und aus Enabled() zurückgegeben werden könnte.

Was noch unklar ist: Wie würde sich das auf authentifizierende Charger auswirken? Können diese im Zustand "disabled" trotzdem noch mit RFID re-aktiviert werden?

premultiply, 2023-08-10:
Enable(true) bedeutet dass eine Ladefreigabe seitens evcc vorliegt. Es DARF (seitens evcc) geladen werden.
Enable(false) bedeutet hingegen dass seitens evcc keine Ladefreigabe vorliegt und der Ladepunkt daher keinesfalls aktiviert (=Strom zum Fahrzeug an) sein/werden darf.

Enabled() fragt vom Ladepunkt ab ob er die Anweisung und den Willen (insbesondere nicht laden zu dürfen) seitens evcc klar und deutlich verstanden hat.

Widerspricht sich hier der rückgemeldete Zustand mit dem von evcc klar geäusserten Willen gibt es den Fehler "charger out of sync".

GrimmiMeloni, 2023-08-11:
Kurz zum enabled setting auf dem Charger: Leider nicht wirklich praktikabel für den evcc usecase. Das schaltet den Charger komplett ab. Das führt unter anderem dazu daß das Auto nicht von Status A nach B wechselt - somit ist dann auch die Fahrzeugerkennung über die Hersteller API nicht mehr verfügbar.

Ich schaue mir nochmal an ob die Definition von Premultiply hilft. Hier liegt der Fokus ja besonders auf "nicht laden dürfen". Bisher war für die Easee Enabled() anders rum gebaut, so daß es den aktivierten Zustand bestätigt. Vielleicht hilft es stattdessen sicher zu stellen, daß deaktiviert ist. Da ist ggf. auch die Liste der möglichen Zustände kleiner...? (op_mode 0,1, und (2 + Reason 55||100 ....?).

allcoolusernamesaregone, 2023-08-11:
> Ich schaue mir nochmal an ob die Definition von Premultiply hilft. Hier liegt der Fokus ja besonders auf "nicht laden dürfen". Bisher war für die Easee Enabled() anders rum gebaut, so daß es den [...]

Ja, man könnte auch bei 

- "darf laden" bei Enable(true): resume_charging und/oder start_charging, wenn man das automatische Autorisieren unterstützen will bzw. zum kontrollierten Start bzgl. Netzbezug unterstützen will, aufrufen.   Und dann einfach Annehmen, dass der weitere Status, welcher auch immer, ok ist - ohne jegliches weiteres warten / prüfen. Wenn die Easee dann sofort oder aber auch erst später startet ist das für "es darf" ja ok.

- "muss deaktiviert sein" bei Enable(false): pause_charging und tatsächlich abwarten, dass der OpMode nach 2 (m.E.n. in Kombination nur mit Reason 52 bei einer Ladeunterbrechnung! 100 habe ich immer nur beim Ladestart gesehen) wechselt (ggf. auch 4 = Completed, falls das Auto beim Ladestop zufällig sagt "reicht mir", ebenso 1 = Disconnected  -  und/oder die tatsächliche Stromstärke < 1A wird auf jeder Phase. Die Ladestärke auswerten wäre im Zweifel sogar das eindeutigere und einfachere Merkmal, dass die Ladung gestoppt wurde.

premultiply, 2023-08-11:
Gedacht ist das als wallboxseitiges Register dass über Enable() gesetzt und via Enabled() abgefragt wird.
Irgendwelche Messwerte und Statuszustände sind da in der Regel kein geeigneter Ersatz.
Im Zweifelsfall muss man dies intern in der Charger-Implementierung als Variable handhaben und dort auch bei Abweichungen mit der Box sebst syncen wenn das Standardzustände sind und der Nutzer kein "out of sync" sehen soll.
Für evcc ist dann alles prima.

andig, 2023-08-14:
> Der Code liesst sich, als sollte auch ein disable darüber abgewickelt werden, aber das kann nie vorkommen.

TLDR: richtig wäre wohl

	enablingRequired := enable != c.chargerEnabled

andig, 2023-08-14:
> und dort auch bei Abweichungen mit der Box sebst syncen wenn das Standardzustände sind und der Nutzer kein "out of sync" sehen soll.

@premultiply das gehört auf keinen Fall in den Charger weil es die Logik ja schon im Loadpoint gibt und die mittlerweile recht nachsichtig ist...