evcc-io/evcc #9310: Request to add Acrel ADW300(electric meter)  to evcc

912929782, 2023-08-09:
I want to use an electric meter in evcc. The model is Acrel ADW300. This is the official website of Acrel: https://de.acrel.de/   
The English instruction manual of this meter is attached below repository:  https://github.com/912929782/File
Please Help

andig, 2023-08-09:
We need the modbus register documentation.

andig, 2023-08-09:
@912929782 looks doable. Could you provide remote access to Modbus port for info@evcc.io for testing?

912929782, 2023-08-10:
@andig My meter is in the LAN, so it can't provide remote access. Could you please add it according to the register document, and then I will test it.

andig, 2023-08-10:
You can setup port forwarding.

andig, 2023-08-10:
Ich hätt gesagt wenn dann richtig? Aber fair, wir können das auch vereinfachen. Danke!

PR #9330 (2023-08-12): Add Acrel ADW300
Fixes #9310

@912929782 Please test!
closed 2023-08-12 by andig