evcc-io/evcc #9317: Option zur Abschaltung der Strom-Prüfung

hr29eg, 2023-08-09:
In der Diskussion [https://github.com/evcc-io/evcc/discussions/9254](url) wurde das Regelverhalten am Beispiel eines Hyundai Kona electric unter der Bedingung, dass im Fahrzeug per Einstellung die Ladeleistung reduziert wurde diskutiert. Dabei kommt es obwohl PV-Überschuss vorhanden ist nicht zur vollständig möglichen Nutzung für die Ladeleistung des EVs.

Als Ursache für die Abregelung der Ladeleistung wurde eine in [https://github.com/evcc-io/evcc/blob/master/core/loadpoint.go](url) zumindest dem dort angegeben Kommentar nach speziell für die ZOE eingebrachte Stromprüfung als hinderlich erkannt. 

```go
// effectiveCurrent returns the currently effective charging current
func (lp *Loadpoint) effectiveCurrent() float64 {
	if !lp.charging() {
		return 0
	}

	// adjust actual current for vehicles like Zoe where it remains below target
	if lp.chargeCurrents != nil {
		cur := max(lp.chargeCurrents)
		return math.Min(cur+2.0, lp.chargeCurrent)
	}

	return lp.chargeCurrent
}
``` 
Um das Regelverhalten zu verbessern, würde ich darum bitten, diese Stromprüfung per Konfiguration (z.B. in der evcc.yaml) abschaltbar zu gestalten.

Gruß Andreas.

andig, 2023-08-09:
> unter der Bedingung, dass im Fahrzeug per Einstellung die Ladeleistung reduziert wurde diskutiert

Was hat das mit dem Zoe-Limit zu tun?

VolkerK62, 2023-08-09:
Es ist nicht unbedingt ein Sonderfall. 
Die Idee mit der reduzierten Fahrzeugleistung ist nicht schlecht für Gegebenheiten in denen eine manuelle 1p/3p Umstellung nicht möglich ist. Dadurch ist bei 3p ein früherer Ladestart möglich.

Die Wallbox gibt 6A@3p frei.
Das Fahrzeug ist auf reduzierte Ladeleistung eingestellt und nimmt nur 2,5A pro Phase.
Bei der Berechnung von `pv charge current` wird jetzt nicht mit 6A + Überschuss gerechnet, sondern mit 2,5A effektiv zzgl. der 2A aus der "Zoe-Regelung". 
Also wird `pv charge current` nur mit 4,5A + Überschuss berechnet. Dadurch steigt die Ladeleistung aber nicht mehr und es wird Überschuss eingespeist.

Siehst du eine andere Möglichkeit, diese Korrektur zu umgehen?

hr29eg, 2023-08-09:
Der Sonderfall ist ja genau die in dem oben auszugsweise dargestellten Quelltext die Strombehandlung unter den Kommentar like Zoe. Ohne diese Sonderbehandlung würde der Regler in die aktuelle Runde mit dem Wert der Stromvorgabe aus der vorangegangen Runde gehen was erstmal der Normalfall wäre. 
Das würde auch passieren wenn die Wallbox keine Stromwerte messen und an EVCC melden würde. Ein Versuch diese Messung für die smartWB in der evcc.yaml abzuschalten hat nicht funktioniert.
Die Details und die Versuche dazu sind aber alle in der verlinkten Diskussion beschrieben.

andig, 2023-08-09:
Ich verstehe das Problem nicht. Warum stört es mit 4,5A zu rechnen wenn das Auto eh auf 2,5A hängen bleibt?

> Dadurch steigt die Ladeleistung aber nicht mehr und es wird Überschuss eingespeist.

Das tut sie ja anscheinend dank Limit sowieso nicht? Ich müsst es bitte erklären.

VolkerK62, 2023-08-09:
Ein Bild sagt mehr als Worte
https://github.com/evcc-io/evcc/discussions/9254#discussioncomment-6670968
Hier kann man die Auswirkungen schön sehen.
Ohne diese Korrektur müssten eigentlich 12,9A freigegeben werden (9,32 (letzte Freigabe) + 3,58 (Überschuss)).
Aufgrund der Korrektur bleibt es aber bei 9,xA 

>Und: warum nicht einfach auf die sinnfreie Option im Fahrzeug verzichten?

Wie oben schon gesagt, würde das die 3p Lademöglichkeiten erweitern.

StefanSchoof, 2023-08-10:
Habe mal aus Neugier im git blame geschaut:

Die Zoe Reglung wurde in #1431 eingebaut.
Es wurde schon kurz nach dem PR überlegt das zurück zu drehen. https://github.com/evcc-io/evcc/issues/1430#issuecomment-911809190
Und es gab schon einmal einen PR dazu konfigurieren #1472

andig, 2023-08-10:
> Ihr müsst es bitte erklären.

Ich hab keine Zeit, den Thread zu reverse-enginieren. Falls nicht möglich können wir hier schließen.

VolkerK62, 2023-08-10:
[image]

Du musst nicht die ganze Diskussion durchgehen. Leider gibt es den Log Ausschnitt nur als Bild.
Um 16:37:11 wird der Überschuss (3,58A) zu 5,72A addiert, obwohl vorher (16:37:05)  9,32A freigegeben wurden.
Die 5,72A ergeben sich aus dem größten Wert der charge currents zzgl. der 2A.

Wenn hier jetzt nicht 5,72A als Basis genommen würde, sondern 9,32A, dann würde pv charge current auf  12,9A gesetzt werden.

In den Issue geht es darum, die Möglichkeit zu haben, diese Phasenprüfung abzuschalten, um bewusst die Reduzierung am Auto nutzen zu können.

premultiply, 2023-08-10:
Die "+2.0" sorgen dafür, dass evcc den dem Fahrzeug signalisierten max. Phasenstrom nicht _beliebig_ weit aufdreht, sondern dies eben auf **+2A** des derzeitigen, gemessenen Phasenstrom begrenzt.

Hier könnte man jetzt überlegen diesen Wert einfach etwas zu erhöhen.

Irgendwann müsste man dann allerdings auch die **Steigerung**srate pro Zyklus auf +1A zu begrenzen um bei nichtlinearen Systemen das Schwingen bzw. Overshoot zu dämpfen...

VolkerK62, 2023-08-10:
Danke @StefanSchoof  für die investigative Tiefenbohrung ;)

Eigentlich ist damals alles diskutiert worden.
Es ist u. U. notwendig, es kann u.U. hinderlich sein.
Eine Möglichkeit dies an die individuellen Gegebenheiten anzupassen, würde zu einem unerwünschten neuen Parameter führen.

Bleibt aktuell zur Umgehung nur die Option, den Loadpoint so zu "manipulieren", dass keine Phasenströme geliefert werden.

andig, 2023-08-10:
Nach 10 Posts bleibt weiterhin unklar was das mit einer Ladebegrenzung im Auto zu tun hat. M.E. können wir hier schließen.

hr29eg, 2023-08-10:
> Nach 10 Posts bleibt weiterhin unklar was das mit einer Ladebegrenzung im Auto zu tun hat. M.E. können wir hier schließen.

Ich will Dir jetzt nicht zu nahe treten, aber mir schleicht sich das Gefühl ein, dass Du das wohl eher nicht verstehen willst. Dagegen war die Kompensationsidee von @VolkerK62 dankenswerter Weise richtig lösungsorientiert, um die problematischen Code-Zeilen zu umschiffen.

closed 2023-08-10 by andig