evcc-io/evcc #9321: MQTT-Output "chargeCurrent" float anstatt Ganzzahl

TheNavy, 2023-08-10:
Ich versuche in meinem EVCC meinen stufenlos Regelbaren PV-Heizstab darzustellen. Dazu habe ich einen Custom Charger erzeugt und greife den Wert "evcc/loadpoints/2/chargeCurrent" per MQTT ab. Funktioniert auch ohne Probleme außer dass er dort nur eine Ganzzahl übermittelt. Und 1A Sprünge bei einem 3P System sind ja direkt fast 700W was mir in einem Stufenlosen System zu viel sind.

Gibt es eine Möglichkeit, dass ein Custom Charger dort auch einen Float-Wert ausgibt? Geht ja bei den Wallboxen auch?!

Gruß
Richard

andig, 2023-08-10:
We would need to add `MaxCurrentMillis` as optional interface.

TheNavy, 2023-08-10:
#7338 Sounds nice :) Is ist possible for me to use some kind of workaround until the new features are implemented?

closed 2023-08-10 by andig