evcc-io/evcc #9339: Siemens Versicharge Modbus Fehlermeldung nach FW-Update

Maxheino, 2023-08-11:
### Describe the bug

Hallo,

meine Siemens Versicharge Wallbox hat sich ein Firmwareupdate gezogen, und seither bekomme ich in der evcc Oberfläche folgende Fehlermeldung:
charger: modbus: exception '2' (illegal data address), function '3'

### Steps to reproduce

1. Versicharge alte FW Version 2.124
2. Versicharge neue FW Version 2.128.24
3.
4.
...

### Configuration details

```yaml
meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.179.113  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.179.113  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.179.112  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery1

chargers:
- type: template
  template: versicharge 
  id: 2  
  host: 192.168.179.111
  port: 502
  modbus: tcpip  
  name: wallbox

vehicles:
- title: e-up 
  type: template
  template: id  
  user: ich@mail.de
  password: Geheim
  vin: XXXXXXXXXXXXXXXXXXXXXX
  capacity: 35
  targetSoc: 90
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: true

site:
  title: EIK-Home
  meters:
    grid: grid1
    pv:
    - pv1
    battery:
    - battery1
```

### Log details

```text
[site  ] DEBUG 2023/08/11 11:03:39 ----                                                                                                
[lp-1  ] DEBUG 2023/08/11 11:03:39 charge power: 3209W                                                                                 
[site  ] DEBUG 2023/08/11 11:03:39 pv power: 6346W                                                                                     
[site  ] DEBUG 2023/08/11 11:03:39 battery soc: 99%                                                                                    
[site  ] DEBUG 2023/08/11 11:03:39 battery power: -689W                                                                                
[site  ] DEBUG 2023/08/11 11:03:39 grid power: -1W                                                                                     
[site  ] DEBUG 2023/08/11 11:03:39 site power: -690W                                                                                   
[lp-1  ] DEBUG 2023/08/11 11:03:39 charge voltages: [229 228 231]V                                                                     
[lp-1  ] DEBUG 2023/08/11 11:03:39 detected connected phases: 3p                                                                       
[lp-1  ] DEBUG 2023/08/11 11:03:39 charge currents: [7 7 0]A                                                                           
[lp-1  ] DEBUG 2023/08/11 11:03:39 detected active phases: 2p                                                                          
[lp-1  ] DEBUG 2023/08/11 11:03:39 charge total import: 253.925kWh                                                                     
[lp-1  ] DEBUG 2023/08/11 11:03:39 charger status: C                                                                                   
[lp-1  ] DEBUG 2023/08/11 11:03:39 soc estimated: 34.10% (vehicle: 34.00%)                                                             
[lp-1  ] DEBUG 2023/08/11 11:03:39 vehicle soc: 34%                                                                                    
[lp-1  ] DEBUG 2023/08/11 11:03:39 vehicle soc limit: 0%                                                                               
[lp-1  ] DEBUG 2023/08/11 11:03:39 vehicle range: 86km                                                                                 
[lp-1  ] ERROR 2023/08/11 11:03:39 charger: modbus: exception '2' (illegal data address), function '3'
```

### What type of operating system are you running?

Docker container

### Version

0.118.11

mdkeil, 2023-08-11:
Habe mal die aktuellen Release-Note angehängt.. zwischen seiner Version und der aktuellsten wurden auf jeden Fall modbus-seitige Fehler gefixed. 
[VersichargeACSeriesFirmwareReleaseNotes_enUS2.128.pdf](https://github.com/evcc-io/evcc/files/12320227/VersichargeACSeriesFirmwareReleaseNotes_enUS2.128.pdf)

andig, 2023-08-11:
Fixed im Nightly.

closed 2023-08-11 by andig