evcc-io/evcc #9358: Loadpoint.syncCharger() not correcting currents

GrimmiMeloni, 2023-08-12:
### Describe the bug

@andig ich glaube der Change aus #9075 ist noch nicht sauber.
Ich hatte hier heute eine Situation in der der Charger mit zu hohen Amps geladen hat. Die neue Version von Loadpoint.syncCharger() hat das aber nicht korrigiert.

Das liegt meiner Ansicht nach daran, daß die Logik aktuell den Charger Current Abgleich nur macht, wenn Uneinigkeit bzgl. enabled zwischen Loadpoint und Charger herrscht. Aber selbst dann muß der Charger disabled sein, damit der Current abgleich überhaupt stattfindet. Auslöser ist 

https://github.com/evcc-io/evcc/blob/5ceafc803b7b91e080118a3106c5da8776c96d13/core/loadpoint.go#L650

Im Fall das der Charger enabled ist, steigt spätestens hier die Funktion immer aus. Damit der Current Abgleich in L665 wirklich hilft, sollte er auch in jedem Intervall erfolgen.  

### Steps to reproduce

N/A

### Configuration details

```yaml
N/A
```

### Log details

```text
[site  ] DEBUG 2023/08/12 18:25:25 ----
[lp-1  ] DEBUG 2023/08/12 18:25:25 charge power: 7696W
[site  ] DEBUG 2023/08/12 18:25:25 pv power: 2112W
[site  ] DEBUG 2023/08/12 18:25:25 battery soc: 99%
[site  ] DEBUG 2023/08/12 18:25:25 battery power: 4550W
[site  ] DEBUG 2023/08/12 18:25:25 grid power: 1054W
[site  ] DEBUG 2023/08/12 18:25:25 site power: 5704W
[lp-1  ] DEBUG 2023/08/12 18:25:25 charge currents: [31.7 0 0]A
[lp-1  ] DEBUG 2023/08/12 18:25:25 detected active phases: 1p
[lp-1  ] DEBUG 2023/08/12 18:25:25 charge total import: 1575.784kWh
[lp-1  ] DEBUG 2023/08/12 18:25:25 charger status: C
[lp-1  ] DEBUG 2023/08/12 18:25:25 soc estimated: 70.00% (vehicle: 70.00%)
[lp-1  ] DEBUG 2023/08/12 18:25:25 vehicle soc: 70%
[lp-1  ] DEBUG 2023/08/12 18:25:25 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/08/12 18:25:25 vehicle range: 350km
[lp-1  ] DEBUG 2023/08/12 18:25:25 phase timer inactive
[lp-1  ] DEBUG 2023/08/12 18:25:25 pv charge current: 0A = 6A + -24.8A (5704W @ 1p)
[site  ] DEBUG 2023/08/12 18:25:40 ----
[lp-1  ] DEBUG 2023/08/12 18:25:40 charge power: 7696W
[site  ] DEBUG 2023/08/12 18:25:40 pv power: 2228W
[site  ] DEBUG 2023/08/12 18:25:40 battery soc: 99%
[site  ] DEBUG 2023/08/12 18:25:40 battery power: 4570W
[site  ] DEBUG 2023/08/12 18:25:40 grid power: 908W
[site  ] DEBUG 2023/08/12 18:25:40 site power: 5578W
[lp-1  ] DEBUG 2023/08/12 18:25:40 charge currents: [31.7 0 0]A
[lp-1  ] DEBUG 2023/08/12 18:25:40 detected active phases: 1p
[lp-1  ] DEBUG 2023/08/12 18:25:40 charge total import: 1575.784kWh
[lp-1  ] DEBUG 2023/08/12 18:25:40 charger status: C
[lp-1  ] DEBUG 2023/08/12 18:25:40 soc estimated: 70.00% (vehicle: 70.00%)
[lp-1  ] DEBUG 2023/08/12 18:25:40 vehicle soc: 70%
[lp-1  ] DEBUG 2023/08/12 18:25:40 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/08/12 18:25:40 vehicle range: 350km
[lp-1  ] DEBUG 2023/08/12 18:25:40 pv charge current: 0A = 6A + -24.3A (5578W @ 1p)
```

### What type of operating system are you running?

Docker container

### Version

nightly incl. commit e02f689 (#9075)

andig, 2023-08-13:
> Die neue Version von Loadpoint.syncCharger() hat das aber nicht korrigiert.

Nur einen Zyklus lang oder gar nicht?

> Das liegt meiner Ansicht nach daran, daß die Logik aktuell den Charger Current Abgleich nur macht, wenn Uneinigkeit bzgl. enabled zwischen Loadpoint und Charger herrscht. 

Das war eigentlich Absicht- wenn ich mir über den Schaltzustand schon nicht einig bin dachte ich macht der Rest wenig Sinn. Ich würde aber erwarten, dass das nach einem Durchlauf wieder korrigiert wird?

andig, 2023-08-13:
Ach quatsch, ich sehe den Punkt. Ich mache einen anderen, etwas einfacheren Vorschlag.

closed 2023-08-13 by andig