evcc-io/evcc #9368: FoxEss Hybrid H3 Wechselrichter - Verbindung Modbus RS485 via Wifi-Adapter funktioniert, aber falsche PV-Power-Werte werden angezeigt

waldemar-doering, 2023-08-13:
### Describe the bug

Hi,
ich habe gestern unseren Fox ESS H3 Hybrid Wechselrichter mit RS485 und einem Elfin EW11 angeschlossen und ihn (nach mehreren Versuchen) erfolgreich mit EVCC verbinden können. 
Auch die Hesotec Ladesäule (Sponsortoken liegt vor) und unsere Fahrzeuge konnten integriert werden.
Die evcc.yaml habe ich nicht verändert. Sie wurde vom Programm erstellt. Soweit läuft es jetzt. 

Aber in der EVCC-Anzeige werden für den Eigenverbrauch und die Einspeisung falsche, zu hohe Werte angezeigt.

Ich habe einmal die Modbus-Register 31002 und 31005 manuell mit einer App ausgelesen. 
Die Werte darin entsprechen den Werten, die auch in der zugehörigen FoxESS-App angegeben werden und der Realität entsprechen.
 
In der auf Github hinterlegten templates/definition/meter/fox-ess-h3-ethernet.yaml sind diese Modbus-Register auch entsprechend angegeben.

Die in EVCC angezeigten Werte sind jedoch zu hoch bzw. falsch (siehe folgende Bilder, im Haus war der Verbrauch zu diesem Zeitpunkt bei ca. 200W). Kann es sein, dass EVCC die falschen Werte abruft oder verarbeitet?

[image]
[image]
[image]

Die evcc.yaml sowie eine Log-Datei habe ich angefügt. 
Im Log werden für "PV Power" Werte angezeigt, die ungefähr um den **Faktor 2** zu hoch sind (Die Werte in der FoxEXX Cloud sind ungefähr halb so groß).

Der Vollständigkeit halber der Hinweis, dass wir zwei PV-Strings am Wechselrichter haben. 
mit dem manuellen Auslesen konnte man sehen, dass in den beiden Modbus-Registern sind die PV-Werte je String enthalten sind (31002 = String 1, 31005 = String 2) enthalten. Wird dies von EVCC berücksichtigt?

Ich hoffe, es hilft weiter und es gibt bald eine Lösung dazu.

Beste Grüße
Waldemar Döring

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxxxxxxxxxxxxxxxx

interval: 10s # control cycle interval

sponsortoken: #vorhanden, aber ausgeblendet

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: fox-ess-h3
  id: 247
  host: 192.168.178.29
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: fox-ess-h3
  id: 247
  host: 192.168.178.29
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: fox-ess-h3
  id: 247
  host: 192.168.178.29
  port: 502
  usage: pv
  modbus: tcpip
  name: pv3
- type: template
  template: fox-ess-h3
  id: 247
  host: 192.168.178.29
  port: 502
  usage: battery
  modbus: tcpip
  name: battery4

chargers:
- type: template
  template: hesotec
  id: 1
  host: 192.168.178.66
  port: 502
  modbus: tcpip
  name: wallbox7

vehicles:
- type: template
  template: hyundai
  title: IONIQ5
  user: xxxxxxxxxxxxxxxxxxx
  password: xxxxxxxxxxxx
  capacity: 72.6
  language: de
  name: ev5
- type: template
  template: enyaq
  title: ENYAQ
  user: xxxxxxxxxxxxxxxxx
  password: xxxxxxxxxxxxxx
  capacity: 77
  name: ev6

loadpoints:
- title: Parkplatz
  charger: wallbox7
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: false

site:
  title: Daheim
  meters:
    grid: grid1
    pv:
    - pv2
    - pv3
    battery:
    - battery4
```

### Log details

```text
[main  ] INFO 2023/08/13 12:55:58 evcc 0.118.11
[main  ] INFO 2023/08/13 12:55:58 using config file: /etc/evcc.yaml
[main  ] INFO 2023/08/13 12:55:58 starting ui and api at :7070
[db    ] INFO 2023/08/13 12:55:58 using sqlite database: /home/pi/.evcc/evcc.db
[site  ] WARN 2023/08/13 12:56:02 battery configured but residualPower is missin                                                                                                                                                             g (add residualPower: 100 to site)
[site  ] INFO 2023/08/13 12:56:02 site config:
[site  ] INFO 2023/08/13 12:56:02   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/08/13 12:56:02     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/08/13 12:56:02     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/08/13 12:56:02     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/08/13 12:56:02     battery 1: power ✓ energy ✗ currents ✗ soc                                                                                                                                                              ✓ capacity ✗
[site  ] INFO 2023/08/13 12:56:02   vehicles:
[site  ] INFO 2023/08/13 12:56:02     vehicle 1: range ✓ finish ✓ status ✓ clima                                                                                                                                                             te ✗ wakeup ✓
[site  ] INFO 2023/08/13 12:56:02     vehicle 2: range ✓ finish ✓ status ✓ clima                                                                                                                                                             te ✗ wakeup ✗
[lp-1  ] INFO 2023/08/13 12:56:02 loadpoint 1:
[lp-1  ] INFO 2023/08/13 12:56:02   mode:        minpv
[lp-1  ] INFO 2023/08/13 12:56:02   charger:     power ✓ energy ✓ currents ✓ pha                                                                                                                                                             ses ✗ wakeup ✗
[lp-1  ] INFO 2023/08/13 12:56:02   meters:      charge ✓
[lp-1  ] INFO 2023/08/13 12:56:02     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/08/13 12:56:02 phase timer inactive
[lp-1  ] DEBUG 2023/08/13 12:56:02 pv timer inactive
[lp-1  ] DEBUG 2023/08/13 12:56:02 guard timer inactive
[lp-1  ] DEBUG 2023/08/13 12:56:02 max charge current: 6A
[site  ] WARN 2023/08/13 12:56:02 interval <30s can lead to unexpected behavior,                                                                                                                                                              see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2023/08/13 12:56:02 ----
[lp-1  ] DEBUG 2023/08/13 12:56:02 charge power: 0W
[site  ] DEBUG 2023/08/13 12:56:02 pv power: 8375W
[site  ] DEBUG 2023/08/13 12:56:03 battery soc: 99%
[site  ] DEBUG 2023/08/13 12:56:03 battery power: 349W
[site  ] DEBUG 2023/08/13 12:56:03 grid power: 189W
[site  ] DEBUG 2023/08/13 12:56:03 site power: 538W
[lp-1  ] DEBUG 2023/08/13 12:56:03 charge voltages: [0 0 0]V
[lp-1  ] DEBUG 2023/08/13 12:56:03 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/08/13 12:56:03 charge total import: 0.000kWh
[lp-1  ] DEBUG 2023/08/13 12:56:03 charger status: C
[lp-1  ] DEBUG 2023/08/13 12:56:03 charge start energy: 0kWh
[lp-1  ] INFO 2023/08/13 12:56:03 car connected
[lp-1  ] DEBUG 2023/08/13 12:56:03 vehicle api refresh
[lp-1  ] DEBUG 2023/08/13 12:56:03 pv timer elapse
[lp-1  ] DEBUG 2023/08/13 12:56:03 pv timer inactive
[lp-1  ] DEBUG 2023/08/13 12:56:03 charger: guard elapse
[lp-1  ] DEBUG 2023/08/13 12:56:03 guard timer inactive
[lp-1  ] DEBUG 2023/08/13 12:56:03 charge total import: 0.000kWh
[lp-1  ] DEBUG 2023/08/13 12:56:03 charge start energy: 0kWh
[lp-1  ] INFO 2023/08/13 12:56:03 start charging ->
[lp-1  ] DEBUG 2023/08/13 12:56:03 wake-up timer: stop
[main  ] DEBUG 2023/08/13 12:56:03 vehicle status: A (IONIQ5)
[main  ] DEBUG 2023/08/13 12:56:04 vehicle status: C (ENYAQ)
[lp-1  ] INFO 2023/08/13 12:56:04 vehicle updated: unknown -> ENYAQ
[lp-1  ] DEBUG 2023/08/13 12:56:04 vehicle status: C
[lp-1  ] DEBUG 2023/08/13 12:56:04 vehicle soc: 85%
[lp-1  ] DEBUG 2023/08/13 12:56:04 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/08/13 12:56:04 vehicle range: 414km
[lp-1  ] DEBUG 2023/08/13 12:56:04 pv charge current: 1.22A = 2A + -0.78A (538W                                                                                                                                                              @ 3p)
[site  ] DEBUG 2023/08/13 12:56:12 ----
[lp-1  ] DEBUG 2023/08/13 12:56:12 charge power: 0W
[site  ] DEBUG 2023/08/13 12:56:12 pv power: 7708W
^C[site  ] DEBUG 2023/08/13 12:56:13 battery soc: 99%
[site  ] DEBUG 2023/08/13 12:56:13 battery power: 816W
[lp-1  ] DEBUG 2023/08/13 12:56:13 charge total import: 0.000kWh
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.118.11

andig, 2023-08-13:
Duplikat von https://github.com/evcc-io/evcc/issues/9265?

jogger166, 2024-06-10:
Hallo Waldemar, ich bin auf Deine Problembeschreibung gestoßen, ich habe ein Problem, was Du scheinbar längst gelöst hast: Das Fox ESS H3 System über einen Elfin EW11 (Modbus RTU nach Modbis TCP) an EVCC verbinden.  Ich habe das evcc auf einem PI gestartet, bekomme aber keinerlei Messwerte vom Fox. In der Elfin Systemübersicht  wird unter Communication State auch keine Server-IP angezeigt, leider kann ich nicht herausbekommen, wie dort die IP des evcc-System (also des PI) reinkommt. Wie hast Du das geschafft? Und muss in evcc zusätzlich noch ein Template hinterlegt werden, welches die Umsetzung von Modbus RTU nach TCP im Elfin definiert? Falls ja, könntest Du das posten? 
Danke im  Voraus.

waldemar-doering, 2024-06-11:
Moin jogger166, in meiner yaml ist unter meters: der host 192.168.178.29 die IP vom Elfin. Genau weiß ich es nicht mehr, aber im Elfin-Setup musste man die Zugangsdaten von seinem WLAN angeben. Im Router konnte man dann sehen, welche IP vergeben wurde. Die habe ich dann fest vergeben und in der yaml eingetragen. 
Aber was genau ich im Elfin Setup zum H3 eingetragen habe, müsste ich noch mal nachschauen. Grüße Waldemar

waldemar-doering, 2024-06-11:
Wenn der Elfin richtig ins Heimnetz eingebunden und die Verbindung zum H3 steht, erhält auch der H3 im Router eine eigene IP. Auch die habe ich statisch gemacht.
Oder geht es um die Werte der Modbusverbindung zwischen Elfin und H3?

waldemar-doering, 2024-06-11:
[image]

waldemar-doering, 2024-06-11:
Das sind meine Einträge im Elfin Setup unter Serial Port Settings

waldemar-doering, 2024-06-11:
Ich musste übrigens im H3 Menü die Modbus ID auf 247 setzen. Ich meine, die war bei 1 und damit funktionierte es nicht ( siehe Eintrag id: 247 in der yaml)