evcc-io/evcc #9373: copy-paste error in em2go.go line 250: func (wb *Em2Go) Diagnose()

RenatusRo, 2023-08-13:
### Describe the bug

em2GoRegMaxCurrent should be em2GoReg**Min**Current

in em2go.go line 250

	if b, err := wb.conn.ReadHoldingRegisters(em2GoRegMaxCurrent, 1); err == nil {
		fmt.Printf("\tEVSE Min. Current:\t%.1fA\n", float64(binary.BigEndian.Uint16(b)/10))
	}

### Steps to reproduce

read the code :-)

### Configuration details

```yaml
doesn't matter
```

### Log details

```text
Power:          0W
Energy:         0.0kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 234V 235V 236V
Charge status:  B
Enabled:        false
Charged:        0.0kWh
Duration:       48m0s

Diagnostic dump:
	Charging Station Status:	2
	Connector State:	1
	Error Code:	0
	EVSE Max. Current:	32.0A
	EVSE Min. Current:	32.0A
	Cable Max. Current:	32.0A
	Serial: EM2Go0800
	Safe Current:	32.0A
	Connection Timeout:	30
	Current Limit:	6.0A
	Charge Mode:	1
	Charge Command:	2
```

### What type of operating system are you running?

Linux

### Version

0.118.11

andig, 2023-08-13:
Fixed, thank you!

closed 2023-08-13 by andig

RenatusRo, 2023-08-13:
incredible fast :-)